espressif/esptool #687: ESP32-S2 32MB, 64MB, 128MB being found by esptool.py

TGiles1998, 2021-10-29:
# Operating system
macOS Mojave
# Python version
3.10
# What Chip
ESP32 S2
# What development board or other hardware is the chip attached to

DevKitC

If your hardware is custom or unusual, please attach a photo to the issue.

Photo Attached

[image]

# Is anything else attached to the development board, except for the serial flasher connections?

External Flash WSON-8.

EFUSE CONFIG

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                           Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                             If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                            If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                               8MHz clock freq override                           = 53 R/W (0x35)
SPI_PAD_CONFIG_CLK (BLOCK0):                       Override SD_CLK pad (GPIO6/SPICLK)                 = 6 R/W (0b00110)
SPI_PAD_CONFIG_Q (BLOCK0):                         Override SD_DATA_0 pad (GPIO7/SPIQ)                = 7 R/W (0b00111)
SPI_PAD_CONFIG_D (BLOCK0):                         Override SD_DATA_1 pad (GPIO8/SPID)                = 8 R/W (0b01000)
SPI_PAD_CONFIG_HD (BLOCK0):                        Override SD_DATA_2 pad (GPIO9/SPIHD)               = 9 R/W (0b01001)
SPI_PAD_CONFIG_CS0 (BLOCK0):                       Override SD_CMD pad (GPIO11/SPICS0)                = 5 R/W (0b00101)
DISABLE_SDIO_HOST (BLOCK0):                        Disable SDIO host                                  = False R/W (0b0)

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

NO IDE, Bash Terminal

# Full esptool.py command line that was run:

esptool.py --no-stub --port /dev/cu.SLAB_USBtoUART flash_id

# Full output from esptool.py
esptool.py v3.3-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:61:05:11:a0:b0
Enabling default SPI flash mode...
Manufacturer: 00
Device: 4048
Detected flash size: Unknown
Hard resetting via RTS pin...

# Do you have any other information from investigating this?

It won't detect the 64MB or 128MB, just the 32MB manufacturer.

Winbond Flash's tried:

W25Q256JVEIM TR

W25Q512JVEIQ

W25Q01JVZEIQ

# Is there any other information you can think of which will help us reproduce this problem?

No

ESP32DE, 2021-10-30:
@TGiles1998 

hi, you wrote ESP32-S2 but you are testing ESP32

```
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)

```

FYI:

W25Q256JVEIM TR
EF 70 `19`
is in da esptool-house

W25Q512JVEIQ
EF 40 `20`
you can have a try : [add here](https://github.com/espressif/esptool/blob/master/esptool.py#L124-L126)

[image]

example: 
0x18: '16MB', 0x19: '32MB', 0x1a: '64MB', `0x20: '64MB',` 0x21: '128MB'

looks then like this
[image]

W25Q01JVZEIQ
EF  40 `21`
is in da esptool-house

esp32-s2, esp32-s3 tested. 

[image]

you can try the last master, it was merged last days.
use the latest esptool - 

try 
esptool.py `--no-stub` --port <PORT> flash_id

TGiles1998, 2021-10-30:
Thank you for your reply!

I have just switched to my S3 board instead and ran with and without the stubb for the 32MB and it unfortunately still doesn't work, I also edited the esptool.py script.

$ esptool.py --no-stub --chip esp32-S3 --port /dev/cu.SLAB_USBtoUART flash_id
esptool.py v3.3-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:b4:72:70:0d:c0
Enabling default SPI flash mode...
Manufacturer: 07
Device: ffff
Detected flash size: Unknown
Hard resetting via RTS pin...

$ esptool.py --chip esp32-S3 --port /dev/cu.SLAB_USBtoUART flash_id
esptool.py v3.3-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting...
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:b4:72:70:0d:c0
Uploading stub...
Running stub...
Stub running...
Manufacturer: 01
Device: ffff
Detected flash size: Unknown
Hard resetting via RTS pin...

Picture of setup

[image]

I hope I am doing something really wrong as Ive been trying to get external flash to wokr above 16MB for quite some time now.

Thanks in advance!

ESP32DE, 2021-10-30:
@TGiles1998 

i add numbers to your pic and the wson.

[image]

the wiring looks from numbering ok - but the one i think is - that the (long unprotected / not twisted ) wire can interfere, because these act like antennas and are very susceptible to interference at high speed >20 MHz. 

you are talented from the pic - so you can try to mount the WSON-8 direct on the SOIC-8 300
this is the challenge :

- isolate the EEpad on WSON-8, so there is no bridge to the individual SOIC pads / landing pads. 

[image]

cut the kapton on edge of the wson-8 and paste and solder it directly.
the resistance to the SoC is very tight but it works.
 
[image]

it should work with the NOR WB1Gbit, and  usually with the 32MB / 64 MB too

if you can not solder directly you can try to shorten the wire  `and`  use gnd wire and twist each dataline also the clk line.
[image]

TGiles1998, 2021-10-31:
Hi,

I attached some Kapton tape to the back of WSON-8 flash.

[image]

Then ran it through on the S3 board in my infrared reflow oven. And Voila!

[image]

$ esptool.py --chip esp32-S3 --port /dev/cu.SLAB_USBtoUART flash_id
esptool.py v3.3-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:b4:72:70:00:78
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4021
Detected flash size: 128MB
Hard resetting via RTS pin...

Worked first time and with the Stub. 

Amazing thank you so much!!

radimkarnis, 2021-11-01:
Great results, @TGiles1998! Thank you @ESP32DE for helping with this issue.

Do you want to discuss some more or can this be closed now?

closed 2021-11-01 by TGiles1998