espressif/esptool #690: Flash firmware bin file

Roadrunner185, 2021-11-11:
Hello,

I want to flash an ESP32 with firmware.bin file using the esptool. It's flashing ("write_flash") without any failure. But it's not starting correctly.
If I use Visual Studio Code and platformio it works.
If I compare both outputs it seems that with the esptool command line the firmware will be written to one location whereas with VS Code the firmware will be flashed to different locations/segments. 

The firmware bin file includes different segments. Is there a problem to write a bin file with more segments with esptool?

closed 2021-11-11 by Roadrunner185

radimkarnis, 2021-11-11:
Hello @Roadrunner185,

It is not enough to simply flash the firmware.bin file that PlatformIO produces. There are also other things, such as a bootloader, that need to be flashed at the right offsets.

> Is there a problem to write a bin file with more segments with esptool?

The short answer is no - PlatfromIO uses the same esptool under the hood. That means that if you manage to get the exact esptool command it runs, you will have your solution. Follow [this great answer](https://community.platformio.org/t/export-of-binary-firmware-files-for-esp32-download-tool/9253/2) to see how to do that!

If that doesn't help, can you please the full esptool command you run?

Radim

Roadrunner185, 2021-11-11:
Hello @radimkarnis,

I've just found and fixed the problem. You're right, the problem was that I have also to flash other bin files...

Best Regards
Sebastian