espressif/esptool #691: idf.py monitor cannot see log, miniterm can see correct output

twlawrenceko, 2021-11-11:
Hello,

When trying to run `idf.py flash monitor` I got the below error:
```
Serial port /dev/ttyUSB0
Connecting.......................
/dev/ttyUSB0 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
No serial ports found. Connect a device, or use '-p PORT' option to set a specific port.
```

However, when running `miniterm /dev/ttyUSB0 115200` it can correctly reads all the log (previously flashed in hello_world example). 

When trying to run `idf.py flash monitor`, miniterm shows the ESP has been rebooted. Moreover, I can see the TXD and RXD light on CP2102 blinking.

dobairoland, 2021-11-12:
Hi @twlawrenceko. I'm sorry but I don't see why is this an esptool issue. `idf.py monitor` works for you but `idf.py flash monitor` isn't? What are the full logs? Or this is happening immediately in the beginning, ie. you've getting the same for `idf.py flash`?

twlawrenceko, 2021-11-12:
> Hi @twlawrenceko. I'm sorry but I don't see why is this an esptool issue. `idf.py monitor` works for you but `idf.py flash monitor` isn't? What are the full logs? Or this is happening immediately in the beginning, ie. you've getting the same for `idf.py flash`?

Hello @dobairoland, both `idf.py flash monitor` and `idf.py monitor` are not detecting the ESP32. However, I can see with using miniterm. 

This happens after populating one cable, but it seems like receiving is doing ok as miniterm can see it.

dobairoland, 2021-11-15:
Could you please try to run the `esptool.py chip_id` command and share the full output (you can remove the MAC address of course)?

twlawrenceko, 2021-11-15:
Hello, after further inspection, the capacitor is not functional anymore, which swapping a new one fixed it

closed 2021-11-15 by twlawrenceko