espressif/esptool #693: Can't burn efuses on ESP32C3 for external flash SPI

SensorNetOrg, 2021-11-20:
* Operating system: Arch Linux
* Python version:  3.9

# Full esptool.py command line as run:

espefuse.py burn_efuse  SPI_PAD_CONFIG_CS  21

# Full output from esptool.py
Connecting...
Detecting chip type... ESP32-C3
espefuse.py v3.2-dev
The efuses to burn:
  from BLOCK1
     - SPI_PAD_CONFIG_CS
    ATTENTION! This BLOCK uses NOT the NONE coding scheme and after 'BURN', these efuses can not be burned in the feature:
               ['MAC', 'SPI_PAD_CONFIG_CLK', 'SPI_PAD_CONFIG_Q', 'SPI_PAD_CONFIG_D', 'SPI_PAD_CONFIG_HD']
               ['SPI_PAD_CONFIG_WP', 'SPI_PAD_CONFIG_DQS', 'SPI_PAD_CONFIG_D4', 'SPI_PAD_CONFIG_D5', 'SPI_PAD_CONFIG_D6']
               ['SPI_PAD_CONFIG_D7', 'WAFER_VERSION', 'PKG_VERSION', 'BLOCK1_VERSION']

Burning efuses (see 'ATTENTION!' above):

    - 'SPI_PAD_CONFIG_CS' (SPI CS pad) 0b000000 -> 0b010101

Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
[01] MAC_SPI_8M_0         is not empty
        (written ): 0x00057a9060b0c258010c00000000000000007cdfa1b0682c
        (to write): 0x000000000000000000000000000000540000000000000000
        (coding scheme = RS)

A fatal error occurred:         Burn into MAC_SPI_8M_0 is forbidden (RS coding scheme does not allow this).

# What is the expected behaviour?

SPI_PAD_CONFIG_CS is burned

# Do you have any other information from investigating this?

Connecting...
Detecting chip type... ESP32-C3
espefuse.py v3.2-dev
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
VDD_SPI_AS_GPIO (BLOCK0)                           Set this bit to vdd spi pin function as gpio       = False R/W (0b0)
BTLC_GPIO_ENABLE (BLOCK0)                          Enable btlc gpio                                   = 0 R/W (0b00)
POWERGLITCH_EN (BLOCK0)                            Set this bit to enable power glitch function       = False R/W (0b0)
POWER_GLITCH_DSENSE (BLOCK0)                       Sample delay configuration of power glitch         = 0 R/W (0b00)
DIS_LEGACY_SPI_BOOT (BLOCK0)                       Disables Legacy SPI boot mode                      = False R/W (0b0)
UART_PRINT_CHANNEL (BLOCK0)                        Selects the default UART for printing boot msg     = UART0 R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FORCE_SEND_RESUME (BLOCK0)                         Force ROM code to send a resume command during SPI = False R/W (0b0)
                                                    bootduring SPI boot                              
BLOCK_USR_DATA (BLOCK3)                            User data                                         
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Efuse fuses:
WR_DIS (BLOCK0)                                    Disables programming of individual eFuses          = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                                    Disables software reading from BLOCK4-10           = 0 R/W (0b0000000)

Flash Config fuses:
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms                                              
FLASH_ECC_MODE (BLOCK0)                            Set this bit to set flsah ecc mode.               
   = flash ecc 16to18 byte mode R/W (0b0)
[... 76 lines cut ...]
SPI_PAD_CONFIG_CLK (BLOCK1)                        SPI CLK pad                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_Q (BLOCK1)                          SPI Q (D1) pad                                     = 0 R/W (0b000000)
SPI_PAD_CONFIG_D (BLOCK1)                          SPI D (D0) pad                                     = 0 R/W (0b000000)
SPI_PAD_CONFIG_CS (BLOCK1)                         SPI CS pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_HD (BLOCK1)                         SPI HD (D3) pad                                    = 0 R/W (0b000000)
SPI_PAD_CONFIG_WP (BLOCK1)                         SPI WP (D2) pad                                    = 0 R/W (0b000000)
SPI_PAD_CONFIG_DQS (BLOCK1)                        SPI DQS pad                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_D4 (BLOCK1)                         SPI D4 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D5 (BLOCK1)                         SPI D5 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D6 (BLOCK1)                         SPI D6 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D7 (BLOCK1)                         SPI D7 pad                                         = 0 R/W (0b000000)

Usb Config fuses:
DIS_USB_JTAG (BLOCK0)                              Disables USB JTAG. JTAG access via pads is control = False R/W (0b0)
                                                   led separately                                    
DIS_USB_DEVICE (BLOCK0)                            Disables USB DEVICE                                = False R/W (0b0)
DIS_USB (BLOCK0)                                   Disables the USB OTG hardware                      = False R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                            Exchanges USB D+ and D- pins                       = False R/W (0b0)
DIS_USB_DOWNLOAD_MODE (BLOCK0)                     Disables use of USB in UART download boot mode     = False R/W (0b0)

Vdd_Spi Config fuses:
PIN_POWER_SELECTION (BLOCK0)                       GPIO33-GPIO37 power supply selection in ROM code   = VDD3P3_CPU R/W (0b0)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = False R/W (0b0)

KonstantinKondrashov, 2021-11-22:
Hi @SensorNetOrg!

It is not a bug. It is the correct behavior. For esp32/s/c chips the eFuse BLOCK1 has [Reed-Solomon coding scheme](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/efuse.html#supported-coding-scheme). The limitation for BLOCK1-10 - they can only be written to one time.

I see that BLOCK1 has data:
```
MAC (BLOCK1) Factory MAC Address
= 7c:df:a1:b0:68:2c (OK) R/W
WAFER_VERSION (BLOCK1) WAFER version = 3 R/W (0b011)
PKG_VERSION (BLOCK1) Package version = ESP32-C3 R/W (0b000)
BLOCK1_VERSION (BLOCK1) BLOCK1 efuse version = 0 R/W (0b000)
```
You can not write new data to BLOCK1-10 if it has already been written.

SensorNetOrg, 2021-11-22:
@KonstantinKondrashov 
I haven't burned anything to BLOCK1.

Why did the factory pre-burned BLOCK1 before they sold the chip?

What is the use of the eFuses if they are pre-burned at the factory?

closed 2021-11-23 by SensorNetOrg

KonstantinKondrashov, 2021-11-23:
Hi @SensorNetOrg!

> Why did the factory pre-burned BLOCK1 before they sold the chip?

because BLOCK1 has: Factory MAC Address, WAFER version, Package version ... they definitely must be set at the Factory stage.
Seems you need to contact Espressif to order chips with pre-burned custom data in eFuses.

ONLY FOR TESTS (not recommended): You can try to suppress this error to burn new data using the espefuse.py (change the espefuse.py so to skip `A fatal error occurred: Burn into MAC_SPI_8M_0 is forbidden (RS coding scheme does not allow this)`.). But the encoding data for the BLOCK (hidden part) will be broken.

skip this part of code in espefuse.py
```py
                        print("\t(coding scheme = RS)")
                        error_msg = "\tBurn into %s is forbidden (RS coding scheme does not allow this)." % (self.name)
                        self.parent.print_error_msg(error_msg)
```

kaseiiro, 2024-08-10:
Hi @KonstantinKondrashov,

Suppressing this error to burn a new data portion is pretty simple to implement, that's not the question.
But it leads us to another stumbling-block: when **reading** this data back, the RS-decoder tries to fix errors, thus ruins the data (as the RS codes are no more valid). Is there a trick to disable the RS-decoder while reading? Nothing special, I'm just curious. Thanks.

KonstantinKondrashov, 2024-08-12:
Hi @kaseiiro!
No, there is no way to disable RS-error correction. From the doc https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf

```
After that, the hardware burns into eFuse the 44-byte codeword consisting of the data bytes followed by the parity bytes. 
When the eFuse block is read back, the eFuse controller automatically decodes the codeword and applies error correction if needed. 
Because the RS check codes are generated on the entire 256-bit eFuse block, each block can only be written once.
```

I don't know if it works but what if you make so many errors that RS-correction will not be done? It is just an idea.

kaseiiro, 2024-08-12:
> I don't know if it works but what if you make so many errors that RS-correction will not be done? It is just an idea.

Unfortunately not. Having 12 error bytes out of 32 (and likely 12 of 12 RS bytes damaged), the RS-decoder still introduses more errors. But still thanks!