espressif/esptool #694: ESP8266 module ESP-12F does not connect to esptool.py

karstengit, 2021-11-20:
I have bought some blank ESP-12F modules like this: https://www.aliexpress.com/item/1005001520382593.html

For these modules I have made my own adapter PCB to connect them, but now I don't get any connection with the esptool.py to these modules.

The connection is:
```
TX --------------- RX with an PL2303 USB adapter
RX --------------- TX with an PL2303 USB adapter
EN --------------- 3.3V
GPIO-0 ----------- GND over button
GPIO-15 ---------- GND over 4K7 resistor
RESET ------------ GND over button
GND -------------- GND
VCC -------------- 3.3V from AMS1117-3.3 blocked with Elko 220µF
```

With a serial terminal I can cleary see with 74880 baud the response for standard boot on TXD (GPIO1):
```
 ets Jan  8 2013,rst cause:2, boot mode:(7,7)

waiting for host
```
and on TXD1 (GPIO2)

```
 ets Jan  8 2013,rst cause:2, boot mode:(7,6)

waiting for host
```
----

For flash mode with RESET and GPIO-0 on TXD (GPIO1):
```
 ets Jan  8 2013,rst cause:2, boot mode:(5,7)

waiting for host
```
and on TXD1 (GPIO2)

```
 ets Jan  8 2013,rst cause:2, boot mode:(5,6)

waiting for host
```
----
I tried to connect with:

python3 esptool.py --port /dev/ttyUSB0 flash_id<br>
or<br>
python3 esptool.py --chip esp8266 --port /dev/ttyUSB0 flash_id<br>
or<br>
python3 esptool.py --no-stub --port /dev/ttyUSB0 flash_id

but no connection can be established.

"A fatal error occurred: Failed to connect to Espressif device: No serial data received."

With the trace option I get:

```
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.082 Read 1 bytes: 72
TRACE +0.000 Read invalid data: 72
TRACE +0.000 Remaining data in serial buffer: 6c00
```

I tried the connection with these two versions:

esptool.py v3.3-dev<br>
esptool.py v2.0-beta2

I assume this is simply bad hardware.<br>
Is there anybody who has mastered this problem?

----

Running on Debian GNU/Linux 11 (bullseye)

with Python 3.9.2

----

```
$ python3 esptool.py --chip esp8266 --port /dev/ttyUSB0 flash_id
esptool.py v3.3-dev
Serial port /dev/ttyUSB0
Connecting......................................

A fatal error occurred: Failed to connect to ESP8266: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

Tracing:
```
TRACE +0.100 No serial data received.
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.009 Read 1 bytes: 72
TRACE +0.000 Read invalid data: 72
TRACE +0.000 Remaining data in serial buffer: 6c
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 No serial data received.
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 No serial data received.
```

jandrassy, 2021-11-20:
you have io 15 HIGH

karstengit, 2021-11-21:
> you have io 15 HIGH. Check the pull-down.

No - it is definitely LOW - i already checked this.

> GPIO-15 ---------- GND over 4K7 resistor

karstengit, 2021-11-21:
Your analyze is correct, although i could not believe it first, because i could measure 5mV at the pin outside, that should be correct.

After a little bit thinking i did check the solder joint to the module and it was cold with a bad contact!

So i resoldered it and now it is running - thank you very much!

Where can i find an explanation of the other numbers?

closed 2021-11-21 by karstengit