espressif/esptool #695: Serial.end() in arduino sketch setup method bricked my ESP32S2

lcsjunior, 2021-11-23:
Serial.end() in arduino sketch setup method bricked my ESP32S2.

Problem:

When uploading to my S2 mini with Arduino IDE, I usually have to upload sometimes. 1x Serial port busy error occurs. 2x upload successfully. In an attempt to solve it, I put Serial.end() at the end of the setup method, but now I can't upload it anymore. My ESP died?

```cpp
void setup() {
  Wire.begin(_SDA_PIN, _SCL_PIN);
  Serial.begin(115200);

  if (sht.init()) {
    sht.setAccuracy(SHTSensor::SHT_ACCURACY_HIGH);
  } else {
    Serial.println(F("SHT initialization failed"));
  }

  display.init(115200);
  u8g2Fonts.begin(display);
  **Serial.end();**
}
```

Operating system: Ubuntu 20.04

Python version: Python 3.6.15

What Chip: ESP32-S2

What development board or other hardware is the chip attached to: S2 Mini V 1.0.0 Lolin
[](https://www.wemos.cc/en/latest/s2/s2_mini.html)

Are you running esptool.py from an IDE such as Arduino or Eclipse?

Example: Arduino IDE

Full esptool.py command line that was run:

**esptool.py --chip esp32s2 erase_flash

Full output from esptool.py
esptool.py v3.2
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting....
/dev/ttyACM0 failed to connect: Write timeout

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.**

radimkarnis, 2021-11-23:
Hello @lcsjunior,

In Arduino, USB does require specifically to be turned on (or Serial selected to go to USB CDC). Else USB is not on and you can't just upload to the board over it.

I presume you are using the [Wemos S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html) board. If you put it into the download mode manually (connect the board to your PC and press the `RST` button while holding the `0` button), the USB flash should work again. Please, let me know if that worked.

Radim

lcsjunior, 2021-11-23:
Hi,

Earlier I tested it with usb TTL and gave the same error.
@radimkarnis I followed your advice, again I put it in the manual download and managed to reset the S2 mini.
`esptool.py --chip esp32s2 erase_flash`

I already had USB CDC enabled in Arduino IDE and managed to upload it to the device.

In time, I will resume uploads via OTA.
But, by the way, do you know why I can only upload the second time?
The first time, it occurs:

esptool.py v3.1
Serial port /dev/ttyACM0
Connecting...
Traceback (most recent call last):
  File "/snap/arduino/61/lib/python3.6/site-packages/serial/serialposix.py", line 621, in write
    n = os.write(self.fd, d)
OSError: [Errno 19] No such device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/lc/snap/arduino/61/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 4582, in <module>
    _main()
  File "/home/lc/snap/arduino/61/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 4575, in _main
    main()
  File "/home/lc/snap/arduino/61/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 4076, in main
    before=args.before)
  File "/home/lc/snap/arduino/61/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 121, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/home/lc/snap/arduino/61/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 629, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False, usb_jtag_serial=usb_jtag_serial)
  File "/home/lc/snap/arduino/61/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 596, in _connect_attempt
    self.sync()
  File "/home/lc/snap/arduino/61/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 487, in sync
    timeout=SYNC_TIMEOUT)
  File "/home/lc/snap/arduino/61/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 425, in command
    self.write(pkt)
  File "/home/lc/snap/arduino/61/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py", line 388, in write
    self._port.write(buf)
  File "/snap/arduino/61/lib/python3.6/site-packages/serial/serialposix.py", line 655, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 19] No such device
the selected serial port serial.serialutil.SerialException: write failed: [Errno 19] No such device
 does not exist or your board is not connected

radimkarnis, 2021-11-23:
To be honest, I am not sure about the exact implementation of USB-CDC mode. Because it is fully implemented in software, there are certain cases when the port just disappears or is not initialized (that is why USB CDC has to be enabled in Arduino IDE). The host OS drivers might also not handle these situations well, that is why it is sometimes necessary to try twice.

Can't tell you any more details. Resetting and entering the download mode manually is a bulletproof solution (of course this doesn't mean it is the right one). 

I will close this issue, feel free to reopen if you feel like further discussion is needed.

closed 2021-11-23 by radimkarnis