espressif/esptool #697: Wemos D1 mini -  Invalid head of packet, Possible serial noise or corruption.

Wi5eJ4rl, 2021-11-27:
# Operating system

Debian GNU/Linux bookworm/sid

# Python version

3.9.9

# What Chip

ESP8266

# What development board or other hardware is the chip attached to

LOLIN(WEMOS) D1 mini (clone)

# Is anything else attached to the development board, except for the serial flasher connections?

none

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

command line & Arduino IDE version 1.8.16

# Full esptool.py command line that was run:

esptool.py --trace -p /dev/ttyUSB1 --baud 9600 chip_id

# Full output from esptool.py

esptool.py v3.2
Serial port /dev/ttyUSB1
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 64 bytes:
    200000c0c0010802 00070712200000c0 |  ........... ...
    c001080200070712 200000c0c0010802 | ........ .......
    200000c0c0010802 00070712200000c0 |  ........... ...
    c001080200070712 200000c0c0010802 | ........ .......
TRACE +0.001 Read invalid data:
    200000c0c0010802 00070712200000c0 |  ........... ...
    c001080200070712 200000c0c0010802 | ........ .......
    200000c0c0010802 00070712200000c0 |  ........... ...
    c001080200070712 200000c0c0010802 | ........ .......
TRACE +0.001 Remaining data in serial buffer:
    200000c0c0010802 00070712200000c0 |  ........... ...
    c001080200070712 200000c0c0010802 | ........ .......
    200000c0c0010802 00070712200000c0 |  ........... ...
    c001080200070712 200000c0c0010802 | ........ .......
    200000c0c0010802 00070712200000c0 |  ........... ...
    c001080200070712 200000c0c0010802 | ........ .......
    200000c0c0010802 00070712200000c0 |  ........... ...
    c001080200070712 200000c0c0010802 | ........ .......
.TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 64 bytes:
    200000c0c0010802 00070712200000c0 |  ........... ...
    c001080200070712 200000c0c0010802 | ........ .......
[... 659 lines cut ...]
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 2 bytes: 0000
TRACE +0.000 Read invalid data: 0000
TRACE +0.000 Remaining data in serial buffer:
.

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0x00): Possible serial noise or corruption.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting

# Do you have any other information from investigating this?

I bought 4 wemos and only one have this issue

# Is there any other information you can think of which will help us reproduce this problem?

Sorry I'm newby, but I thing that I created *.bin in IDE by "compliled program export", and i've put it with: /esptool.py -p /dev/ttyUSB1 write_flash --flash_mode=dio --flash_size=4MB 0x00000 dht22.ino.d1_mini.bin. I thing that maybe I've erase some bootloader, but I'm not sure because I don't know which board it was...

The board is bliking, but there is problem to connect via usb. I don't know if I bought it dead or I've "burned" it...

radimkarnis, 2021-11-29:
Hello @Wi5eJ4rl,

if you have 4 identical boards and only 1 has his issue, it is a sign of faulty hardware. 

Can you try connecting with a serial terminal program and see what the board is printing? There are many serial terminal programs suitable for debugging & serial interaction. The pySerial module (which is required for esptool) includes one such command line terminal program - miniterm.py. Run `python -m serial.tools.miniterm "PORT" 74880`, you should see a [boot log like this](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#boot-log) when you reset the chip. If not, please post the output here.

Wi5eJ4rl, 2021-11-29:
Thank you @radimkarnis for your reply.

I'm not sure but the output is similar but longer:
` python3 -m serial.tools.miniterm /dev/ttyUSB0 74880
--- Miniterm on /dev/ttyUSB0  74880,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
-�-�-�-�-�-�-�-�-�-�-�-�-�-�-�-�-�␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀16
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
tail 8
chksum 0xe5
load 016
[... 292 lines cut ...]
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile
SDK ver: 2.0.0(5a875ba) compile␀`
... and a lot of '␀␀␀' some char...

If I understand, maybe I've erased a bootloader? I've read that in ESP-12 it is located in 0x0000 so if I put there sketch, it erases bootloader. I also find information that, if I want to flash new bootloader, the board can switch to boot mode by it self - mine not - and there is the 'force' method with soldering GPIO 0 with GND - this also do nothing... So I don't know what can I do...

The other option is, the chip CH340 is fine so it "talking" with computer, or maybe the this chip is failure and the ESP12 board is fine.... I don't know...

I'm wandering if I can fix this board or it's a junk...

radimkarnis, 2021-11-30:
Thank you for the log. 
The lines of the log look ok to me, the chip is definitely booting (it is weird these lines are repeated that many times).

> ... and a lot of '␀␀␀' some char...

Anyway, baud rate 74880 is what the Espressif firmware bootloader uses. Hence, at that rate, you can decipher its messages but see everything else as gibberish. The apps on top of the Espressif SDK (e.g. Arduino sketch) talk at 115200. That's why you see the following messages as gibberish at that rate. 

If you retry the miniterm command with 115200 baud rate, maybe we get more information.

> method with soldering GPIO 0 with GND - this also do nothing

If you disconnect everything from the board, bridge GPIO0 (marked `D3` on the board) and GND, and reset the chip it doesn't switch into the download mode? (more information about this process is [here](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html))

Wi5eJ4rl, 2021-11-30:
Hello again,

With `python3 -m serial.tools.miniterm /dev/ttyUSB1 115200` command I get:

1. couple of "ERROR" lines
2. Rubbish - I think that this is in 74880 baud rate
3. ready line in loop

But, if I set the 74880 baud rate, plug board in usb, connect D3 to GND, reset, I get "boot mode:(1,7)" in loop, so I don't understand. If I do the same and I put `esptool.py -p /dev/ttyUSB1 chip_id` I get again nothing:

esptool.py v3.2
Serial port /dev/ttyUSB1
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0x00): Possible serial noise or corruption.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting

I tried also connect it with windows but it the same problem. What I can do?

Wi5eJ4rl, 2021-11-30:
What is more, if I have blank board (without any bridge in pins), I have tree (or more) "states" after reset:

#############################################################################################
First:
#############################################################################################

␀chksum 0xe5
load 0x3ffe8000, lechksum 0xe5
load 0x3ffe8000, lechksum 0xe5
(in loop)
load 0x3ffe8000, le␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀v␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀m 0xd8

2nd boot version : 1.6m 0xd8

2nd boot version : 1.6m 0xd8

(in loop)

2nd boot version : 1.6␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀m␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀
jump to run user1 @ 1000

jump to run user1 @ 1000

(in loop)

jump to run user1 @ 1000

␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀d @ Aug  9 2016 15:12:27
phy ved @ Aug  9 2016 15:12:27
phy ved @ Aug  9 2016 15:12:27
(in loop)

#############################################################################################
And second after reset:
#############################################################################################

phy ve␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀boot mode:(3,7)

boot mode:(3,7)

boot mode:(3,7)

(in loop)

boot mode:(3,7)

␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀l 8
chksum 0x84
load 0x3ffe831l 8
(in loop)
load 0x3ffe831l 8
chksum 0x84
load 0x3ffe831␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀: 40MHz
: 40MHzode       : QIO
: 40MHzode       : QIO
(in loop)
: 40MHzode       : QIO
␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀(inloop)
␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀]␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀rf cal sector: 1017
rf[112] : 0rf cal sector: 1017
rf[112] : 0rf cal sector: 1017
(in loop)
rf[112] : 0rf cal sector: 1017
rf[112] : 0␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀r: 1055, pp ver: 10.2

r: 1055, pp ver: 10.2

r: 1055, pp ver: 10.2

(in loop)
[... 4 lines cut ...]

␀␀␀␀␀␀␀␀␀␀␀␀load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room load 0x40100000, len 2408, room ␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀0, len 632, room 0
tail 8
chk0, len 632, room 0
(in loop)
tail 8
chk␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀
  SPI Flash Size & Map: 32Mbit(
  SPI Flash Size & Map: 32Mbit(
(in loop)
  SPI Flash Size & Map: 32Mbit(
[... cut ...]
␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀��j� ��j� 

#############################################################################################

and after reset one of above... Maybe there is a problem with memory? Some "bad sectors"? This is not stable board...

radimkarnis, 2021-12-01:
> if I set the 74880 baud rate, plug board in usb, connect D3 to GND, reset, I get "boot mode:(1,7)" in loop

`boot mode:(1,X)` means switching to download mode works (see [this page for explanation](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#explanation)), but the looping message and other flaky logs are very unusual.

This is some weird behavior, I am sorry, but at this point it is clearly not an esptool issue, but a HW one. I would probably advise you to save the hassle of digging through this and just replace the board.

Wi5eJ4rl, 2021-12-01:
Ok, thanks for your help!

closed 2021-12-01 by Wi5eJ4rl