espressif/esptool #698: esptool chip detection appears broken in latest esp-idf release/v4.3 for ESP32

mostthingsweb, 2021-11-30:
### Operating System

Ubuntu 20.04

### Version

v3.2

### Python Version

3.8.10

### Chip Description

ESP32-WROOM32

### Device Description

SparkFun Thing Plus (from here: https://www.sparkfun.com/products/15663)

### Hardware Configuration

Nothing else is attached to the board.

### How is Esptool Run

No IDE

### Full Esptool Command Line that Was Run

esptool.py -p /dev/ttyUSB0 chip_id

### Esptool Output

```shell
esptool.py v3.2-dev
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.056 Read 1 bytes: 65
TRACE +0.000 Read invalid data: 65
TRACE +0.000 Remaining data in serial buffer: 
    7473204a756e2020 3820323031362030 | ts Jun  8 2016 0
    303a32323a35370d 0a0d0a7273743a30 | 0:22:57....rst:0
    7831302028525443 5744545f5254435f | x10 (RTCWDT_RTC_
    5245534554292c62 6f6f743a30783133 | RESET),boot:0x13
[... 661 lines cut ...]
.

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0x1B)
```

### More Information

Everything was working fine until I updated esp-idf today. I am on the release/v4.3 branch. The last time I updated it was about a month ago (still on the release/v4.3 branch).

Interestingly, when I switch to release/v4.4, esptool.py and idf.py work fine. I get this kind of output:

```
Executing action: flash
Serial port /dev/ttyUSB0
Connecting.....................
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.........
Detecting chip type... ESP32
```

I think chip detection in esp-idf release/v4.3 is broken.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

mostthingsweb, 2021-11-30:
For reference this is what happens when I run the same esptool command using esp-idf release/v4.4 branch:

```
esptool.py v3.2-dev
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 No serial data received.
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 No serial data received.
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.055 Read 1 bytes: 65
TRACE +0.000 Read invalid data: 65
TRACE +0.000 Remaining data in serial buffer: 
    7473204a756e2020 3820323031362030 | ts Jun  8 2016 0
    303a32323a35370d 0a0d0a7273743a30 | 0:22:57....rst:0
    7831302028525443 5744545f5254435f | x10 (RTCWDT_RTC_
    5245534554292c62 6f6f743a30783133 | RESET),boot:0x13
    20285350495f4641 53545f464c415348 |  (SPI_FAST_FLASH
    5f424f4f54290d0a 636f6e6669677369 | _BOOT)..configsi
    703a20302c205350 4957503a30786565 | p: 0, SPIWP:0xee
    0d0a636c6b5f6472 763a307830302c71 | ..clk_drv:0x00,q
    5f6472763a307830 302c645f6472763a | _drv:0x00,d_drv:
    307830302c637330 5f6472763a307830 | 0x00,cs0_drv:0x0
    302c68645f647276 3a307830302c7770 | 0,hd_drv:0x00,wp
    5f6472763a307830 300d0a6d6f64653a | _drv:0x00..mode:
    44494f2c20636c6f 636b206469763a32 | DIO, clock div:2
    0d0a6c6f61643a30 7833666666303033 | ..load:0x3fff003
    302c6c656e3a3732 30300d0a6c6f6164 | 0,len:7200..load
    3a30783430303738 3030302c6c656e   | :0x40078000,len
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.022 Read 1 bytes: 0a
TRACE +0.000 Read invalid data: 0a
TRACE +0.000 Remaining data in serial buffer: 
    1b5b303b33326d49 202836312920626f | .[0;32mI (61) bo
    6f743a202323204c 6162656c20202020 | ot: ## Label    
    2020202020202020 5573616765202020 |         Usage   
    2020202020202054 797065205354204f |        Type ST O
    6666736574202020 4c656e6774681b5b | ffset   Length.[
    306d0d0a1b5b303b 33326d4920283639 | 0m...[0;32mI (69
    2920626f6f743a20 2030206e76732020 | ) boot:  0 nvs  
    2020202020202020 2020202057694669 |             WiFi
    2064617461202020 2020202020303120 |  data        01 
[... 915 lines cut ...]
TRACE +0.000 Write 18 bytes: 
    c000060800000000 0000000000ace50b | ................
    40c0                              | @.
TRACE +0.004 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 0106040084b28da200000000c0
TRACE +0.000 Received full packet: 0106040084b28da200000000
TRACE +0.003 Read 1 bytes: c0
TRACE +0.000 Read 5 bytes: 4f484149c0
TRACE +0.000 Received full packet: 4f484149
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=08a0f53f
TRACE +0.000 Write 14 bytes: c0000a04000000000008a0f53fc0
TRACE +0.004 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200cff441000000c0
TRACE +0.000 Received full packet: 010a0200cff441000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=04a0f53f
TRACE +0.000 Write 14 bytes: c0000a04000000000004a0f53fc0
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020084b28da20000c0
TRACE +0.000 Received full packet: 010a020084b28da20000
MAC: f4:cf:a2:8d:b2:84
Hard resetting via RTS pin...

```

radimkarnis, 2021-11-30:
Hello @mostthingsweb,

Thank you for noticing. ESP-IDF v4.3 uses an older version of esptool, which doesn't have this commit: https://github.com/espressif/esptool/commit/b082b0ed2d86b3330134c4854a021dfd14c29b08. There was a bug that prevented ESP32 revisions 0 and 1 from automatically resetting to the download mode. The fix hasn't been backported to v4.3 yet. Will take care of this!

mostthingsweb, 2021-11-30:
Hi @radimkarnis, great thank you! I thought I had fried a board or something :).

radimkarnis, 2021-12-28:
@mostthingsweb, backported to v4.3 in https://github.com/espressif/esp-idf/commit/4f19ef06fb2e1406d14edcfe984577fdfdda9073.
Thanks for reporting this!

closed 2021-12-28 by radimkarnis