espressif/esptool #699: Add warning/confirmation when uploading binary to an encrypted or secure boot enabled target to avoid bricking devices unknowingly

m-sja, 2021-11-30:
### Is your feature request related to a problem?

I bricked a device by uploading an alternative code to an esp32.
I was not aware that the target has secureboot and flash encryption enabled.
Uploading code using esptool worked flawlesly, without a warning or confirmation. Even verification showed no issue.
This bricked my device as i do not have the pre-encrypted or signed bootloader to restore the device.

ESP32 is widely used by DIYers and Hobyists who try to improve products lacking functionality or firmware updates.
There are a lot of programming interfaces utilising esptool to upload binarys blindly.
Due to this "gamification" users may be not aware that they can brick a device if the efuses are set.

### Describe the solution you'd like

when uploading code to a target which has some security efuses like secureboot or flashencryption set, present a warning the user must confirm or an command option to force flashing to a secured device.
This would prevent the "normal" flashing procedure to brick the device unknowingly if secure options are set on the target.

### Describe alternatives you've considered

Integrating a pre flash check using espfuse on the frontend side would be an option, but due to the fact that there are so many flashing-frontends and the secure options are not common on the DIY/maker universe there is no other practical solution i see.

### Additional context

Boot loop after flashing unencrypted binary to encrypted target:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```
espfuse summary of the encrypted target:
```
>espefuse.exe -p com7 -b 115200 -c esp32 summary
Connecting...
Device PID identification is only supported on COM and /dev/ serial ports.
...........
espefuse.py v3.2
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):                        BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                                 Voltage reference calibration                      = 1100 R/W (0b10000)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                           Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                             If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                            If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                               8MHz clock freq override                           = 50 R/W (0x32)
SPI_PAD_CONFIG_CLK (BLOCK0):                       Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):                         Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):                         Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):                        Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):                       Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):                        Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                                   Efuse write disable mask                           = 388 R/W (0x0184)
RD_DIS (BLOCK0):                                   Efuse read disable mask                            = 3 R/W (0x3)
CODING_SCHEME (BLOCK0):                            Efuse variable block length scheme
  = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                               Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                                      Factory MAC Address
  = 40:f5:20:81:b4:04 (CRC 0xe0 OK) R/W
MAC_CRC (BLOCK0):                                  CRC8 for factory MAC address                       = 224 R/W (0xe0)
CHIP_VER_REV1 (BLOCK0):                            Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                            Silicon Revision 2                                 = False R/W (0b0)
CHIP_VERSION (BLOCK0):                             Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                             Chip package identifier                            = 1 R/W (0b001)
MAC_VERSION (BLOCK3):                              Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 1 R/- (0b0000001)
UART_DOWNLOAD_DIS (BLOCK0):                        Disable UART download mode (ESP32 rev3 only)       = False R/- (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):                       Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):                    Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                               Secure boot V1 is enabled for bootloader image     = True R/W (0b1)
ABS_DONE_1 (BLOCK0):                               Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                             Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):                       Disable flash encryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_DECRYPT (BLOCK0):                       Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):                         Disable flash cache in UART bootloader             = True R/W (0b1)
BLOCK1 (BLOCK1):                                   Flash encryption key
  = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK2 (BLOCK2):                                   Secure boot key
  = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK3 (BLOCK3):                                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

m-sja, 2021-12-14:
Hello @radimkarnis,
thank you for you reply.

Sure you are absolutely right that if doing things never been reported successfully done before, some extra caution should be taken.
But manufacturers suddenly enabling security features in a new product revision could brick a ton of devices.
Also a lot of IoT manufactures do not provide security updates (or any updates at all) making the use case for 3rd party firmware even bigger.
Maybe you can look at it from a sustainability point of view.
I thought about the threats you pointed out and i am confident that my request may not interfere with production environments due to some reasons:

1. Reprogramming already encrypted ESPs in production environment should be uncommon (except while bugfixing messed up production runs which should not be an everyday situation)
2. If i understand correct the secureboot and flash_encryption fuses are set **after** the flash is programmed, so my requested check will not require any input or parameter as in this state the ESP is not encrypted or secured yet. Also no warning message would appear.
3. The "never change a running system" fact: It is highly unlikely that manufacturers will update their production environment without the need for new features of esptool. Even if new software versions are used/updated they are being double, triple and quadruple- checked by people who are responsible for this production step before using it in production.

As esptool also is used by Arduino and PlatformIO environments (lowering the barrier to reflash devices) the positive impact would be even bigger once implemented in those environments.
Also it would prevent users from frustration spending hours trying to fix this unfixable problem. The boot-message does not clearly state that secureboot or flash_encryption is preventing from successfully booting. So searching for the errormessage leads to a ton of possible problems and solutions which all are obsolete in this case.
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```
You will not find something like this: "Congratulations, you successfully bricked your device, now grab your reflow soldering kit and replace the ESP ;-)"

I hope you can take my reasons into account and think about my request again.
Best regards
Marian

devZer0, 2021-12-20:
> On the other hand, it would not be convenient for production scripts, and at this point, adding this confirmation 
>step would be a breaking change - all the custom scripts used by manufacturers would have to be changed.

you could simply check for encryption and refuse to re-flash without further confirmation.

i don't think that in production deployments devices getting re-flashed all to often and simply adding some "-y" for some "i know what i'm doing, please disable security checks and skip confirmation dialogue" should be reasonable change , even in a production script.

furthermore, i don't think production setups won't be updated all too often.  

i would also rate security/device-anti-brick-countermeasure far more important then comfort/compatibility.  

if you can update your esptool.py, you can also update your calling scripts. that's a non brainer.

radimkarnis, 2022-01-31:
Thank you all for your valuable opinions and the discussion. 

We have thought about this again and reconsidered. This will be implemented as a part of a future major release, as this changes the public API.

closed 2022-04-06 by espressif-bot