espressif/esptool #700: esptool.py does not work in Mac hosted virtual machines

MMI, 2021-12-01:
### Operating System

macos 11.6.1

### Version

v3.2

### Python Version

Python 3.10

### Chip Description

ESP32-D0WD-V3

### Device Description

Failure is visible on a devkitC (using a cp210x usb bridge) or other custom devices using an ftdi bridge. 

### How is Esptool Run

terminal

### Full Esptool Command Line that Was Run

esptool -p $PORT read_mac

### Esptool Output

```shell
esptool.py -p /dev/ttyUSB0 read_mac
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: Wrong boot mode detected (0x17)! The chip needs to be in download mode.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
```

### More Information

Device is connected to an iMac.
It works correctly running directly on the Mac.

The failure occurs on virtual machines. I have tried:

Under Parallels:
* ubuntu 20.4
* ubuntu 20.10
* ubuntu with custom kernel and some ftdi_sio driver hacking by me
* Windows 10

Under VirtualBox:
* debian 11

All fail.  From the trace, I can see that the device gets reset but never in the correct bootloader mode.

### Other Steps to Reproduce

I have also tried on a Windows 11 PC. There it works under windows _and_ under virtualbox (!!!)

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2021-12-06:
Hello @MMI,

thank you for reporting this! 

>  From the trace, I can see that the device gets reset but never in the correct bootloader mode.

This is true, the timing of [DTR/RST control line signals needed for automatic bootloader](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#automatic-bootloader) probably deteriorates or they do not propagate at all. There are some things you can try:

1)  [Manually put your ESP32 board into Firmware Download mode.](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader)
2) [Connect over a networked remote serial port to bypass the VM->OS DTR/RST bottleneck.](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/remote-serial-ports.html)
3) Try connecting the chip to a USB hub (using an older USB 2 hub might help).

We are trying hard to support all of the possible OS/Python/ESP chip/USB-UART bridge combinations, but it is simply not possible to cover all of these, especially when a layer of running esptool in a VM is introduced. Esptool is a serial flasher utility - if for some reason the serial connection cannot be established or DTR/RST signals do not propagate, it cannot function properly. For this reason, I believe this issue should be addressed elsewhere than in esptool.

closed 2021-12-17 by radimkarnis

MMI, 2021-12-17:
Thanks for your reply. I actually didn't know about the remote serial port hack -- pretty cool.

For my part, I already had various work arounds and moved on. My motivations for reporting here were:

1. Something for the next person to google.
2. A naive hope that somebody more important than me would file a bug with Apple about their USB virtualization (which is my best guess as to root cause).