espressif/esptool #701: SerialTimeoutException on Linux while flashing NodeMCU firmware

theGeekyLad, 2021-12-03:
### Operating System

Arch Linux

### Version

v3.2

### Python Version

3.9.7

### Chip Description

ESEP8266

### Device Description

## NodeMCU

[image]

[image]

### Hardware Configuration

Nothing else is attached.

### How is Esptool Run

No IDE, Linux Terminal

### Full Esptool Command Line that Was Run

[~ esptool-v3.2-linux-amd64]$ ./esptool --port /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 --baud 9600 write_flash -fm qio 0x00000 ~/Downloads/nodemcu-release-8-modules-2021-12-03-06-35-26-float.bin

### Esptool Output

```shell
esptool.py v3.2
Serial port /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 48:3f:da:49:81:50
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0006ffff...
Flash params set to 0x0020
Compressed 458752 bytes to 299449...
Writing at 0x00000000... (5 %)Traceback (most recent call last):
  File "esptool.py", line 5143, in <module>
  File "esptool.py", line 5136, in _main
  File "esptool.py", line 4602, in main
  File "esptool.py", line 3873, in write_flash
  File "esptool.py", line 154, in inner
  File "esptool.py", line 918, in flash_defl_block
  File "esptool.py", line 495, in check_command
  File "esptool.py", line 458, in command
  File "esptool.py", line 421, in write
  File "serial/serialposix.py", line 636, in write
serial.serialutil.SerialTimeoutException: Write timeout
[299409] Failed to execute script 'esptool' due to unhandled exception!
```
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2021-12-03:
Hello @theGeekyLad,

This can be the result of a low-quality USB cable or a generally noisy serial connection. Does the error occur always at the same spot (5%)?

To get more information, would you mind installing the python version of esptool (not the prebuilt binary) and trying with that? Instructions are [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/contributing.html#development-setup). Also. running the command with `--trace` can help us debug this.

theGeekyLad, 2021-12-03:
Hi @radimkarnis, thanks for the prompt assistance! The exception always occurs at 5%.

I'll also tried changing the cable but to no avail. Shall try the Python version and post an update here. 👍🏼

radimkarnis, 2021-12-10:
@theGeekyLad any progress on this?

If you have time to debug some more, it would be also worth it to run the command with the `--no-stub` option.

moefear85, 2021-12-10:
Did you try other chips? If the error always occurs at the same spot/percentage, then there is the possibility the flash is bad. If it is due to noise, then the errors should happen at random times.

radimkarnis, 2021-12-17:
Closing this issue due to inactivity, feel free to reopen if further discussion is needed.

closed 2021-12-17 by radimkarnis