espressif/esptool #702: Using esptool to flash boards using a gui generated in pycharm and pyqt5

anurag16doshi, 2021-12-07:
### Operating System

Windows 10

### Version

v3.2

### Python Version

Python 3.9.9

### Chip Description

ESP32

### Device Description

Using esptool to flash boards. Accessing the esptool from a gui generated in pycharm and pyqt5. I have connected the clicked event of a pushButton to flash a few files to esp32. The command has previously worked without pycharm and pyqt5.

### How is Esptool Run

GUI-Pyqt5, Pycharm

### Full Esptool Command Line that Was Run

esptool -p COM11 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0xd000 ota_data_initial.bin 0x1000 bootloader.bin 0x10000 v1.00.004.bin 0x8000 partitions_two_ota.bin

### Esptool Output

```shell
Traceback (most recent call last):
  File "...\esptool.py", line 485, in _connect_attempt
    self.sync()
  File "...\esptool.py", line 423, in sync
    self.command(self.ESP_SYNC, b'\x07\x07\x12\x20' + 32 * b'\x55',
  File "...\esptool.py", line 373, in command
    p = self.read()
  File "...\esptool.py", line 318, in read
    return next(self._slip_reader)
  File "...\esptool.py", line 2195, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "...\esptool.py", line 505, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "...\esptool.py", line 491, in _connect_attempt
    print('.', end='')
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "...\venv\Scripts\esptool.exe\__main__.py", line 7, in <module>
  File "...\esptool.py", line 3416, in _main
    main()
  File "...\esptool.py", line 3081, in main
    esp.connect(args.before, args.connect_attempts)
  File "...\esptool.py", line 512, in connect
    print('Connecting')
OSError: [Errno 22] Invalid argument
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2021-12-07:
Hello @anurag16doshi,

> The command has previously worked without pycharm and pyqt5.

Am I understanding correctly that esptool works if you use it standalone? If you don't know, could you please verify?

This looks like a problem introduced by using the custom GUI made with `GUI-Pyqt5`. If that would be the case, I am afraid I cannot help you as this is not an issue with esptool itself.

Please install esptool in [standard mode](https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html) or in [development mode](https://docs.espressif.com/projects/esptool/en/latest/esp32/contributing.html#development-setup) and provide more information about running the command without gui.

anurag16doshi, 2021-12-07:
Hello @radimkarnis 

Thank you for your response, I have been using python's os.system command, it works with the gui and executes the command.

radimkarnis, 2021-12-08:
> it works with the gui and executes the command.

So the issue has been resolved? If not, can you please try using esptool standalone?

radimkarnis, 2021-12-17:
Closing this issue, feel free to reopen if necessary.

closed 2021-12-17 by radimkarnis