espressif/esptool #703: serial.serialutil.SerialException: Cannot configure port

pw2017, 2021-12-07:
### Operating System

Windows 10

### Version

v3.1

### Python Version

Python 3.8.7

### Chip Description

ESP32-WROOM-32

### Device Description

I am following the Espressif getting started document (hello_world) with the DEVKITV1 board.

### Hardware Configuration

Nothing attached

### How is Esptool Run

No IDE, Windows terminal

### Full Esptool Command Line that Was Run

idf.py -p COM3 flash

### Esptool Output

```shell
C:\Users\patrick\esp\hello_world>idf.py -p COM3 flash
Executing action: flash
Running ninja in directory c:\users\patrick\esp\hello_world\build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\Users\patrick\esp-idf\component...trick/esp-idf/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32 -p COM3 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
esptool.py v3.1-dev
Serial port COM3
Traceback (most recent call last):
  File "C:/Users/patrick/esp-idf/components/esptool_py/esptool/esptool.py", line 4535, in <module>
    _main()
  File "C:/Users/patrick/esp-idf/components/esptool_py/esptool/esptool.py", line 4528, in _main
    main()
  File "C:/Users/patrick/esp-idf/components/esptool_py/esptool/esptool.py", line 4027, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "C:/Users/patrick/esp-idf/components/esptool_py/esptool/esptool.py", line 120, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "C:/Users/patrick/esp-idf/components/esptool_py/esptool/esptool.py", line 310, in __init__
    self._port = serial.serial_for_url(port)
  File "C:\Users\patrick\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "C:\Users\patrick\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages\serial\serialwin32.py", line 80, in open
    self._reconfigure_port()
  File "C:\Users\patrick\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
CMake Error at run_serial_tool.cmake:50 (message):
  C:/Users/patrick/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe
  C:/Users/patrick/esp-idf/components/esptool_py/esptool/esptool.py --chip
  esp32 failed

FAILED: CMakeFiles/flash
cmd.exe /C "cd /D C:\Users\patrick\esp-idf\components\esptool_py && C:\Users\patrick\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="C:/Users/patrick/esp-idf" -D SERIAL_TOOL="C:/Users/patrick/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe C:/Users/patrick/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="C:/Users/patrick/esp/hello_world/build" -P C:/Users/patrick/esp-idf/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### More Information

I also tried with a different board - same error.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2021-12-07:
Hello @pw2017,

please check if `COM3` is the right serial port and if you have the [USB-UART bridge drivers](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers) installed. Then reboot and try again.

radimkarnis, 2021-12-08:
Thank you for trying. 

Can you run esptool on its own (without using IDF) or connect to the chip with any [serial terminal program](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#serial-terminal-programs) (e.g. run `python -m serial.tools.miniterm PORT 115200`)?

If this fails, there might be several reasons why the connection cannot be established, such as a low-quality USB cable, faulty board (USB-UART bridge chip), drivers. Nevertheless, without being able to connect to the chip it is impossible for esptool (or any other serial program, really) to function. I recommend reading [this guide in IDF documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/establish-serial-connection.html) for more tips and information.

It might be worth it to try on another device to confirm or disprove any cable/board-related problems.

pw2017, 2021-12-08:
Thanks for your reply.
I used a borrowed pc and was able to connect and flash firmware using Arduino IDE.
Back to my pc and I am able to connect to the board successfully using Putty and can monitor the port.
I also tried using Arduino IDE and get the same message when trying to flash but I can open the monitor and view the serial output.
So it seems the connection to the port is working but for some reason IDF cannot access the port to flash.

radimkarnis, 2021-12-08:
Great! It means your HW is not faulty and esptool and the drivers work. 

One last thing to try is to [install esptool separately](https://docs.espressif.com/projects/esptool/en/latest/esp32/contributing.html#development-setup) and run the command `esptool.py esp32 -p COM3 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin` (Taken from the log provided by you. Please note the file paths in the command are relative.). If that succeeds, I would suggest opening an issue in the IDF repository - there may be an issue with the IDF build system accessing the port.

pw2017, 2021-12-10:
Thanks @radimkarnis. I did the istall using the Windows installation but running the command give the same error. But I'm not sure now if you mean that I should install esptool a different way? I was a bit lost with the link you sent.

moefear85, 2021-12-10:
The error messages clearly mention "PermissionError", which means pyserial (the library that python uses to communicate with serial ports), does not have required permissions to access the port. On linux this would usually mean you have to add the user to the dialout group, or permissions have to be set in "/etc/udev/rules..." for that device. But since you're on windows, I assume you might be trying to concurrently access the serial port while something else (such as a serial monitor of another application), is already accessing it.

Try to restart your PC, and without opening anything unnecessary, just open a command prompt, and type "esptool.py -p com3 chip_id". Does it work or produce the same error?

pw2017, 2021-12-16:
Thanks @moefear85, what eventually worked was to plug the board in via a powered USB hub (it was plugged in directly before).
I don't know why but that has resolved the issue.

Thanks again @moefear85 and @radimkarnis

closed 2021-12-16 by pw2017