espressif/esptool #705: Failed to communicate with the flash chip, read/write operations will fail

photomatix18, 2021-12-12:
### Operating System

Windows 11

### Version

v3.2

### Python Version

Python 3.10.1

### Chip Description

ESP32-D0WDQ6 (revision 1)

### Device Description

ESP32 ESP-32S

Original pruchase link - https://www.amazon.com/gp/product/B08BYJR9Y4/

### Hardware Configuration

Nothing.

### How is Esptool Run

Windows terminal

### Full Esptool Command Line that Was Run

python -m esptool chip_id

### Esptool Output

```shell
esptool.py v3.2
Found 1 serial ports
Serial port COM10
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b8:f0:09:cd:f2:c4
Uploading stub...
Running stub...
Stub running...
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: b8:f0:09:cd:f2:c4
Hard resetting via RTS pin...
```

### More Information

I've used Tozmotizer, ESPHome-Flasher (3 different versions), and esptools to no avail. I can't get anything to flash. I have two identical boards and neither works so I don't suspect faulty hardware. I have tried 3 different cables, 3 computers and different ports on the comptuers. 

One of the boards has never been flashed, the other I flashed about a year ago and now trying to flash with new firmware. I don't remember haven't any issues when I first flashed about a year ago.

### Other Steps to Reproduce

This is what I get in esphome-flasher 1.4.0.

```Using 'COM10' as serial port.
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......
Detecting chip type... ESP32
Connecting.......

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: B8:F0:09:CD:F2:C4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Unexpected error: Error while erasing flash: Packet content transfer stopped (received 8 bytes)
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2021-12-13:
Hello @photomatix18, 

Thank you for trying different solutions before opening an issue.

> I've used Tozmotizer, ESPHome-Flasher (3 different versions), and esptools to no avail. ... I have tried 3 different cables, 3 computers and different ports on the computers.

This is a clear indication that the problem are faulty devkits - specifically their flash chips. According to your logs, communication with the ESP works fine until the point when a flash chip read/write operation chip happens.

> WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.

This warning is tied directly to faulty hardware. Try running `python -m esptool flash_id`, this command asks the ESP to interrogate the flash chip and report its manufacturer and device number. If this fails, only `0000`s or `ffff`s are reported back. 

For some reason, the flash chips are not answering back - they might be dead, browning out ([more info](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#insufficient-power)), or have some of their pins shorted. There are several bad reviews of the board on the amazon page you've linked. These describe cold solder joints and other problems.

If you have the devkit connected to a breadboard or other hardware, please try disconnecting that first. Make sure you are using a sufficient power supply and try checking your devkit closely for any visual signs of a problem.

photomatix18, 2021-12-13:
Hi @radimkarnis,

Thanks for the response! 

I suspected faulty hardware, but after my known (at one time) good unit stopped working, and I opened my second brand new unit and it had the same issue, I had a hard time believing both were bad.

I just reflowed both ESP modules on the dev boards and finally got a response from one of the flash chips. 🙌🏼

I'll keep working on the other one. Thanks for pointing me in the right direction!

closed 2021-12-13 by photomatix18

MCR21-arch, 2024-07-04:
@photomatix18 How did you reflow the ESP module ?

photomatix18, 2024-07-04:
I just used my soldering iron and heated up each contact on the chip, one at a time.