espressif/esptool #707: windows 11 not detecting

settlelol, 2021-12-18:
### Operating System

Windows 11

### Version

v3.2

### Python Version

Python 3.9.7

### Chip Description

ESP8226

### Device Description

[aliexpress link](https://www.aliexpress.com/item/1005001952641270.html)

### How is Esptool Run

Latest version of n2d [https://github.com/a4004/n2d]

### Other Steps to Reproduce

I have the Silabs driver and I have installed the CH340 driver although it doesnt appear on the device manager port section for some reason

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2021-12-18:
Hi @settlelol. Please try with the drivers posted in https://github.com/espressif/esptool/issues/706.

settlelol, 2021-12-18:
https://i.imgur.com/WCRnGZR.png

i get this when trying to install it

settlelol, 2021-12-18:
> Hi @settlelol. Please try with the drivers posted in #706.

https://i.imgur.com/WCRnGZR.png

i get this when trying to install it

radimkarnis, 2021-12-30:
Closing this issue. Feel free to reopen if you think further discussion is needed.

closed 2021-12-30 by radimkarnis