espressif/esptool #708: ESP32-S2 boot fails / SPIFFs doesn't work

GiovanniCmpaner, 2021-12-20:
### Operating System

Windows 11

### Version

v3.1

### Python Version

Python 3.9.8

### Chip Description

ESP32-S2

### Device Description

Genuine ESP32-S2-Saola-1M

### Hardware Configuration

Nothing

### How is Esptool Run

Windows terminal

### More Information

Related issues: https://github.com/espressif/arduino-esp32/issues/6039

Three command variations, with resulting outputs:

1. Boot works, SPIFFS doesn't works (default from PlatformIO):
```
"C:\Users\Giovanni\.platformio\penv\Scripts\python.exe" "C:\Users\Giovanni\.platformio\packages\tool-esptoolpy\esptool.py" 
--chip esp32s2 
--port "COM5" 
--baud 921600 
--before default_reset 
--after hard_reset write_flash -z 
--flash_mode dio 
--flash_freq 40m 
--flash_size detect 
0x1000 C:\Users\Giovanni\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32s2\bin\bootloader_dio_40m.bin 
0x8000 C:\Users\Giovanni\Desktop\auto2teste\.pio\build\esp32dev\partitions.bin 
0xe000 C:\Users\Giovanni\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin 
0x10000 .pio\build\esp32dev\firmware.bin
```
ESP32-S2 output:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x4e4
load:0x4004c000,len:0x82c
load:0x40050000,len:0x277c
entry 0x4004c174
E (544) SPIFFS: mount failed, -10025
[   960][E][esp32-hal-misc.c:125] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT
E (3236) SPIFFS: mount failed, -10025
[  3652][E][SPIFFS.cpp:114] format(): Formatting SPIFFS failed! Error: -1
[  3652][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
[...]
```

2. Boot doesn't work (with `flash_mode = qio` and `board_build.f_flash = 80000000L` on PlatformIO):
```
"C:\Users\Giovanni\.platformio\penv\Scripts\python.exe" "C:\Users\Giovanni\.platformio\packages\tool-esptoolpy\esptool.py" 
--chip esp32s2 
--port "COM5" 
--baud 921600 
--before default_reset 
--after hard_reset write_flash -z 
--flash_mode qio 
--flash_freq 80m 
--flash_size detect 
0x1000 C:\Users\Giovanni\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32s2\bin\bootloader_qio_80m.bin 
0x8000 C:\Users\Giovanni\Desktop\auto2teste\.pio\build\esp32dev\partitions.bin 
0xe000 C:\Users\Giovanni\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin 
0x10000 .pio\build\esp32dev\firmware.bin
```
ESP32-S2 output:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:0x3ffe6100,len:0x570
ets_loader.c 65 
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40010d75
SPIWP:0xee
mode:QIO, clock div:1
load:0x3ffe6100,len:0x570
ets_loader.c 65
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:0x3ffe6100,len:0x570
ets_loader.c 65
[...]
```

3. Boot works, SPIFFS works (manual modified command):
```
"C:\Users\Giovanni\.platformio\penv\Scripts\python.exe" "C:\Users\Giovanni\.platformio\packages\tool-esptoolpy\esptool.py" 
--chip esp32s2 
--port "COM5" 
--baud 921600 
--before default_reset 
--after hard_reset write_flash -z 
--flash_size detect 
0x1000 C:\Users\Giovanni\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32s2\bin\bootloader_qio_80m.bin 
0x8000 C:\Users\Giovanni\Desktop\auto2teste\.pio\build\esp32dev\partitions.bin 
0xe000 C:\Users\Giovanni\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin 
0x10000 .pio\build\esp32dev\firmware.bin
```
ESP32-S2 output:
```
No error messages
```

Why the default 1 or variation 2 on PlatformIO doesn't work?

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

GiovanniCmpaner, 2021-12-21:
It is a bootloader issue:
https://github.com/espressif/arduino-esp32/issues/6039#issuecomment-998798677

closed 2021-12-21 by GiovanniCmpaner

radimkarnis, 2021-12-21:
Thanks for letting us know!