espressif/esptool #709: Can't flash / erase - Failed to read register address 3ff00054 (result was 0100: Unknown result)

iKrushYou, 2021-12-21:
### Operating System

macOS 12.0

### Version

v3.2

### Python Version

2.7.16

### Chip Description

ESP8266 on TYWE3S

### Device Description

[Treatlife Smart Switch](https://www.amazon.com/Treatlife-Smart-Light-Switch-Assistant/dp/B07R4MFWCQ/ref=sr_1_1_sspa?crid=IGV6YAD3W80S&keywords=treat+life+switch&qid=1640056974&sprefix=treat+life+switch%2Caps%2C64&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExUkw3NEE5SFgzSFo0JmVuY3J5cHRlZElkPUEwMjM1Mjk5MVdFSEtMMDIyRE0wViZlbmNyeXB0ZWRBZElkPUEwNTk1NTQ0WERFN01ETlNFSkIzJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==) uses a TYWE3S

### Hardware Configuration

Connected TXD/RXD/GND/VCC according to https://developer.tuya.com/en/docs/iot/wifie3smodule?id=K9605ua1cx9tv
and use GPIO0 to enable boot mode

### How is Esptool Run

terminal

### Full Esptool Command Line that Was Run

esptool.py -p /dev/cu.usbserial-AB0KPFGB -b 115200 write_flash -e 0x0 /Users/alexkrush/Downloads/DS02S.bin 

### Esptool Output

```shell
alexkrush@Alexs-MacBook-Pro ~ % esptool.py -p /dev/cu.usbserial-AB0KPFGB -b 115200 -t write_flash -e 0x0 /Users/alexkrush/Downloads/DS02S.bin 
esptool.py v3.2
Serial port /dev/cu.usbserial-AB0KPFGB
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.007 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.011 Read 1 bytes: c0
TRACE +0.000 Read 61 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c001080200 070712200000c0c0 | ........... ....
    0108020007071220 0000c0c001       | ....... .....
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.014 Read 1 bytes: 08
TRACE +0.000 Read 33 bytes: 
    0200070712200000 c0c0010802000707 | ..... ..........
    12200000c0c00108 0200070712200000 | . ........... ..
    c0                                | .
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000

Detecting chip type...TRACE +0.000 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.000 Write 10 bytes: c00014000000000000c0
TRACE +0.007 Read 1 bytes: c0
[... 101 lines cut ...]
TRACE +0.015 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020001c1f0ff0000c0
TRACE +0.000 Received full packet: 010a020001c1f0ff0000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5c00f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +0.016 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020032f42c000000c0
TRACE +0.000 Received full packet: 010a020032f42c000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5800f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005800f03fc0
TRACE +0.015 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020000b000b30000c0
TRACE +0.000 Received full packet: 010a020000b000b30000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5400f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005400f03fc0
TRACE +0.016 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a02003e3800020100c0
TRACE +0.000 Received full packet: 010a02003e3800020100

A fatal error occurred: Failed to read register address 3ff00054 (result was 0100: Unknown result)
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2021-12-22:
Hello @iKrushYou,

I've never seen this issue! It seems like the esp returns an error when esptool tries to read some of its eFuses (see [this line](https://github.com/espressif/esptool/blob/master/esptool.py#L1298), notice the address). This might indicate some eFuse corruption in the chip. Sadly, there is not any helpful ESP8266 eFuses documentation available to verify this.

The eFuses are read in order to identify the chip and package. You can bypass this and try flashing without this step by modifying the `get_chip_description` function in `esptool.py` ([this one](https://github.com/espressif/esptool/blob/master/esptool.py#L1321-L1332)) to leave out the identification steps:
```python
    def get_chip_description(self):
        return "ESP8266EX"
```

I am afraid solving an eFuse issue is impossible, so jumping over that step might be the only option. Please try this and let me know if that helped!

iKrushYou, 2021-12-22:
> 

Hey @radimkarnis thanks for the super quick reply! I had tried running with the `--chip ESP8266` flag, not sure if that's the same. I can try this as well

however, I bought another flasher and it arrived a few minutes ago. It seems to be working! I have no knowledge of how these works but it seems odd to me that the old one was able to read certain things from the chip but not everything

radimkarnis, 2021-12-22:
> I had tried running with the --chip ESP8266 flag

Not the same thing - a package identification is still attempted, even if the chip type is specified. You can try my recommendation just to verify.

> I bought another flasher and it arrived a few minutes ago. It seems to be working! 

Great news! A USB-Serial bridge (flasher) issue is also a possibility (a better one than corrupted eFuses). This is also supported by the fact that you had this issue on multiple chips. 

Thanks for investigating more!

radimkarnis, 2021-12-30:
Closing this issue. If you plan to debug the faulty flasher, feel free to share your observations with us!

closed 2021-12-30 by radimkarnis