espressif/esptool #711: esptool building fails under the latest Python 3.10 and Linux 5.15

maximtrp, 2021-12-28:
### Operating System

Linux 5.15.11

### Version

v3.2

### Python Version

Python 3.10.1

### Full Esptool Command Line that Was Run

pip install esptool

### Esptool Output

```shell
Defaulting to user installation because normal site-packages is not writeable
Collecting esptool
  Using cached esptool-3.2-py3-none-any.whl
Collecting bitstring>=3.1.6
  Using cached bitstring-3.1.9-py3-none-any.whl (38 kB)
Requirement already satisfied: cryptography>=2.1.4 in /usr/lib/python3.10/site-packages (from esptool) (36.0.1)
Collecting reedsolo<=1.5.4,>=1.5.3
  Using cached reedsolo-1.5.4.tar.gz (271 kB)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3.10/site-packages (from esptool) (3.5)
Collecting ecdsa>=0.16.0
  Using cached ecdsa-0.17.0-py2.py3-none-any.whl (119 kB)
Requirement already satisfied: cffi>=1.12 in /usr/lib/python3.10/site-packages (from cryptography>=2.1.4->esptool) (1.15.0)
Requirement already satisfied: pycparser in /usr/lib/python3.10/site-packages (from cffi>=1.12->cryptography>=2.1.4->esptool) (2.21)
Requirement already satisfied: six>=1.9.0 in /usr/lib/python3.10/site-packages (from ecdsa>=0.16.0->esptool) (1.16.0)
Building wheels for collected packages: reedsolo
  Building wheel for reedsolo (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-sw_ip2_e/reedsolo_fece4f6a45c042bf9331b09563b190d1/setup.py'"'"'; __file__='"'"'/tmp/pip-install-sw_ip2_e/reedsolo_fece4f6a45c042bf9331b09563b190d1/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-4n9mkxr8
       cwd: /tmp/pip-install-sw_ip2_e/reedsolo_fece4f6a45c042bf9331b09563b190d1/
  Complete output (161 lines):
  Cython is installed, building creedsolo module
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.10
  copying reedsolo.py -> build/lib.linux-x86_64-3.10
  running build_ext
  building 'creedsolo' extension
  creating build/temp.linux-x86_64-3.10
  gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fPIC -I/usr/include/python3.10 -c creedsolo.c -o build/temp.linux-x86_64-3.10/creedsolo.o
  creedsolo.c: In function ‘__pyx_tp_dealloc_array’:
  creedsolo.c:32027:5: error: lvalue required as increment operand
  32027 |     ++Py_REFCNT(o);
        |     ^~
  creedsolo.c:32029:5: error: lvalue required as decrement operand
  32029 |     --Py_REFCNT(o);
        |     ^~
  creedsolo.c: In function ‘__pyx_tp_dealloc_memoryview’:
  creedsolo.c:32338:5: error: lvalue required as increment operand
  32338 |     ++Py_REFCNT(o);
        |     ^~
  creedsolo.c:32340:5: error: lvalue required as decrement operand
  32340 |     --Py_REFCNT(o);
        |     ^~
  creedsolo.c: In function ‘__pyx_tp_dealloc__memoryviewslice’:
  creedsolo.c:32588:5: error: lvalue required as increment operand
  32588 |     ++Py_REFCNT(o);
        |     ^~
  creedsolo.c:32590:5: error: lvalue required as decrement operand
  32590 |     --Py_REFCNT(o);
[... 283 lines cut ...]
          |                         ^
    In file included from /usr/include/python3.10/unicodeobject.h:1046,
                     from /usr/include/python3.10/Python.h:83,
                     from creedsolo.c:16:
    /usr/include/python3.10/cpython/unicodeobject.h:446:26: note: declared here
      446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
          |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
    creedsolo.c: In function ‘__Pyx_decode_c_string’:
    creedsolo.c:37955:9: warning: ‘PyUnicode_FromUnicode’ is deprecated [-Wdeprecated-declarations]
    37955 |         return PyUnicode_FromUnicode(NULL, 0);
          |         ^~~~~~
    In file included from /usr/include/python3.10/unicodeobject.h:1046,
                     from /usr/include/python3.10/Python.h:83,
                     from creedsolo.c:16:
    /usr/include/python3.10/cpython/unicodeobject.h:551:42: note: declared here
      551 | Py_DEPRECATED(3.3) PyAPI_FUNC(PyObject*) PyUnicode_FromUnicode(
          |                                          ^~~~~~~~~~~~~~~~~~~~~
    error: command '/usr/bin/gcc' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-sw_ip2_e/reedsolo_fece4f6a45c042bf9331b09563b190d1/setup.py'"'"'; __file__='"'"'/tmp/pip-install-sw_ip2_e/reedsolo_fece4f6a45c042bf9331b09563b190d1/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-5ap1p51c/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/maxim/.local/include/python3.10/reedsolo Check the logs for full command output.
```

### What is the Expected Behaviour?

Successful installation

dobairoland, 2021-12-28:
Hi @maximtrp. This seems to be an issue with the `reedsolo` package. The issue was already reported: https://github.com/tomerfiliba/reedsolomon/issues/36

I think a suitable workaround for you would be to use our prebuilt wheel database:

```sh
pip install --only-binary :all: --extra-index-url https://dl.espressif.com/pypi/ reedsolo
pip install esptool
```

Feel free to use any other online wheel database where a suitable wheel package is available.

dobairoland, 2022-01-10:
Thanks for the feedback.

I'll keep this open until the proper fix which would be to set a dependency on that `reedsolo` version which solves the build issue.

closed 2022-12-12 by espressif-bot

kdschlosser, 2022-12-27:
all that has to be done is Cython has to be uninstalled. Then it will install without an issue.