espressif/esptool #716: RTS and DTR being used as active low is not properly documented outside of source code

FrozenDroid, 2022-01-21:
### Operating System

N/A

### Version

v3.2

### Python Version

N/A

### Full Esptool Command Line that Was Run

N/A

### More Information

We are designing a board using the ESP32C3, and we ran into the issue with "Download mode successfully detected, but getting no sync reply: The serial TX path seems to be down.".
We wrote our own STM32 <-> ESP32 UART proxy, and wired the DTR and RTS high signals to BOOTOPTION and ENABLE respectively. After looking into the source code of Esptool, we found out that these signals need to actually be inverted, since Esptool treats DTR and RTS as active low, which was not obvious to us, and was not documented anywhere in the docs.

radimkarnis, 2022-01-21:
Hello @FrozenDroid,
thanks for the report, we will mention this in the docs to prevent any future confusion.

> Download mode successfully detected, but getting no sync reply: The serial TX path seems to be down.".

This implies the chip successfully switches into the download mode, but further communication from the host to the chip doesn't work. Was this fully solved by inverting the logic?

FrozenDroid, 2022-01-21:
@radimkarnis It was indeed. By some miracle it did make its way into download mode, but as far as we could tell, as soon as it got into download mode, the enable pin dropped back to low, and thus obviously the ESP wouldn't reply.

cbaurtx, 2022-01-21:
The documentation on the boot sequence is generally poor. If deciphered from the source code correctly the flashing sequence for GPIO0 and EN are
DTR       RTS      
H           H     
Wait
L         H
Wait
L         L
H         L
Wait
H         H

But than I read that GPIO0 must be 0 during reset to get into flashing mode, Where am i wrong?
Some boards have an XOR circuit added, but if the sequence above is correct that will not help

radimkarnis, 2022-01-21:
@cbaurtx, you have switched the `DTR` and `RTS` signals. Keep in mind that both are active low - for instance `_setRTS(True)` sets `RTS` low and therefore pulls `EN` low. To be clear - `True` means `L` and `False` means `H`.

| RTS (EN) | DTR (IO0) | Note |
|-----|-----|-----|
| H | H | |
| wait | wait | |
| L | H | - EN pin pulled low, chip in reset |
| wait | wait | |
| L | L | - EN and IO0 pulled low |
| H | L | - EN high while IO0 low, chip out of reset - download mode |
| wait | wait | |
| H | H | - EN high and IO0 high |

It is really the same thing you do when you press the buttons manually (release `RST` button while holding `IO0 / BOOT` button), just written in code. The only caveat is the fact that `True` means pull low and vice versa - which is the reason for this issue ticket.

closed 2022-01-25 by espressif-bot

radimkarnis, 2022-01-25:
Understandable, thanks for sharing your opinion.

We have to keep in mind the fact that this is still **esptool** documentation. It is not meant to serve as a guide for developing custom projects. In such cases, it is preferred to read the Technical Reference Manuals (see [C3 TRM](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf)) and the serial port library (in our case [pySerial](https://pyserial.readthedocs.io/en/latest/pyserial_api.html)) docs.

Nevertheless, the note `When developing esptool.py, keep in mind DTR and RTS are active low signals` doesn't prevent you from keeping that in mind even when you are not doing any development work for esptool!