espressif/esptool #717: ESP32-S3 SPI Efuse Config

TGiles1998, 2022-01-22:
### Operating System

macOS 12.1

### Version

latest master

### Python Version

Python 3.10.0

### Chip Description

ESP32S3

### Device Description

DevKitC

### Hardware Configuration

Nothing 

### How is Esptool Run

Mac OS Terminal

### Full Esptool Command Line that Was Run

espefuse.py --port /dev/cu.usbserial-1440 burn_efuse SPI_PAD_CONFIG_HD 28

### Esptool Output

```shell
espefuse.py --port /dev/cu.usbserial-1440 burn_efuse SPI_PAD_CONFIG_HD 28
Connecting....
Detecting chip type... ESP32-S3
espefuse.py v3.3-dev
The efuses to burn:
  from BLOCK1
     - SPI_PAD_CONFIG_HD
    ATTENTION! This BLOCK uses NOT the NONE coding scheme and after 'BURN', these efuses can not be burned in the feature:
               ['MAC', 'SPI_PAD_CONFIG_CLK', 'SPI_PAD_CONFIG_Q', 'SPI_PAD_CONFIG_D', 'SPI_PAD_CONFIG_CS']
               ['SPI_PAD_CONFIG_WP', 'SPI_PAD_CONFIG_DQS', 'SPI_PAD_CONFIG_D4', 'SPI_PAD_CONFIG_D5', 'SPI_PAD_CONFIG_D6']
               ['SPI_PAD_CONFIG_D7', 'WAFER_VERSION', 'PKG_VERSION', 'BLOCK1_VERSION']

Burning efuses (see 'ATTENTION!' above):

    - 'SPI_PAD_CONFIG_HD' (SPI HD (D3) pad) 0b000000 -> 0b011100

Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
[01] MAC_SPI_8M_0         is not empty
	(written ): 0x868accb48498fe000100000000000000000034b472700cac
	(to write): 0x00000000000000000000000000001c000000000000000000
	(coding scheme = RS)

A fatal error occurred: 	Burn into MAC_SPI_8M_0 is forbidden (RS coding scheme does not allow this).
```

### More Information

I would like to change the efuse config for the esp32s3 spi pins

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

KonstantinKondrashov, 2022-01-25:
Hi @TGiles1998!
> A fatal error occurred: 	Burn into MAC_SPI_8M_0 is forbidden (RS coding scheme does not allow this).

The error is correct. Your chip already has written eFuse block#1. You can not burn new data into this block.

Blocks 1 - 10 have RS coding scheme which does not support the second burn. https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/efuse.html#supported-coding-scheme

For TEST ONLY, I can suggest trying to suppress this error but you will get the error of RS coding. The hidden part of the eFuse block (RS coding data) will be damaged but actual data in the block will be well. I believe this should not affect anything.

Some commands have `--force-write-always` flag to suppress this error, but not `efuse_burn`.

TGiles1998, 2022-01-25:
Thank you for your response! I hope espressif can manufacture some silicon that isn't already burnt.

closed 2022-01-29 by TGiles1998