espressif/esptool #719: Failed to connect to Espressif device: No serial data received

suchencjusz, 2022-02-08:
### Operating System

Ubuntu 20.04

### Version

v3.2

### Python Version

Python 3.8.10

### Chip Description

ESP8266 ESP-12N CP2102

### Device Description

NodeMCU V1 ESP8266 ESP-12N CP2102

### Hardware Configuration

SCL SDA FROM SSD1306 to D1 and D2, RX From gps to D3 TX from gps to D4, (sd card reader) SCLK to D5, DO to D6, DI to D7, CS to D8

### How is Esptool Run

CLI

### Full Esptool Command Line that Was Run

`esptool.py -p /dev/ttyUSB0 chip_id`

### Esptool Output

```shell
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
```

### More Information

Problem was discovered when I tried upload some code on platformio.

### Other Steps to Reproduce

I have no way  to check it on other OS

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-02-08:
Hello @suchencjusz,

this error could be caused by many different reasons. Most of them are covered in the [troubleshooting guide](https://docs.espressif.com/projects/esptool/en/latest/esp8266/troubleshooting.html). Most likely there is something wrong with the connection or the chip is not switching into the download mode. Sadly, your issue report doesn't really tell if you have tried any of the recommended steps.

Have you tried opening a serial connection with any of [serial terminal programs](https://docs.espressif.com/projects/esptool/en/latest/esp8266/troubleshooting.html#serial-terminal-programs)? If you see any data coming in, we can rule out bad RX/TX connection (bad USB cable, insufficient power, ..).

Have you tried disconnecting other HW from the devkit and [manually switching into the download mode](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#manual-bootloader)?

If this doesn't help, please provide a trace ([run esptool with the `--trace` option](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#manual-bootloader)).

radimkarnis, 2022-02-08:
For instance, the D3 pin of NodeMCU V1 connects to the GPIO0 of ESP8266. This strapping pin decides the mode on reset - download mode is entered if pulled low. I see you are connecting RX from GPS to this pin.

suchencjusz, 2022-02-08:
I don't believe... it worked!!! Thanks! <3

closed 2022-02-08 by suchencjusz

imsaku, 2022-10-02:
> For instance, the D3 pin of NodeMCU V1 connects to the GPIO0 of ESP8266. This strapping pin decides the mode on reset - download mode is entered if pulled low. I see you are connecting RX from GPS to this pin.

I'm sorry, but what do you exactly mean "pulled low"? I'm on ESP8266 and have an identical issue, and all these troubleshoot guides have been telling me this, but I don't understand what it means. Thanks for any help beforehand.