espressif/esptool #721: ESP32-S2FN4R2 support

YujiTFD, 2022-03-04:
### Operating System

Windows 10 x64

### Version

v3.2

### Python Version

Windows executable

### Chip Description

ESP32-S2FN4R2

### Device Description

Wemos [S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html)

### Hardware Configuration

USB-C cable connected to the board

### How is Esptool Run

No IDE, Windows terminal

### Full Esptool Command Line that Was Run

esptool.exe -p COM3 -b 115200 read_flash 0x00000 0x400000 flash_4MB.bin

### Esptool Output

```shell
GPIO0 is not pressed
===
esptool.py v3.2
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0x55): Possible serial noise or corruption.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
===

GPIO0 pressed before boot
===
esptool.py v3.2                                                                                                                                                                
Serial port COM4                                                                                                                                                               
Connecting...                                                                                                                                                                  
Detecting chip type... Unsupported detection protocol, switching and trying again...                                                                                           
Connecting...                                                                                                                                                                  
Traceback (most recent call last):                                                                                                                                             
  File "esptool.py", line 377, in detect_chip                                                                                                                                  
struct.error: unpack requires a buffer of 16 bytes                                                                                                                             
                                                                                                                                                                               
During handling of the above exception, another exception occurred:                                                                                                            
                                                                                                                                                                               
Traceback (most recent call last):                                                                                                                                             
  File "esptool.py", line 5143, in <module>                                                                                                                                    
  File "esptool.py", line 5136, in _main                                                                                                                                       
  File "esptool.py", line 4541, in main                                                                                                                                        
  File "esptool.py", line 121, in get_default_connected_device                                                                                                                 
  File "esptool.py", line 388, in detect_chip                                                                                                                                  
  File "esptool.py", line 663, in connect                                                                                                                                      
  File "esptool.py", line 628, in _connect_attempt                                                                                                                             
  File "esptool.py", line 519, in sync                                                                                                                                         
  File "esptool.py", line 451, in command                                                                                                                                      
  File "serial\serialutil.py", line 372, in timeout                                                                                                                            
  File "serial\serialwin32.py", line 222, in _reconfigure_port serial.serialutil.SerialException: Cannot 
 configure port, something went wrong. Original message: PermissionError(13, 'Присоединенное к системе устройство не работает.', None, 31)                                                                                                                                                                           
[21188] Failed to execute script 'esptool' due to unhandled exception!
===
```

### More Information

- I've read troubleshhoting guide thrice,
- tried several USB cables connected to USB 2.0/3.0/C controllers,
- VDC voltage measured with voltmeter returned 3.311V,
- tried pressing GPIO0 button before booting (COM-port number changes this way) and using **--before default_reset** parameter, all with no luck.

At the same time, very same cabling and same **esptool.exe** binary I was able to read and flash Wemos [D1 mini](https://www.wemos.cc/en/latest/d1/d1_mini.html) like a charm. [WLED web-install](https://install.wled.me) also worked fine for ESP8266EX, returning error for S2 again.

### Other Steps to Reproduce

Also tried reading and flashing with esptool.py 3.2 from Raspbian with very same fail.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2022-03-04:
Hi @YujiTFD.

The D1 mini uses a UART-to-USB bridge while the S2 mini board uses the directly available USB interface of ESP32-S2. This is causing the difference.

Until there is no USB console application uploaded, you need to enter bootloader mode manually. This is explained in the intitial upload section of https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/usb-otg-console.html#initial-upload.

Have you tried not just holding GPIO0 but also manually reseting to bootloader mode, and only after that invoke esptool?

Do you always get PermissionError (like even after restarting your PC)? This is usually happening when two applications are accessing the same port.

dobairoland, 2022-03-04:
You could find some tips in https://github.com/espressif/esptool/issues?q=is%3Aissue+permissionerror+is%3Aclosed as well.

radimkarnis, 2022-03-04:
Hello @YujiTFD,
I would recommend using latest esptool from this repository. 

This issue has been fixed in https://github.com/espressif/esptool/commit/b6546510bf6c1fe8ad8c208bfff432ba2e76dd0e. This fix is not yet included in v3.2. 

Please, let us know if this helps!

YujiTFD, 2022-03-04:
> Hello @YujiTFD, I would recommend using latest esptool from this repository.
> 
> A similar issue has been discussed in #685 and fixed in [b654651](https://github.com/espressif/esptool/commit/b6546510bf6c1fe8ad8c208bfff432ba2e76dd0e). This fix is not yet included in v3.2.
> 
> Please, let us know if this helps!

Yep, that did it:
```
esptool.py v3.3-dev
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:d8:00:06
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0013dfff...
Compressed 1301488 bytes to 765686...
Wrote 1301488 bytes (765686 compressed) at 0x00000000 in 14.4 seconds (effective 724.1 kbit/s)...
Hash of data verified.

Leaving...
WARNING: ESP32-S2FNR2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
```

A-a-a-nd now I reliased I got a wrong devboard: was planning on [WLED](https://github.com/Aircoookie/WLED) install or some ESPHome project, but no firmware goes live after flashing :D Time to invest in studying.

Thank you kindly for your assistance.

closed 2022-03-04 by YujiTFD

dobairoland, 2022-03-04:
Thank you @radimkarnis. I forgot about that fix.