espressif/esptool #722: Error connecting - LILYGO TTGO T-Display ESP32 module

GuicoRM, 2022-03-08:
### Operating System

Windows 10

### Version

v3.0

### Python Version

No using Python

### Chip Description

LILYGO TTGO T-Display ESP32 

### Device Description

LILYGO TTGO T-Display ESP32 

### Hardware Configuration

No

### How is Esptool Run

IDE Arduino

### More Information

esptool.py v3.0-dev
Serial port COM3
Connecting......
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:11:ae:a6:94:8c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-03-08:
Hi @GuicoRM,
the line `Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB` implies that the flash chip is not responding. It might be disconnected (bad soldering - check the devkit closely), read-protected or damaged.

To verify this, please install latest esptool from this repository or from PyPI ([guide here](https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html)) and try running `esptool.py flash_id`. This command interrogates the flash chip, esptool should return an error if this really is the case.

radimkarnis, 2022-03-14:
Hello @GuicoRM,
have you tried my recommendation? 

Please provide feedback, so we can solve this issue or close it, thank you!

GuicoRM, 2022-03-17:
Unfortunately I couldn't try it due to I changed my ESP32 chips and the problem disappeared.

radimkarnis, 2022-03-17:
Ok, closing the issue.

closed 2022-03-17 by radimkarnis