espressif/esptool #724: firmware connection problem

Standardzero1, 2022-03-15:
### Operating System

Window 10

### Version

v3.2

### Python Version

Python 2.7.17

### Chip Description

ESP8266 -12F

### Device Description

 NUCLEO-F303RE

### How is Esptool Run

无IDE

### Full Esptool Command Line that Was Run

esptool.py --port COM3 --baud 230400 write_flash --flash_size=detect  0  E:\esptool_0.4.11_win32\websocket_mini.ino.generic.bin

### Esptool Output

```shell
E:\esptool_0.4.11_win32>esptool.py --port COM3 --baud 230400 write_flash --flash_size=detect  0  E:\esptool_0.4.11_win32\websocket_mini.ino.generic.bin
esptool.py v3.2
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0xFC): Possible serial noise or corruption.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
```

### More Information

When downloading the firmware, I tested the pins of the ESP8266.GPIO0 is at 0V.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-03-15:
Hello @Standardzero1,
it seems like the chip is resetting into the download mode. To verify this, please capture the serial interaction with any of [serial terminal programs](https://docs.espressif.com/projects/esptool/en/latest/esp8266/troubleshooting.html#serial-terminal-programs) (connect and try resetting the chip). You should see a [boot log like this.](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#boot-log) 

If you see such a boot log, it means the connection works fine and the problem is in the reset sequence. 
> When downloading the firmware, I tested the pins of the ESP8266.GPIO0 is at 0V.

If GPIO0 is correctly held low, the only thing left is to reset the chip, maybe that isn't happening. You can try [entering the download mode manually](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#manual-bootloader).

Please let me know how this went and provide any captured logs.

Standardzero1, 2022-03-15:
Hello, I entered the download mode manually and connected GPIO0 to ground by pressing the button.

radimkarnis, 2022-03-15:
Did you see the boot log? Did entering the download mode manually solve the issue?

Standardzero1, 2022-03-15:
How can I see the startup log? Manual download did not solve the problem.

radimkarnis, 2022-03-15:
run `python -m serial.tools.miniterm PORT 74880` (replace PORT with your serial port) and try resetting the chip. 

Please read the links I have attached to my previous replies, these lead to esptool documentation and explain every step.

Standardzero1, 2022-03-15:
!U0�010E1B0E04720BB71ED0
!E0000
!R010000000000L��XwcQLCA4<02/+)*)13*),(;:63-**(((()())*)'((()()))())(+*))((')))))))(*+,)()+'+)+45)0/>;DhkcqYYM=\lc^OJI1<7L?[U*2);:CMWRu||\__\DNET38,(-1)**--(.-*-.)(*-0:-/(+.+/,.3=;-(),-.=4))*)*))**,()))*+)*(,1*)()*+)-/-+(*:42.65..:7.13)-,))/,(/0:8,,-8,CMD=UF>FE.).-)+/)*
!C010000000000YVUbf``_\YWUTUTQMKGDBA@>=;;:;;<<<<<<<;;;:999999999999988999999999999::::;<<>ADFIKMNOQSUUTTRRSUYZ[\[[[\^^^]\[\_abcdba^^[XVTRRQTTUUVVVUUROKIFECB@?==<=====>>>>>>>???>==>>>>>==<=<=>=<;;<<<;;:::::::::;<<<====>????>>>>>?>>>>?@AAAAABCCDEEEFGGHGEDDBBCDCCDEFGFFDCJH
!T0�0006CwF6DB0000102A4;77CD0000209B9qE58E000030A5BlCD87000040CE4|FF56000051155=A37200006122D=01D6000071942M0FE0000081993U56D90000919E4F02950000A0000"00000000B0000"00000000C0000"00000000D0000"00000000E0000"00000000F0000"00000000

I can see that the serial port has data. Because I flashed the firmware of the microcontroller through the serial port before.

Standardzero1, 2022-03-15:
I leave GPIO0 floating, enter run mode, my computer can connect to WIFI. (I disconnected the USB cable at this point)

Standardzero1, 2022-03-17:
Yes。Before brushing the firmware of STM32 through the serial port。

radimkarnis, 2022-03-17:
Esptool is a tool made exclusively for ESP8266 and ESP32 series of Espressif SoC's. It won't work with STM32 chips. 

If you are trying to flash a separate ESP8266 module (to use together with the STM32 for its networking capabilities), you will need a separate TTL adapter.

Or maybe I just don't understand your situation. In that case, you would need to provide more information, maybe pictures, about what and how you are connecting and trying to flash.

radimkarnis, 2022-03-25:
Closing this issue. Feel free to reopen if you think we should discuss more.

closed 2022-03-25 by radimkarnis