espressif/esptool #725: espfuse not available

S3j5b0, 2022-03-16:
### Operating System

ubuntu 20.04

### Version

v3.2

### Python Version

python 3.8

### Full Esptool Command Line that Was Run

$ esptool.py esptool.py v3.2 usage: esptool [-h]                [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2,esp8684}]                [--port PORT] [--baud BAUD]                [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]                [--after {hard_reset,soft_reset,no_reset,no_reset_stub}]                [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]                [--connect-attempts CONNECT_ATTEMPTS]                {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}                ...  esptool.py v3.2 - ESP8266 ROM Bootloader Utility  positional arguments:   {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}                         Run esptool {command} -h for additional help     load_ram            Download an image to RAM and execute     dump_mem            Dump arbitrary memory to disk     read_mem            Read arbitrary memory location     write_mem           Read-modify-write to arbitrary memory location     write_flash         Write a binary blob to flash     run                 Run application code in flash     image_info          Dump headers from an application image     make_image          Create an application image from binary files     elf2image           Create an application image from ELF file     read_mac            Read MAC address from OTP ROM     chip_id             Read Chip ID from OTP ROM     flash_id            Read SPI flash manufacturer and device ID     read_flash_status   Read SPI flash status register     write_flash_status  Write SPI flash status register     read_flash          Read SPI flash content     verify_flash        Verify a binary blob against flash     erase_flash         Perform Chip Erase on SPI flash     erase_region        Erase a region of the flash     merge_bin           Merge multiple raw binary files into a single file for                         later flashing     version             Print esptool version     get_security_info   Get some security-related data  optional arguments:   -h, --help            show this help message and exit   --chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2,esp8684}, -c {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2,esp8684}                         Target chip type   --port PORT, -p PORT  Serial port device   --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading   --before {default_reset,usb_reset,no_reset,no_reset_no_sync}                         What to do before connecting to the chip   --after {hard_reset,soft_reset,no_reset,no_reset_stub}, -a {hard_reset,soft_reset,no_reset,no_reset_stub}                         What to do after esptool.py is finished   --no-stub             Disable launching the flasher stub, only talk to ROM                         bootloader. Some features will not be available.   --trace, -t           Enable trace-level output of esptool.py interactions.   --override-vddsdio [{1.8V,1.9V,OFF}]                         Override ESP32 VDDSDIO internal voltage regulator (use                         with care)   --connect-attempts CONNECT_ATTEMPTS                         Number of attempts to connect, negative or 0 for                         infinite. Default: 7. ask@Garsy:~/Git$ espfuse.py espfuse.py: command not found

### Esptool Output

```shell
$ espfuse.py
espfuse.py: command not found
```

### What is the Expected Behaviour?

I intalled esptool like so:

`t$ pip install esptool`
which installed nicely.
I can then run esptool like described in the docs:

```
$ esptool.py
esptool.py v3.2
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2,esp8684}]
```
but not espfuse:

```
$ espfuse.py
espfuse.py: command not found
```
According to the docs, they should be installed alongside each other.What might be the issue here?

radimkarnis, 2022-03-16:
Hi @S3j5b0, 

you are running `espfuse.py`, but the script is named `espefuse.py` - joint words `esp` and `eFuse`.

You are missing one `e`.

S3j5b0, 2022-03-16:
Oh my jesus, thanks a lot

closed 2022-03-16 by S3j5b0