espressif/esptool #726: Use CTS pin instead of RTS

ygoe, 2022-03-20:
### Is your feature request related to a problem?

I have one of the FT232RL USB to serial adapters that you can get on AliExpress, with the jumper to select 3.3/5V. All of them look the same. They have these pins exposed on the short edge:

* DTR
* RX
* TX
* VCC
* CTS
* GND

From the ESP32-Pico-Kit schematic (I'm using that board for now) I see how the serial port's DTR and RTS lines are connected to the ESP32's EN and IO0 lines to support automatic booting while flashing.

When using the FT232RL adapter for other setups with the plain ESP32-WROOM-32D/U module, there's a pin mismatch here. DTR is available, but RTS is not. (I could solder a pin header on the long edge to get it but it's more complicated to attach two different connectors than just using the single one that's already there.) Instead the adapter has the CTS line.

Can the CTS line be used in place of the RTS line for automatic booting?

Could it be an option and configurable for use inside the VS Code extension? Or would it be safe to always set both pins to support all types of USB to serial adapters?

After all, why does the ESP32 normally use the RTS pin when all available adapters only feature the CTS pin instead? That feels a bit like missing the reality.

### Describe the solution you'd like

See above.

### Describe alternatives you've considered

See above.

radimkarnis, 2022-03-21:
Hello @ygoe,

the simplest explanation for this is that the DTR and RTS pins are the only ones you can set as output (control them). These are also called DTE-only signals, RS232 standard says:

> The basis is that a DTE (data terminal equipment) is talking to a DCE (Data Circuit-terminating equipment)!
    typically we'd think of a DTE as the computer, and a DCE as the teletype (the ESP device in this case)

[image]

You can check this in [pySerial API guide](https://pyserial.readthedocs.io/en/latest/pyserial_api.html?highlight=cts#serial.Serial.cts). Notice how the `dtr` and `rts` properties have setters (pySerial allows to set their state), but `cts` has only a getter (you can only read its state).

ygoe, 2022-03-21:
I see. So then there's nothing that software could do here.

I've soldered all the pins on one of my USB-serial boards and use those pins then. Next I'll put together the transistors and resistors on the breadboard according to the provided schematic and try to understand how they work here. That's going to be interesting!

closed 2022-03-21 by ygoe