espressif/esptool #727: ESP32-S3 `--flash_size 8MB` does not modify bootpartition.bin

Jason2866, 2022-03-20:
### Operating System

macOS 12.3

### Version

latest master

### Python Version

3.8.12

### Chip Description

ESP32-S3

### Device Description

AI-Thinker ESPS3-12K (NodeMCU)

### Hardware Configuration

nothing

### How is Esptool Run

VSC and directly via command line

### Full Esptool Command Line that Was Run

esptool.py --chip esp32s3 write_flash --flash_size 8MB 0x0 bootloader.bin 0xe000 boot_app0.bin 0x8000 partition-table.bin 0x10000 app.bin

### Esptool Output

```shell
Finishes normally. 
Reading bootloader first bytes back 
`E903022F C0613B40 EE000000 09000000`
so this is still the orig. 4MB bootloader provided from Arduino framework
Correct for 8MB would be
`E903023F C0613B40 EE000000 09000000`
```

### Other Steps to Reproduce

Flashing a 8MB ESP32-S3 device with [this](https://github.com/espressif/arduino-esp32/blob/esp32-s3-support/tools/sdk/esp32s3/bin/bootloader_qio_80m.bin) Arduino bootpartition.bin

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-03-21:
Hello @Jason2866,

esptool is not updating the image header, because it doesn't think it is a bootloader image. This is because you are flashing the bootloader at offset `0x0`, but it should be flashed at `0x1000`.

The check in question is [here](https://github.com/espressif/esptool/blob/master/esptool.py#L3795). You can see it skips updating the image header if its offset is not `0x1000`.

To verify, try flashing the bootloader to `0x1000`, the flash_size byte should be updated.

Jason2866, 2022-03-21:
@radimkarnis The bootloader for S3 is indeed at 0x0 and has to be changed.

closed 2022-03-21 by dobairoland