espressif/esptool #729: Divide by zero exception esptool.py v3.0

owenduffy, 2022-03-23:
### Operating System

Win10

### Version

latest master

### Python Version

3.9

### Chip Description

Esp8266

### Device Description

esp01s

### How is Esptool Run

Windows command window

### Full Esptool Command Line that Was Run

D:\Users\owen\AppData\Local\Programs\Python\Python39\scripts\esptool.exe -c ESP8266 -b 921600 -p com8 read_flash 0xfe000 0x80 wifisettings.bin

### Esptool Output

```shell
D:\src\EspRelay>D:\Users\owen\AppData\Local\Programs\Python\Python39\scripts\esptool.exe -c ESP8266 -b 921600 -p com8 read_flash 0xfe000 0x80 wifisettings.bin
esptool.py v3.3
Serial port com8
Connecting...
Failed to get PID of a device on com8, using standard reset sequence.
.
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a4:cf:12:b9:80:27
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
128 (100 %)
128 (100 %)
Traceback (most recent call last):
  File "D:\Users\owen\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\Users\owen\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\Users\owen\AppData\Local\Programs\Python\Python39\scripts\esptool.exe\__main__.py", line 7, in <module>
  File "D:\Users\owen\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 5380, in _main
    main()
  File "D:\Users\owen\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 4812, in main
    operation_func(esp, args)
  File "D:\Users\owen\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool.py", line 4267, in read_flash
    % (len(data), args.address, t, len(data) / t * 8 / 1000), last_line=True)
ZeroDivisionError: float division by zero
```

### More Information

Divide zero exception... possibly due to calculation of transfer speed for short file at high speed giving a time of zero. Needs try/catch ... and in any event, it does not make sense to calculate transfer speed if the total time is less than say 10 times the minimum time increment.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-03-25:
Hi @owenduffy,
thanks for reporting this. 

We will patch this soon. Or if you feel like it, we welcome contributions!

owenduffy, 2022-03-25:
The simplest fix might be at line 4266 of esptool.py (https://github.com/espressif/esptool/commit/7e3496d1636a0c88c062de54f18d9ed2c28b773e), change:

```
    print_overwrite('Read %d bytes at 0x%x in %.1f seconds (%.1f kbit/s)...'
                    % (len(data), args.address, t, len(data) / t * 8 / 1000), last_line=True)
```

to:

```
    try
        print_overwrite('Read %d bytes at 0x%x in %.1f seconds (%.1f kbit/s)...'
                    % (len(data), args.address, t, len(data) / t * 8 / 1000), last_line=True)
    except
        pass

```

Not tested though.

Owen

closed 2022-04-04 by espressif-bot