espressif/esptool #730: S3 Flash / PSRAM type (QIO / OPI) detection commands

Jason2866, 2022-03-26:
### Is your feature request related to a problem?

Since S3 introduced a new Flash type (OPI) it is mandantory to flash the correct
combination of used Flash / PSRAM type. Without the device does fail to boot.
It is a problem for the user to know which flash mode is used for Flash / PSRAM.
Since there are many different variants already to get, and it is not possible to identify the used chip modes. 

### Describe the solution you'd like

A command which prints out the supported flash modes for S3 for flash and PSRAM

### Describe alternatives you've considered

Guessing and trying all variants until S3 is booting.

radimkarnis, 2022-04-01:
Hello @Jason2866, thank you for the feature request.

We will look into this! Currently, I can't promise this to be done in any specific timeframe, though.

squix78, 2022-06-02:
+1, please add support for opi flash

arendst, 2022-06-02:
+1, please add support for opi flash.

Enhances Tasmota user experience a lot.

egnor, 2022-07-14:
Would it be possible to clarify this bug report? Is it a documentation bug or a functionality bug? What would the better documentation look like? What _are_ the correct flash modes to use? Does "S3" here refer to the ESP32-S3?

Jason2866, 2022-07-14:
@egnor It is not a bug report it is a feature request for the ESP32-S3 for a command which identifies the type of flash/psram and prints this out.

closed 2022-11-25 by espressif-bot

dobairoland, 2022-11-25:
The `flash_id` command has been extended with information about the flash type. We have not added a separate command for this. If needed, the required information can be obtained from the eFuse as it was done in https://github.com/espressif/esptool/commit/b25606b95920bd06df87aff9202c7a15377d4a30.

Implementing the PSRAM part is not possible at the moment. Please see the linked ESP-IDF request.