espressif/esptool #734: d1mini pro ROM bootloader communication problem on Linux host

galuschka, 2022-04-14:
### Operating System

Ubuntu 20.04

### Version

latest master

### Python Version

Python 3.8.10

### Chip Description

ESP8266 (d1mini pro with 10c4:ea60 USB to UART)

### Device Description

I always get timeout, when I try read chip_id, flash_id, read_mac or whatever esptool command.
```
TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 39 bytes: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555c0                    | UUUUUU.
TRACE +0.000 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.101 Serial data stream stopped: Possible serial noise or corruption.
```
write seems good: c0 00 08  -> 00 = request
on read I expect c0 01 08  -> 01 = response, but after c0 immediately 08 seems to be received.

The linux usb driver is cp210x: (unknown version)
Bus 001 Device 002: ID 10c4:ea60 Silicon Labs CP210x UART Bridge

### Hardware Configuration

no GPIO connected

### How is Esptool Run

arduino IDE as well as direct python call and same behavior by several esptool.py versions (2.4.0, 2.8, 3.3)

### Full Esptool Command Line that Was Run

python ~/.espressif/python_env/rtos3.4_py3.8_env/lib/python3.8/site-packages/esptool.py --port /dev/ttyUSB0 --baud 115200 --trace --chip esp8266 chip_id

### Esptool Output

```shell
esptool.py v3.3
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 39 bytes: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555c0                    | UUUUUU.
TRACE +0.000 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.100 Serial data stream stopped: Possible serial noise or corruption.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 39 bytes: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555c0                    | UUUUUU.
TRACE +0.000 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.100 Serial data stream stopped: Possible serial noise or corruption.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 39 bytes: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555c0                    | UUUUUU.
TRACE +0.000 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.100 Serial data stream stopped: Possible serial noise or corruption.
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 39 bytes: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555c0                    | UUUUUU.
TRACE +0.001 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.101 Serial data stream stopped: Possible serial noise or corruption.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 39 bytes: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555c0                    | UUUUUU.
TRACE +0.000 Received full packet: 
    0824070712205555 5555555555555555 | .$... UUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    555555555555                      | UUUUUU
TRACE +0.101 Serial data stream stopped: Possible serial noise or corruption.
.

A fatal error occurred: Failed to connect to ESP8266: Serial data stream stopped: Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

### More Information

It might relate to a bug in the linux USB driver, where bulk transfer with zero length packets fails. At least there was such a bug in a similar USB driver.

However: I would like to somehow program the 2nd stage bootloader plus an initial image (ESP8266_RTOS_SDK with 2 image OTA).

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

galuschka, 2022-04-14:
works fine with d1mini (not pro)
Bus 001 Device 008: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

radimkarnis, 2022-04-14:
Hi @galuschka,

this definitely looks like a driver issue. The synchronization sequence should look [like this](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/serial-protocol.html#:~:text=packet%E2%80%9D%20log%20lines.-,Here,-is%20a%20second). Can you try reinstalling the cp210x driver?

If you need to program the board just once to upload OTA app, you can use the working d1mini with the HL-340 USB-Serial adapter to do so. It is possible to connect the TX and RX lines of both devkits, disable the esp chip of the non-pro d1mini (by pulling the `CH_PD/EN` pin low), [manually](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#manual-bootloader) put the second chip into download mode, and flash. This way, you effectively "replace" the cp210x with an HL-340.

radimkarnis, 2022-05-09:
Closing this. Feel free to reopen if needed.

closed 2022-05-09 by radimkarnis