espressif/esptool #735: Unable to verify flash chip connection (No serial data received.).

danielkucera, 2022-05-01:
### Operating System

Ubuntu

### Version

other

### Python Version

Python 3.9.7 (default, Sep 10 2021, 14:59:43)

### Chip Description

ESP32-C3

### Device Description

plain module on breakout PCB with 3V3 regulator from 5V on USB
[image]

### Hardware Configuration

nothing else, just plain board on breakout PCB

### How is Esptool Run

Linux terminal

### Full Esptool Command Line that Was Run

esptool.py -p /dev/ttyACM0 -t erase_flash

### Esptool Output

```shell
$ esptool.py -p /dev/ttyACM0 -t erase_flash
esptool.py v3.3
Serial port /dev/ttyACM0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 41 bytes: 
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0               | ... .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 56 bytes: 
    c001080400070712 2000000000c0c001 | ........ .......
    0804000707122000 000000c0c0010804 | ...... .........
    0007071220000000 00c0c00108040007 | .... ...........
    07122000000000c0                  | .. .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Read 14 bytes: c0010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000

Detecting chip type...TRACE +0.000 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.000 Write 10 bytes: c00014000000000000c0
TRACE +0.000 Read 1 bytes: c0
[... 550 lines cut ...]
    main()
  File "/home/danman/.local/bin/esptool.py", line 4812, in main
    operation_func(esp, args)
  File "/home/danman/.local/bin/esptool.py", line 4229, in erase_flash
    esp.erase_flash()
  File "/home/danman/.local/bin/esptool.py", line 143, in inner
    return func(*args, **kwargs)
  File "/home/danman/.local/bin/esptool.py", line 974, in erase_flash
    self.check_command("erase flash", self.ESP_ERASE_FLASH,
  File "/home/danman/.local/bin/esptool.py", line 495, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/home/danman/.local/bin/esptool.py", line 468, in command
    p = self.read()
  File "/home/danman/.local/bin/esptool.py", line 413, in read
    return next(self._slip_reader)
StopIteration
```

### More Information

It was working successfully until I flashed one firmware but maybe it is not related?

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

danielkucera, 2022-05-01:
To reproduce, unzip attached file and run following commands (with GPIO9 High):
[app.zip](https://github.com/espressif/esptool/files/8598683/app.zip)
- `esptool.py --chip esp32c3 --port "/dev/ttyACM0" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x0000 bootloader_qio_80m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 failing-usb-upload.bin `
- `esptool.py --chip esp32c3 --port "/dev/ttyACM0" erase_flash`

result:
```
$ esptool.py --chip esp32c3 --port "/dev/ttyACM0" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x0000 bootloader_qio_80m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 failing-usb-upload.bin 
esptool.py v3.3
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 34:b4:72:4f:55:dc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00064fff...
Flash params set to 0x022f
Compressed 18864 bytes to 11758...
Wrote 18864 bytes (11758 compressed) at 0x00000000 in 0.4 seconds (effective 354.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 402.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 526.5 kbit/s)...
Hash of data verified.
Compressed 345472 bytes to 174201...
Wrote 345472 bytes (174201 compressed) at 0x00010000 in 4.5 seconds (effective 608.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
$ esptool.py --chip esp32c3 --port "/dev/ttyACM0" erase_flash
esptool.py v3.3
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 34:b4:72:4f:55:dc
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Traceback (most recent call last):
  File "/home/danman/.local/bin/esptool.py", line 5387, in <module>
    _main()
  File "/home/danman/.local/bin/esptool.py", line 5380, in _main
    main()
  File "/home/danman/.local/bin/esptool.py", line 4812, in main
    operation_func(esp, args)
  File "/home/danman/.local/bin/esptool.py", line 4229, in erase_flash
    esp.erase_flash()
  File "/home/danman/.local/bin/esptool.py", line 143, in inner
    return func(*args, **kwargs)
  File "/home/danman/.local/bin/esptool.py", line 974, in erase_flash
    self.check_command("erase flash", self.ESP_ERASE_FLASH,
  File "/home/danman/.local/bin/esptool.py", line 495, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/home/danman/.local/bin/esptool.py", line 468, in command
    p = self.read()
  File "/home/danman/.local/bin/esptool.py", line 413, in read
    return next(self._slip_reader)
StopIteration
```

After flashing that firmware, esptool no longer works until manually bringing GPIO9 to GND.

radimkarnis, 2022-05-02:
Hello @danielkucera,

This is unusual behavior - the flash happens successfully, and after that the flash chip becomes unavailable.

Is there any possibility your app sets any of the flash chip status register contents? It seems like the flash chip gets write-/erase-protected.

A good way to check this is to run [`read_flash_status` esptool command](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/advanced-commands.html#read-flash-chip-registers-read-flash-status) and compare the non-working chip and a working one

> After flashing that firmware, esptool no longer works until manually bringing GPIO9 to GND.

So let me understand this correctly - [manually entering the download mode](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html#manual-bootloader) solves the problem? According to your logs, the chip is able to enter the download mode automatically. There should be no difference after a manual or automatic reset. 

I will try to reproduce the issue!

danielkucera, 2022-05-02:
Hi Radim,
I don't touch any registers. The only thing I do is `usb_serial_jtag_write_bytes()` in loop() (I was testing the max USB speed).

radimkarnis, 2022-05-30:
Hi @danielkucera,

Sorry for not getting back to you sooner. I tried to reproduce the issue but couldn't do it. I am using the same chip (`ESP32-C3 (revision 3)`), same binaries, same commands but all works as intended. Erasing flash happens successfully. 

This leads me to believe that your problems might be caused by a faulty flash chip. I would recommend to just replace the module.

Sorry I couldn't help more. I will close the issue now, feel free to reopen if you feel like further discussion is needed.

closed 2022-05-30 by radimkarnis