espressif/esptool #736: LittleFS upload failed with LOLIN D1 mini 4.0

thierry7100, 2022-05-06:
### Operating System

Ubuntu 20.04

### Version

v3.0

### Python Version

3.8.10

### Chip Description

ESP8266

### Device Description

LOLIN D1 Mini, v4.0

### Hardware Configuration

Nothing

### How is Esptool Run

IDE Arduino, v1.8.16, ESP8266 packqge v3.02

### Full Esptool Command Line that Was Run

[LittleFS] data    : /home/thierry/ProjetsDivers/HollowClock/HollowClock_NTP/data [LittleFS] size    : 3048 [LittleFS] page    : 256 [LittleFS] block   : 8192 /HollowClock [LittleFS] upload  : /tmp/arduino_build_692164/HollowClock_NTP.mklittlefs.bin [LittleFS] address : 0x100000 [LittleFS] reset   : --before default_reset --after hard_reset [LittleFS] port    : /dev/ttyUSB0 [LittleFS] speed   : 921600 [LittleFS] python   : /home/thierry/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 [LittleFS] uploader : /home/thierry/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/upload.py

### Esptool Output

```shell
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: c4:5b:be:dd:b3:39
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 3121152 bytes to 4474...
Writing at 0x00100000... (100 %)
Wrote 3121152 bytes (4474 compressed) at 0x00100000 in 0.1 seconds (effective 242481.5 kbit/s)...
Traceback (most recent call last):
  File "/home/thierry/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/upload.py", line 66, in <module>
    esptool.main(cmdline)
  File "/home/thierry/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 3604, in main
    operation_func(esp, args)
  File "/home/thierry/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 3011, in write_flash
    res = esp.flash_md5sum(address, uncsize)
  File "/home/thierry/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 113, in inner
    return func(*args, **kwargs)
  File "/home/thierry/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 780, in flash_md5sum
    res = self.check_command('calculate md5sum', self.ESP_SPI_FLASH_MD5, struct.pack('<IIII', addr, size, 0, 0),
  File "/home/thierry/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 413, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/home/thierry/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 386, in command
    p = self.read()
  File "/home/thierry/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 331, in read
    return next(self._slip_reader)
  File "/home/thierry/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 2636, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
LittleFS Upload failed!
```

### More Information

This is working with Lolin D1 Mini v3.0 boards, but not with LOLIN D1 Mini v4.0

### Other Steps to Reproduce

When I launch ESPtool install via apt install esptool (old version 2.8), this works.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-05-10:
Hello @thierry7100,
would you mind trying with the latest esptool version (can be downloaded with pip or pulled from this repository, [installation guide here](https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html))? 

If this doesn't work, I would greatly appreciate if you could help us find the change that introduced the regression. A great tool for doing so is [git bisect](https://git-scm.com/docs/git-bisect).

radimkarnis, 2022-05-23:
Closing this issue for inactivity. Feel free to reopen if needed.

closed 2022-05-23 by radimkarnis