espressif/esptool #737: "Timed out waiting for packet header" on OSX BIG SUE for CH340 + ESP8266

hupiv, 2022-05-07:
### Operating System

mac os big sur

### Version

v3.2

### Python Version

Python 3.9.12

### Chip Description

ESP8266

### Device Description

ESP01

### Hardware Configuration

I'm using the AZDelivery with CH340 chip to upload firmware

### How is Esptool Run

I'm running from command line

### Full Esptool Command Line that Was Run

esptool.py  --baud 74880 --chip esp8266 --trace --port /dev/cu.usbserial-1410 chip_id

### Esptool Output

```shell
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 No serial data received.
.TRACE +0.053 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.104 No serial data received.
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
```

### More Information

I tried to change the baud rate to lower to upper (9600, ... 115 200..) without no effect :-(

I'm sure my ESP8266/ESP01 card is working because when I plug on my USB/UART devices I can see the default wifi of my card working.

I checked for different discussion and I found an error with Linux Kernel (I'm under OSX but perhaps it could be the same ?)

Regards

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

hupiv, 2022-05-07:
the issue seems related to #653

radimkarnis, 2022-05-09:
Hello @hupiv,

Probably there is a problem with the CH340 USB-UART board or with any of the serial lines.

> I'm sure my ESP8266/ESP01 card is working because when I plug on my USB/UART devices I can see the default wifi of my card working.

This means the ESP chip is getting power, but `No serial data received` implies there is no communication between the chip and the host Mac. To verify this, you can try opening the port with any of the [serial terminal programs](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#serial-terminal-programs) and seeing if the [boot log](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#boot-log) appears when you unplug and plug back the ESP8266/ESP01. If it doesn't, the USB-UART board is at fault.

> I checked for different discussion and I found an error with Linux Kernel 

I am pretty sure this is not related to the Linux kernel issue.

hupiv, 2022-05-10:
Thanks for your response.

Yes I can see the boot log :
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- **Miniterm on /dev/cu.usbserial-1410**  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
␛[0;36m[D][wifi:388]: Found networks:␛[0m
␛[0;32m[I][wifi:431]: - 'Livebox-C4F0' ␛[5m(34:49:5B:B7:C4:F0) ␛[6m␛[0;33m▂▄▆␛[0;37m█␛[0m␛[0m
␛[0;36m[D][wifi:433]:     Channel: 6␛[0m
␛[0;36m[D][wifi:434]:     RSSI: -60 dB␛[0m
␛[0;36m[D][wifi:436]: - ␛[5m'chambre'␛[6m ␛[5m(14:59:C0:B1:17:B8) ␛[6m␛[0;33m▂▄▆␛[0;37m█␛[0m␛[

radimkarnis, 2022-05-10:
This means serial communication is working, but the chip is not [entering the download mode](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#automatic-bootloader). Is there some kind of switch on the USB-UART CH340 board? If so, can you switch it into programming mode and try again?

hupiv, 2022-05-10:
I check on the card and no I didn't see any switch :-(

radimkarnis, 2022-05-10:
So I have opened the reference and I'm afraid the board is missing the auto-reset circuitry:

> It also does not use DTR or RTS to control the reset or programming pins, so IDE’s that use this feature will not work and you will need extra switches to program your ESP.

> However, there is something that I really don't understand, who on earth engineers a programming interface, without the actual button needed to program? because that's missing here.
I get that you may want to save £ 0.02 of the push button, but at least find an easier way rather than have to squeeze a think cable between the contacts of the ESP-01 on the PCB.

> Additionally, this is not suitable for programming ESP-01 modules at all. It lacks the ability to bring the required pin low, so it won't ever program the board.

You will need to pull GPIO0 down in order to [manually enter the download mode](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#manual-bootloader). Without this, the board cannot be programmed.

hupiv, 2022-05-10:
Big big thanks for your help! really !
I will buy a better one so :-)

closed 2022-05-10 by hupiv