espressif/esptool #738: fatal error

skdush, 2022-05-14:
### Operating System

windows 10

### Version

v3.0

### Python Version

Python 3.9.7

### Chip Description

ESP8266

### Device Description

HELLO GUYS . PLEASE I  NEED SOME HELP!!
WHEN I RUN A SIMPLE BLINK CODE  I GET THIS ERROR .I'M USING ESP8266

Arduino : 1.8.16 (Windows 10), Carte : "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Serial1, None, Only Sketch, 3000000"

Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 231788          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27629   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 
DATA   : 1496  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 992   ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25608 )         - zeroed variables      (global, static) in RAM/HEAP 

Le croquis utilise 261905 octets (25%) de l'espace de stockage de programmes. Le maximum est de 1044464 octets.
Les variables globales utilisent 28096 octets (34%) de mémoire dynamique, ce qui laisse 53824 octets pour les variables locales. Le maximum est de 81920 octets.
esptool.py v3.0
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
_
  File "C:\Users\Dell\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/upload.py", line 66, in <module>
    esptool.main(cmdline)
  File "C:/Users/Dell/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 3552, in main
    esp.connect(args.before, args.connect_attempts)
  File "C:/Users/Dell/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool\esptool.py", line 529, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

### How is Esptool Run

ARDUINO IDE

### Full Esptool Command Line that Was Run

esptool.py v3.0 Serial port COM3

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-05-16:
Hello @skdush,
the problem here is not your code (it really doesn't matter which app you are trying to upload). For some reason, esptool can't communicate with the chip and fails to connect.

Either there is something wrong with the connection our your chip is not [automatically switching into the download mode](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#automatic-bootloader). Please read the [troubleshooting guide](https://docs.espressif.com/projects/esptool/en/latest/esp8266/troubleshooting.html) - all of the most common fixes for bad connection are explained there. If that doesn't help, try [switching into the download mode manually](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#manual-bootloader). You can verify the connection with any of [serial terminal programs](https://docs.espressif.com/projects/esptool/en/latest/esp8266/troubleshooting.html#serial-terminal-programs) - if you connect to the chip and see some output when you reset it, serial communication is working fine and most likely it's the mentioned wrong boot mode.

radimkarnis, 2022-05-27:
Closing for inactivity, this issue is probably not relevant anymore. Feel free to reopen if needed.

closed 2022-05-27 by radimkarnis