espressif/esptool #740: Syntax error in version 4.0

AlfredoCubitos, 2022-05-16:
### Operating System

Linux OpenSuse Tumbleweed

### Version

other

### Python Version

3.8.13

### Full Esptool Command Line that Was Run

esptool.py

### Esptool Output

```shell
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 27, in <module>
    esptool._main()
AttributeError: module 'esptool' has no attribute '_main'
```

### What is the Expected Behaviour?

no error

### More Information

This issue is refers  esptool.py  version 4.0

### Other Steps to Reproduce

run esptool.py

radimkarnis, 2022-05-17:
Hello @AlfredoCubitos,
we have just recently tried to fix this issue in https://github.com/espressif/esptool/commit/30821a3f936e13e178afb0bd0e9ff21b5b68e970.

Do you simply install esptool as `pip install esptool` and run it on commandline like `esptool.py -h` or are there any other steps to reproduce? Thank you.

AlfredoCubitos, 2022-05-17:
Hello @radimkarnis,

an update with `pip install esptool` does not work ` pip install --upgrade esptool` neither.
Seems you have to deploy new version. The current version remains to 4.0

I also tried to `uninstall` and `install` the package again. This doesn't work neither.

radimkarnis, 2022-05-18:
A possible solution is mentioned here: https://github.com/espressif/esptool/pull/742#issuecomment-1129674887.

@AlfredoCubitos, would you mind trying these changes to verify this fixes the issue? You can [install esptool in development mode](https://docs.espressif.com/projects/esptool/en/latest/esp32/contributing.html#development-setup) and just change the lines of code.

igor-d-n, 2022-05-18:
I'm getting the same error after installing it into docker container. Version 4.0 is broken( 3.3 still works good.

dotlambda, 2022-05-18:
@AlfredoCubitos Can you post the output of `which python3` and `which esptool.py`?

AlfredoCubitos, 2022-05-19:
`which python3     /usr/bin/python3
`
```
which esptool.py
/home/alfredo/.local/bin/esptool.py
```
This output is because I have installed the development environment as @radimkarnis requested

When I use `pip3 install esptool `the path is` /usr/local/bin/esptool.py`

As I mentioned above, the problem is to set the correct  shebang in all modules during setup.
In /`usr/local/bin/esptool.py` the shebang is correct set, according to the python3 environment variable,  but in the related modules not

AlfredoCubitos, 2022-05-19:
ah, and removing the not needed shebangs may work as well.
There is a pull request pending

WouterGritter, 2022-05-22:
I am also getting this exact behavior when installing esptool for the very first time. Using venv (through PyCharm), the tool works fine.

closed 2022-05-24 by espressif-bot

dobairoland, 2022-05-24:
v4.0.1 was release to address this issue. Please let us know if you still experience this issue in any of the supported setups.

radimkarnis, 2022-05-24:
@AlfredoCubitos thanks for confirming and your cooperation!