espressif/esptool #741: SOLVED  ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.

goldwing06, 2022-05-17:
### Operating System

ubuntu 20.04

### Version

v3.2

### Python Version

3

### Full Esptool Command Line that Was Run

python3 -m esptool --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 Downloads/esp32-20220117-v1.18.bin 

### Esptool Output

```shell
python3 -m esptool --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 Downloads/esp32-20220117-v1.18.bin 
esptool.py v4.0
Serial port /dev/ttyUSB0
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

### What is the Expected Behaviour?

Should be load firmware on esp32

### More Information

ı share solution

goldwing06, 2022-05-17:
Solution is press boot button on the esp32 board when run the code, its simple

closed 2022-05-18 by radimkarnis

iiKurt, 2022-09-05:
> Solution is press boot button on the esp32 board when run the code, its simple

this comment is actually more helpful than the documentation. The docs say to press both buttons, in order to start the ESP in download mode, but esptool appears to reset the board over USB, restarting it in regular mode. The key here is to hold the boot button while esptool is resetting the board over USB, so the ESP restarts in download mode.

benreu, 2022-09-12:
What's strange about this is, in Windows you don't have to press the button every time. The root problem is not in the ESP32, but in the code or something in Linux.

nicohofdtz, 2022-10-14:
> this comment is actually more helpful than the documentation. The docs say to press both buttons to start the ESP in download mode, but esptool appears to reset the board over USB, restarting it in regular mode. The key here is to hold the boot button while esptool is resetting the board over USB, so the ESP restarts in download mode.

Thank you, that worked for me and is something entirely different from what the documentation says

`For example, to manually reset a development board, hold down the Boot button (GPIO0) and press the EN button (EN (CHIP_PU)).`

danieldeschamps, 2022-10-29:
I found the same behavior when comparing ESP-IDF over windows and ESP-IDF running in Ubuntu virtual box. When in windows, I can flash normally. When running in Ubuntu, I need to press the Boot key in the board. This must be related to virtual serial port driver implementation because ESP-IDF python scripts use RTS and DTR RS-232 signals to force the board into programming mode. Somehow when in linux, these signals are not being set properly. Hence, user needs to manually place the board into programming mode by pressing the Boot key.

danieldeschamps, 2022-10-30:
The "press Boot switch" solution always works fine in Linux. But as I wanted to avoid pressing the mechanical switch every flashing, then I started tweaking esptool.py 
Incrementing some delays when seting RTS and DTR actually helped. See here the lines of code I have changed:
[image]

benreu, 2022-10-31:
Setting the second `time.sleep(0.5)` did the trick for me. I didn't have to touch the first line. Not that it matters, but your second time.sleep() is line 597, mine was line 594. 

@danieldeschamps I much appreciate you posting this update. 

Now if we could convince Espressif to update this officially...

dobairoland, 2022-10-31:
Thanks for sharing the right delays for you. We will use these....

Napoleon-BlownApart, 2022-11-10:
> Solution is press boot button on the esp32 board when run the code, its simple

This needs to be HIGHLIGHTED in the documentation, and everywhere else.
I've spent around 4 hours trying to work out why this problem was occurring.

daz, 2022-11-15:
> Neither values above helped me, nor did the increase til 5 seconds. But it was worth a try.

Yeah adjusting the delays aren't working for me either. DOIT dev kit, macOS Ventura.

benreu, 2022-11-15:
Occasionally, my ESP32 will still get 'stuck'. Unplugging and reconnecting seems to fix it.

daz, 2022-11-15:
I solved it by putting a 10uF capacitor between EN and GND

ajieh89, 2022-11-21:
I solved it with this 
https://github.com/espressif/esptool/commit/9806318ad5da9468170bfdd3455208960c129f76#diff-075b42fc6a0cd6c91c04e4a547ece428345e7c370c94cc2a4904b9015788b980

QGB, 2022-11-23:
```
Rebooting...
▒xESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375708
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x404
load:0x403b6000,len:0x6ec
load:0x403ba000,len:0x26b4
entry 0x403b6138

################################################

arch:ESP32S3,0
company:Ai-Thinker|B&T
wifi_mac:F412FA41649C
ble_mac:F412FA41649E
sdk_version:v4.4-dev-3042-g220590d599-dirty
firmware_version:1.0.1
compile_time:Feb  8 2022 14:37:32

ready

################################################
command '
################################################

arch:ESP32S3,0
company:Ai-Thinker|B&T
wifi_mac:F412FA41649C
ble_mac:F412FA41649E
sdk_version:v4.4-dev-3042-g220590d599-dirty
firmware_version:1.0.1
compile_time:Feb  8 2022 14:37:32

ready

################################################' not found
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420061c4  PS      : 0x00060a30  A0      : 0x8037f131  A1      : 0x3fc9f840
A2      : 0x3fc9d7c0  A3      : 0x00000104  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x820061c4  A9      : 0x3fc9f820
A10     : 0x3fc9d7c0  A11     : 0x00000000  A12     : 0x00000100  A13     : 0x3fc9d8c0
A14     : 0x00000001  A15     : 0x00000010  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0x00000000

Backtrace:0x420061c1:0x3fc9f8400x4037f12e:0x68637261  |<-CORRUPTED

ELF file SHA256: 4f1cc2960d991065

Rebooting...
▒xESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (R

```

QGB, 2022-11-23:
```

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download

```

QGB, 2022-11-23:
```
esptool --chip esp32s3 --port {COM} erase_flash
esptool.py v3.2
Serial port COM5
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting

```

tofublock, 2022-11-25:
> I solved it with this 
> https://github.com/espressif/esptool/commit/9806318ad5da9468170bfdd3455208960c129f76#diff-075b42fc6a0cd6c91c04e4a547ece428345e7c370c94cc2a4904b9015788b980

This flawlessly works for me under Debian when invoked from the command line, but doesn't work with `import esptool` and `esptool.main(...)` from another Python script so far. Will investigate further. `os.system("esptool.py. ...")` does work, unsurprisingly.

This issue should be reopened since it's clearly a bad behaviour of esptool under certain circumstances, and "press the button" is not an acceptable solution to it.

reneaye, 2022-11-30:
> 

Using a ESP32 Dev Kit C v4 on macOS Montery and PlatformIO Core 6.1.5 which uses esptool.py v3.3
I also managed auto bootloadring by using a 47uF capacitor between GND and EN

SharpEdgeMarshall, 2022-12-06:
> I solved it with this [9806318#diff-075b42fc6a0cd6c91c04e4a547ece428345e7c370c94cc2a4904b9015788b980](https://github.com/espressif/esptool/commit/9806318ad5da9468170bfdd3455208960c129f76#diff-075b42fc6a0cd6c91c04e4a547ece428345e7c370c94cc2a4904b9015788b980)

This worked also for me, why this is not merged?

radimkarnis, 2022-12-06:
Thank you all for the feedback!

> This worked also for me, why this is not merged?

@SharpEdgeMarshall because it wasn't fully tested and doesn't work on all OSes. It will be probably integrated soon. See https://github.com/espressif/esptool/issues/712#issuecomment-1337920344

farhan2077, 2022-12-28:
> What works for me is: Press and hold Boot button, click EN button, click Upload, release Boot button when "Connecting...." is displayed.

Thank you @nicohofdtz. That worked perfectly.

radimkarnis, 2023-01-04:
@tofublock @SharpEdgeMarshall @ajieh89 this has now been merged into master. Feel free to test!

ryan-mcclue, 2023-01-07:
@radimkarnis I discovered this thread as I recently bought an ESP32 and was following the startup guide today and encountered this issue. I'm flashing via `idf.py -p PORT flash` and still have to hold down the Boot button for it to work

radimkarnis, 2023-01-09:
Hi @ryan-mcclue, this issue has been only fixed in the [esptool](https://github.com/espressif/esptool) repository so far. IDF (which uses esptool under the hood) still ships with an older version, which doesn't include this bugfix yet. That's why running `idf.py flash` might still fail.

We will update the esptool version in IDF today. In the meantime, you can try to use esptool independently of IDF and see if that helps.

developeralgo8888, 2023-01-12:
Erasing or uploading anything to ESP32-S3-WROOM-1 Module using Espressif ESP-PROG  seems to be a epic failure everytime. I have tried pressing and holding the boot button and that has failed to enter bootloader mode .

but i have also tried only pressing reset button on ESP-PROG, The error is serial not detected. The voltage that i can measure on on EN  is 3.28V and GPIO0  is 3.20V . The input voltage through ESP-PROG is around 3.3V

i have tried everything  possible and no luck to have it work. Even using different Laptops, PCs, OSes but nothing works. The Serial connection seems to be fine. I have test  ESP32-S3-WROOM-1 devboard that works fine since i am able to erase_flash and write_flash on the dev board. 

i have reached the end of my  endless hours trying to solve it . i help.

`# esptool.py -p /dev/ttyUSB2 -b 115200 --before default_reset --after hard_reset --chip esp32s3 erase_flash
esptool.py v4.4
Serial port /dev/ttyUSB2
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: Wrong boot mode detected (0x14)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html`

When i do screen monitor 

`#screen /dev/ttyUSB2 115200 `

i can see that the flash seems to be empty i.e has no image 

`invalid header:  0xa5ff005a
invalid header:  0xa5ff005a
invalid header:  0xa5ff005a
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TGOWDT_SYS_RST), boot:0x1c (SPI_FAST_FLASH_BOOT)
Saved PC:0x400048836
invalid header:  0xa5ff005a
invalid header:  0xa5ff005a
invalid header:  0xa5ff005a
`

radimkarnis, 2023-01-12:
@developeralgo8888 how do you connect the `ESP32-S3-WROOM-1` to the `ESP-PROG`?

If you switch the module into download mode manually (hold the BOOT button and press the RESET button) and then run the esptool command with `--before no_reset`, it should work. If not, your IO0 and EN connections are wrong.

developeralgo8888, 2023-01-12:
@radimkarnis , connecting using Tag-Connect 2030 cable and module .  UBUNTU 22.04  ==> ttyUSB2  => USB cable ==> ESP-PROG ==> Tag-Connect Cable 2030 cable ===> Tag-Connect pad on the PCB ==> ESP32-S3-WROOM-1 Module

radimkarnis, 2023-01-12:
@developeralgo8888 does the PCB connect the Tag-Connect pad with the GPIO0 and EN pins of the room module? ([more info here](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/advanced-topics/boot-mode-selection.html)). Is there maybe a jumper or a switch on the PCB for bridging GPIO0 to GND?

developeralgo8888, 2023-01-12:
@radimkarnis , Yes they do. i am not sure i have to confirm on the PCB if there is bridging GPIO0 to GND

radimkarnis, 2023-01-12:
@developeralgo8888 Can you verify that GPIO0 is indeed pulled low when you press the BOOT button? If yes, simply hold the BOOT button and disconnect and reconnect the USB cable (to cut and restore the power). It then has to work if you run the commands with `--before no_reset`.

developeralgo8888, 2023-01-12:
@radimkarnis, The GPIO0 is pulled low when i press boot. When i measure voltage on GPIO0 pin itself on the module, and also at the Tag-Connect-2030 pad pin 6 , i get

Boot Not Pressed  = 3.20V 
Boot Pressed = 0.0 V  ( Pulldown )

When i try it with Boot button pressed  and `--before no_reset `, i get the error below 
`
A fatal error occurred: Failed to connect to ESP32-S3: Invalid head of packet (0x66): Possible serial noise or corruption.`

radimkarnis, 2023-01-12:
@developeralgo8888 ok so we are moving somewhere. Now you can try monitoring the output (with `#screen /dev/ttyUSB2 115200` or `python -m serial.tools.miniterm /dev/ttyUSB2 115200`) and try pressing the combination (hold BOOT, press RESET). The log you see should stop and you should see something like this:

```
...
invalid header:  0xa5ff005a
invalid header:  0xa5ff005a
invalid header:  0xa5ff005a
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON_RESET),boot:0x0 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_FEO_V2))
waiting for download
```

If you get to this state, you have successfully entered the download mode and you can run the commands. 
`Invalid head of packet (0x66): Possible serial noise or corruption.` this tells us it still isn't happening correctly.

developeralgo8888, 2023-01-12:
@radimkarnis,

On the ESP-PROG Board i have noted the following:

With ESP-PROG Boot button Pressed and held, its alternating between but also when continually pressed and released:

```
invalid header: 0xa5ff005a
invalid header: 0xa5ff005a
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x14 (SPI_FLASH_BOOT)
Saved PC:0x40049b21
```

-------------------------

```
invalid header: 0xa5ff005a
invalid header: 0xa5ff005a
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x14 (SPI_FLASH_BOOT)
```

--------------------------------

When ESP-PROG Reset button Pressed and held : No logs are available but if button is released it prints
-------------------------------------------------

```
invalid header: 0xa5ff005a
invalid header: 0xa5ff005a
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
```

When ESP-PROG Reset Button is continually pressed and released
------------------------------------------------------------------
```
invalid header: 0xa5ff005a
invalid header: 0xa5ff005a
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
```

radimkarnis, 2023-01-12:
@developeralgo8888 you've confirmed that both of the buttons on the ESP-PROG work as intended. The behavior you've described is correct, there is nothing wrong. Now I just need you to follow EXACTLY what I tell you:

1) Start monitoring
2) Press and HOLD the BOOT button (keep holding)
3) Press and release the RESET button (do not hold it)
4) The `waiting for download` log should appear
5) Release the BOOT button
6) Stop monitoring
7) Run the esptool commands with `--before no_reset`

developeralgo8888, 2023-01-12:
@radimkarnis, is there a way to log slower ttyUSB output  , it's scrolling extremely fast to even read it in real-time without logging it to a file. But let me try

developeralgo8888, 2023-01-13:
@radimkarnis,  i followed exactly what you outlined but step  4:  The `waiting for download `log should appear  . The `waiting for download ` does not print or appear on the monitor

Is the scrolling log on the ttyUSB2 monitor  supposed to stop when `waiting for download` appear . I have gone through the process several times but it does not appear. Does it appear immediately after the release on of the Reset button ?

henriquenunez, 2023-03-06:
> I solved it by putting a 10uF capacitor between EN and GND

Thank you so much! I have the same setup and it worked now. Not sure if is the OS version or because the board was unused for some time XD. Saved me :)

Klrfl, 2023-07-26:
I didn't plug the output pin when uploading. That apparently fixes the issue

lockie555, 2023-07-29:
My TTGO-T4 board flashed OK on two occasions but now consistently gives the "Wrong boot mode detected " error. Nothing has changed.  Could experienced contributors explain the failure modes of flash memory and its support chips. My TFT screen flashes white about 6 times and then indicates that the previous program is running. I've tried all the combinations of boot and reset suggested but nothing is now working, hence my belief I have a failure somewhere on the T4. Understanding failure scenarios may save countless hours of fruitless attempts to flash.
Thanks in advance.

lockie555, 2023-07-29:
Some additional information.: I tried 8 brand new, (never used) micro USB cables and 6 of them would not allow the USB port to be seen, while two did find USB port 10. I thought that was unusual. The micro USB connectror looks fine under a magnifier.

xAlpharax, 2023-11-04:
> Solution is press boot button on the esp32 board when run the code, its simple

Thank you.

chrgraf, 2023-12-29:
> I solved it by putting a 10uF capacitor between EN and GND

This worked for me as well. Thanks alot for this suggestion. Funny thing is, that the same board can be flashed perfectly on one win11 PC, while the other WIN11 PC always fails. Adding the capacitor did the trick and now flashing the ESP32 works fine on all mine PC's 

thanks

NicoCaldo, 2024-01-31:
I'm having the same issue on Ubuntu running on a VM. With WIndows 10 everything works ok

`A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x3f)! The chip needs to be in download mode.`

> I solved it with this [9806318#diff-075b42fc6a0cd6c91c04e4a547ece428345e7c370c94cc2a4904b9015788b980](https://github.com/espressif/esptool/commit/9806318ad5da9468170bfdd3455208960c129f76#diff-075b42fc6a0cd6c91c04e4a547ece428345e7c370c94cc2a4904b9015788b980)

WHere do I find this file? I'm using ESP-IDF on VS Code

Ramirezassasin, 2024-02-26:
> > this comment is actually more helpful than the documentation. The docs say to press both buttons to start the ESP in download mode, but esptool appears to reset the board over USB, restarting it in regular mode. The key here is to hold the boot button while esptool is resetting the board over USB, so the ESP restarts in download mode.
> 
> Thank you, that pointed me in the right direction. The documentation is not very clear to me here. It says: `For example, to manually reset a development board, hold down the Boot button (GPIO0) and press the EN button (EN (CHIP_PU)).` But from that it was not clear to me that I had to keep holding the Boot button while uploading.
> 
> What works for me is: Press and hold Boot button, click EN button, click Upload, release Boot button when "Connecting...." is displayed.

ty ude

messplay, 2024-03-14:
> I solved it by putting a 10uF capacitor between EN and GND

this was a solution for me too, I tested with a RC pair R=10K and C=10uF (1uF like datasheet say not work for me)

villiambo1997, 2024-04-06:
I placed a 2.2 muF capacitor from EN to Ground. Worked immediatley for me. Im using Windows, PlatformIO, ESP-WROOM-32

https://www.youtube.com/watch?v=SlAG6DFLnBE

Himanshu-N, 2024-04-25:
I had the same problem.
Following this "Tools>Programmer>Esptool" worked for me.

Lyroz-my, 2024-06-14:
Try to unplug the SD card before upload

marco-not-polo, 2024-06-21:
So If you have connected other devices to GPIO pins, try removing them and see if esptool starts working. 
Once the upload is finished, just reconnect the GPIO... 
This worked for me...

klaatu2a, 2024-08-06:
Where is the documentation which describes how to use this tool?

pabloldz, 2024-09-13:
> Thank you, that pointed me in the right direction. The documentation is not very clear to me here. It says: `For example, to manually reset a development board, hold down the Boot button (GPIO0) and [...]

works!!!

nicolasgomez1, 2024-10-06:
> I'm having the same issue on Ubuntu running on a VM. With WIndows 10 everything works ok
> 
> `A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x3f)! The chip needs to be in download mode.`
> 
> > I solved it with this [9806318#diff-075b42fc6a0cd6c91c04e4a547ece428345e7c370c94cc2a4904b9015788b980](https://github.com/espressif/esptool/commit/9806318ad5da9468170bfdd3455208960c129f76#diff-075b42fc6a0cd6c91c04e4a547ece428345e7c370c94cc2a4904b9015788b980)
> 
> This worked

sorry, i'm newbie, what's exactly i have to do?

PepeTux, 2024-11-08:
> sorry, i'm newbie, what's exactly i have to do?

1. Presiona el boton de reset y manten presionado.
2. Presiona el boton de Boot y manten presionado.
3. Suelta el boton de reset pero manten presionado el boton de boot hasta que tu programador establezca la comunicacion.

Hazlo en esa secuencia para que tu ESP32 pueda entrar en modo Bootloader. ✌️

De preferencia usa estos cores que son mas nuevos y no tienen ese problema:

ESP32-C3
ESP32-S2
ESP32-S3

Saludos
JL

t-radomski, 2025-01-23:
> [@developeralgo8888](https://github.com/developeralgo8888) Can you verify that GPIO0 is indeed pulled low when you press the BOOT button? If yes, simply hold the BOOT button and disconnect and [...]

This worked for me. ESP32-WROOM-32E

20388076, 2025-11-08:
I think i found the problem. i used a FireBeetle Board ESP32 E SKU DFR0654 with ESP-WROOM-32E processor and it auto flash the file only in Itel cpu and not on Amd. i used 3 Amd laptops with 3000, 5000 and 7000 series cpus and 3 Itel (but i didn't note the series of cpus). I think the problem comes to the cpu - mcu timing that the new RISK based board don't have and messing with time values for every Amd cpu is not a valid approach. I also believe the issue remain and the the documentation is incomplete. 
It work only when you see the (Connecting.... and it counts with dots appear) then press the RESET button and when the dots stop appear you release the button. 
But for auto upload the best solution i found on this page: https://www.makerguides.com/wrong-boot-mode-esp32/ is to put a 1μF capacitor on reset or enable pin with ground.