espressif/esptool #743: Error after upgrade from V3.3.1 to V4.0

water5, 2022-05-20:
### Operating System

Fedora 34

### Version

latest master

### Python Version

Python 3.9.12

### Full Esptool Command Line that Was Run

esptool.py

### Esptool Output

```shell
Traceback (most recent call last):
  File "/home/mc3/.local/bin/esptool.py", line 27, in <module>
    esptool._main()
AttributeError: module 'esptool' has no attribute '_main'
```

### What is the Expected Behaviour?

Output `esptool` version information, and usage information.

### More Information

Upgrade `esptool` via `python -m pip install --upgrade esptool` from V3.3.1 to V4.0.

On same system, downgrade to V3.3.1 via `python -m pip install --upgrade esptool==3.3.1`, `esptool` works fine.

fiechr, 2022-05-21:
I get the same error on Manjaro.

water5, 2022-05-21:
On Windows, not have this issue, I think some environment configure incorrect.

fiechr, 2022-05-21:
I also don't have issues, if I use pipx to install the esptool package (pipx isolates each installed application inside its own virtualenv).

```
$ pipx install esptool
⚠️  Note: espefuse.py was already on your PATH at /usr/bin/espefuse.py
⚠️  Note: espsecure.py was already on your PATH at /usr/bin/espsecure.py
⚠️  Note: esptool.py was already on your PATH at /usr/bin/esptool.py
  installed package esptool 4.0, installed using Python 3.10.4
  These apps are now globally available
    - espefuse.py
    - espsecure.py
    - esptool.py
done! ✨ 🌟 ✨
$ pipx list
venvs are in /home/cf/.local/pipx/venvs
apps are exposed on your $PATH at /home/cf/.local/bin
   package esptool 4.0, installed using Python 3.10.4
    - espefuse.py
    - espsecure.py
    - esptool.py
$ which esptool.py
/home/cf/.local/bin/esptool.py
$ esptool.py version
esptool.py v4.0
4.0
```

water5, 2022-05-21:
I try install from source code using `python setup.py install`, have same error.

dobairoland, 2022-05-21:
Duplicate of https://github.com/espressif/esptool/issues/740. Please follow that one.

closed 2022-05-21 by dobairoland