espressif/esptool #744: [4.0] espefuse.exe - Win64 build attempts to open `/dev/ttyUSB0`

henrygab, 2022-05-23:
### Operating System

Windows 21H2 (OS Build 22000.675)

### Version

other

### Python Version

stand-alone compiled exe from release (if needed, I have 3.9.13 installed)

### Full Esptool Command Line that Was Run

espefuse.exe adc_info

### Esptool Output

```shell
X:\esptool-v4.0-win64>espefuse.exe adc_info
Traceback (most recent call last):
  File "espefuse.py", line 28, in <module>
  File "espefuse\__init__.py", line 268, in _main
  File "espefuse\__init__.py", line 211, in main
  File "espefuse\__init__.py", line 80, in get_esp
  File "esptool\cmds.py", line 88, in detect_chip
  File "esptool\loader.py", line 246, in __init__
  File "serial\__init__.py", line 90, in serial_for_url
  File "serial\serialwin32.py", line 64, in open
serial.serialutil.SerialException: could not open port '/dev/ttyUSB0': FileNotFoundError(2, 'The system cannot find the path specified.', None, 3)
[15732] Failed to execute script 'espefuse' due to unhandled exception!
```

### What is the Expected Behaviour?

Any of the following behaviors would be acceptable when the user does not specify a port (and only one COM port exists on the system):

1. auto-detect the port (similar to esptool.exe)
2. print a message indicating that port must be specified

### More Information

Running the same command with explicit port works:
```
X:\esptool-v4.0-win64>espefuse.exe --port COM6 adc_info
Connecting....
Detecting chip type... ESP32-S3
espefuse.py v4.0

=== Run "adc_info" command ===

BLOCK2_VERSION = 3
```

Running `esptool.exe` with no explicit port specified works:
```
C:\T\Pixels\esptool-v4.0-win64>esptool chip_id
esptool.py v4.0
Found 1 serial ports
Serial port COM6
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 84:f7:03:c0:84:a4
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-S3 has no Chip ID. Reading MAC instead.
MAC: 84:f7:03:c0:84:a4
Hard resetting via RTS pin...
```

radimkarnis, 2022-05-24:
Hello @henrygab,

this is completely intentional. We refrain from any kind of port auto-detection or auto-selection for safety reasons. We don't want the users to burn any eFuses by accident.

> print a message indicating that port must be specified

I agree this would be more user-friendly. I am also open to the idea of enabling port-autodetection for non-destructive commands (`dump`, `summary`, `adc_info`, ...). 

Thanks for the feature request. Let me know what you think about this.

henrygab, 2022-05-24:
@radimkarnis ... I thought it might be intentional to not auto-select the port, which is why I provided the two alternatives.

The error message was:
```
serial.serialutil.SerialException: could not open port '/dev/ttyUSB0': FileNotFoundError(2, 'The system cannot find the path specified.', None, 3)
```

The tool requires the user to specify a port, but when the do not, the tool throws an exception, rather than printing a useful error message that the `--port` parameter is non-optional.

As to the weirdness about printing *nix-style paths (`/dev/ttyUSB0`) rather than Windows style path for the COM port:  I'm OK punting that down the line, with a fix for the mandatory parameter.

closed 2022-07-26 by espressif-bot