espressif/esptool #745: [4.0] read_flash - stub cannot read full 16MB

henrygab, 2022-05-24:
### Operating System

Windows 11 (21H2 Build 22000.675)

### Version

other

### Python Version

.EXE release v4.0 of tool (although I have 3.9.13 installed)

### Chip Description

ESP32-S3-DevKitC-1 (N32R8V)

### Device Description

ESP32-S3-DevKitC-1 (N32R8V) ... which has 32MB of flash, but I'm only trying to dump 16MB using the stub....

### Hardware Configuration

Nothing but a USB cable connecting the `UART` port to the PC.

### How is Esptool Run

No IDE, straight from cmd.exe

### Full Esptool Command Line that Was Run

esptool --baud 115200 read_flash 16252928  524288  ..\ESP32-S3-DevKitC-1-N32R8__start-at-15.5MB.bin

### Esptool Output

```shell
esptool.py v4.0
Found 1 serial ports
Serial port COM8
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e0:f0:4c
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Invalid head of packet (0x6F): Possible serial noise or corruption.
```

### More Information

This is a fresh-from-the-factory device, never written to, and still running the default blink sketch.

When trying to dump 32MB, the tool reports (as expected) that the stub supports up to 16MB:

    WARNING: Flasher stub doesn't fully support flash size
    larger than 16MB, in case of failure use --no-stub.

The following tables show 100% reproducible results for various combinations of start address and size being dumped.

<details><summary>Fails immediately when end address == 16MB</summary><P/>

The following is the fatal error printed for each failure below:

    `A fatal error occurred: Invalid head of packet (0x6F): Possible serial noise or corruption.`

address | addressMB | size | SizeMB
-----|-----|-----|-----
0 | 0MB | 16777216 | 16MB
12582912 | 12MB | 4194304 | 4MB
13631488 | 13MB | 3145728 | 3MB
14680064 | 14MB | 2097152 | 2MB
15728640 | 15MB | 1048576 | 1MB

Seems the stub cannot actually be used to dump 16MB flash directly, because it does not like the end address being equal to 16MB.

</details><HR/>
<details><summary>Starting at 0, fails dumping >8MB</summary><P/>

address | addressMB | size | SizeMB | error | address @ error
-----|-----|-----|-----|-----|-----
0 | 0MB | 8388608 | 8MB | Succeeds | 0MB-8MB ok
0 | 0MB | 9437184 | 9MB | Fails at **_77%_** | <8MB
0 | 0MB | 10485760 | 10MB | Fails at **_60%_** | 6MB
0 | 0MB | 11534336 | 11MB | Fails at **_45%_** | <5MB
0 | 0MB | 12582912 | 12MB | Fails at **_33%_** | 4MB
0 | 0MB | 13631488 | 13MB | Fails at **_23%_** | 
0 | 0MB | 14680064 | 14MB | Fails at **_14%_** | 
0 | 0MB | 15728640 | 15MB | Fails at **_6%_** |

I have not noticed how the inputs affect these 100% repro failures, based on this evidence.

</details><HR/>
<details><summary>100% repro with varied start/size</summary><P/>

address | addressMB | size | SizeMB | error | address @ error
-----|-----|-----|-----|-----|-----
9437184 | 9MB | 4194304 | 4MB | Fails at **_75%_** | 13MB
9437184 | 9MB | 3145728 | 3MB | Succeeds | 9MB-12MB ok
9437184 | 9MB | 2097152 | 2MB | Succeeds | 9MB-11MB ok
10485760 | 10MB | 4194304 | 4MB | Fails at **_50%_** | **12MB**
10485760 | 10MB | 3145728 | 3MB | Succeeds(!) | 10MB-13MB ok
11534336 | 11MB | 4194304 | 4MB | Fails at **_25%_** | 12MB
11534336 | 11MB | 3145728 | 3MB | Fails at **_66%_** | 13MB
11534336 | 11MB | 2097152 | 2MB | Succeeds(!) | 11MB-13MB ok
12582912 | 12MB | 3145728 | 3MB | Fails at **_33%_** | 13MB
12582912 | 12MB | 2097152 | 2MB | Succeeds(!) | 12MB-14MB ok
13631488 | 13MB | 2097152 | 2MB | Fails at **_50%_** | 14MB
13631488 | 13MB | 1572864 | 1.5MB | Succeeds(!) | 13MB-14.5MB ok
13107200 | 12.5MB | 2097152 | 2MB | Fails at **_75%_** | 14MB
14155776 | 13.5MB | 2097152 | 2MB | Fails at **_25%_** | 14MB
14155776 | 13.5MB | 1048576 | 1MB | Succeeds(!) | 13.5MB-14.5MB ok

The above table confuses my guesses ... although 100% reproducible, the failure point does not appear to correspond to an address, does not appear to correspond to an amount completed, ... Not sure what else I could offer here.

</details><HR/>

So, in summary, the stub can read many of the addresses, but there's something wrong with its handling of >8MB (not >16MB as suggested).  See [attached traces](https://github.com/espressif/esptool/files/8766311/TraceCommands.zip) for trace output of two commands that immediately report the `Invalid head of packet (0x6F)` error.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

henrygab, 2022-05-24:
Reading with `--no-stub` also fails to read much of the flash.
Has this been validated on `N32R8V` (WROOM-2) ESP32-S3?

How much of the flash should I expect to be usable currently?

I will adjust my partition table accordingly, even if temporarily must limit to 8MB flash....

henrygab, 2022-09-27:
Great!  Based on the bot comment (ESPTOOL-460), I understand that this entered the Espressif bug triage database.

(( In other words, someone will at least look at this issue some more. ))

Looking forward to the stub (or even the no-stub) working for these 32MB Flash devkits....

dobairoland, 2022-09-27:
Hi @henrygab. I'm sorry no one has left a note for you. Could you please try the latest v.4.3 release. Maybe https://github.com/espressif/esptool/commit/31908948d87da61b40332a7628177de222e2d40f already solves this.

henrygab, 2022-09-27:
Hi @dobairland,

With 4.3, the `flash_id` subcommand now properly reports the 32MB flash size.  In addition, when using `--no-stub`, the attempt to read the full flash would start. (but so **_slow_**).  I'm happy to pretend this is really a 16MB flash, to be able to use the stub, and thus flash in less than an hour.  🙂 

Unfortunately, the issue persists with v4.3:

```
.\esptool.exe --baud 115200 read_flash 0 16777216 .\image.bin
...
A fatal error occurred: Invalid head of packet (0x6F): Possible serial noise or corruption.
```

FYI, the following works reliably at higher baud rate, so the cables are not suspect:
```
.\esptool.exe --baud 921600 read_flash 0 8388608   .\image.bin
...
8388608 (100 %)
Read 8388608 bytes at 0x00000000 in 131.8 seconds (509.2 kbit/s)...
Hard resetting via RTS pin...
```

Anyways, the failure is 100% repro, and takes less than 10 seconds: Simply connect a DevKit-C1 (N32R8) using its UART port.

dobairoland, 2022-09-27:
Thanks for checking.  We will try to address this soon....

radimkarnis, 2022-10-20:
Hello @henrygab,
thanks for checking in.

I can verify that the issue is completely reproducible, your detailed report is spot on.

The root cause is not fully known yet. You can see `opi_flash.c 270` appear in the serial terminal when the fail happens. I tracked it down to a failing assertion in the ROM (so the problem is not in the stub, that's why `--no-stub` fails too). 

We haven't come up with a fix, yet.

henrygab, 2022-11-23:
Hello @radimkarnis ,

First, I want to thank you for your open communication on what you discovered in the last post.  I appreciate your sharing the status.

Another month has flown by, so I thought I'd check in again ... see if any updates (even if only that you've tried three ideas, and each one hit a blocking issue; or that a new ROM / chip revision will be necessary).

I realize it's not possible to change the ROM, so any software workaround would require some amazing hack.   If you do end up creating such a hack, I'd love to hear the story ... and both cringe and cheer at your creative solution!

Thanks again for the last update, letting us all know it's related to a failing assertion in the ROM, and good luck!

Jason2866, 2022-11-28:
@henrygab Imho it should be possible to solve with the stub. Since the stub does not have the need to use the (faulty?) rom routines. Everything needed can be baked in the stub code.

daviddohm, 2022-12-19:
I thought I had found an answer on an unrelated forum (forum shown below). 

While it doesn't fail with the "Invalid head of packet (0x6F):" error immediately, read_flash fails at 2097152 (14%) regardless of what baud rate I attempt.

e.g. "esptool -b 115200 read_flash 0 0xE00000 flash_contents.bin" running on ubuntu 18.04.

https://forum.micropython.org/viewtopic.php?t=7138
" on a 16MB chip, the size should be changed to 0xE00000. "

henrygab, 2022-12-22:
@radimkarnis -- Monthly request for an update.

Can you verify if the [potential fix](https://github.com/espressif/esptool/issues/795#issuecomment-1321487813) Jason2866 found will be sufficient to resolve the issue?  It does appear to be a bug in the code, and would seem to be directly related.  If this is the right fix, can you push for it to be integrated into a release ASAP?

dobairoland, 2022-12-23:
I'm working on this actively. That potential fix with other similar ones solves only the 16 MB boundary issue. I thought that I will fix it together with the 32 MB support issue but since that is more complicated, I'll separate it and prepare this one earlier.

radimkarnis, 2022-12-23:
@henrygab although the potential fix @Jason2866 made fixes the issue of not being able to read/write the last byte of 16M flash, it doesn't explain the behavior you reported. This isn't related. Nevertheless, I have tried it and it still fails exactly as you described.

henrygab, 2022-12-23:
Bummer!  Would have been great if such a simple root cause.  Happy holidays and new year!

radimkarnis, 2022-12-24:
Wishing you happy holidays and a bright 2023 from your friends at Espressif!

closed 2023-02-03 by espressif-bot

JedidiahPaterson, 2024-05-30:
Just wondering if anyone has a fix to this? 

I am getting a failure right at 2MB 

```
(idf5.1_py3.8_env) $ esptool.py --chip esp32s3 --port=/dev/ttyACM0 -b 115200 --no-stub read_flash 0x0b0000 0x181000 test.bin
esptool.py v4.7.dev1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 30:30:f9:57:30:ec
Enabling default SPI flash mode...
1376256 (87 %)
A fatal error occurred: Failed to read flash block (result was 01090000: CRC or checksum was invalid)
```
the failure address + the offset = 2MB seems weird

I can do the verify_flash command 

`esptool.py verify_flash --diff yes 0x0B0000 build/test.bin`

which returns the correct response digest matching/mismatching