espressif/esptool #746: Timeout packet header

MrJ0se, 2022-05-25:
### Operating System

Windows 11 and KDE Plasma linux

### Version

v3.2

### Python Version

Python 2.7

### Chip Description

ESP8266 (ESP-01)

### Hardware Configuration

GPIO 0, GPIO 2

### How is Esptool Run

Arduino IDE

### Esptool Output

```shell
Original exception was:
Traceback (most recent call last):
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/upload.py", line 66, in <module>
    esptool.main(cmdline)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 3552, in main
    esp.connect(args.before, args.connect_attempts)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 534, in connect
    chip_magic_value = self.read_reg(ESPLoader.CHIP_DETECT_MAGIC_REG_ADDR)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 562, in read_reg
    val, data = self.command(self.ESP_READ_REG, struct.pack('<I', addr))
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 386, in command
    p = self.read()
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 331, in read
    return next(self._slip_reader)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 2636, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header
```

### More Information

i "solved it" by commenting the Python code of esptool from line ~ 2632 to 2636 (timeout throw), just it solved all my day of problems with esp, i tryed various "solutions", as reset esp after "connecting...", or change baurate, as many sites suggested, now bring this little repport to help someone with same problem.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-05-26:
Hi @MrJ0se,
I am glad you could manage to resolve your issue.

You report using esptool v3.2, but according to the log, it seems like you are using an older one.
Please report the actual version you are using, otherwise the lines `2632 to 2636` don't make sense.

radimkarnis, 2022-05-27:
Thanks for specifying the version for future reference. Will close this issue now.

closed 2022-05-27 by radimkarnis