espressif/esptool #747: 4.1: pytest is failing

kloczek, 2022-05-27:
### Operating System

Linux/x86_64

### Version

latest master

### Python Version

3.8.13

### Full Esptool Command Line that Was Run

N/A

### Esptool Output

```shell
N/A
```

### What is the Expected Behaviour?

pytest should pass without errors.

### More Information

I'm trying to package your module as an rpm package. So I'm using the typical PEP517 based build, install and test cycle used on building packages from non-root account.
- `python3 -sBm build -w --no-isolation`
- because I'm calling `build` with `--no-isolation` I'm using during all processes only locally installed modules
- install .whl file in </install/prefix>
- run pytest with PYTHONPATH pointing to sitearch and sitelib inside </install/prefix>

Here is pytest output:
<details>

```console
+ PYTHONPATH=/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.1-2.fc35.x86_64/usr/lib64/python3.8/site-packages:/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.1-2.fc35.x86_64/usr/lib/python3.8/site-packages
+ /usr/bin/pytest -ra
=========================================================================== test session starts ============================================================================
platform linux -- Python 3.8.13, pytest-7.1.2, pluggy-1.0.0
rootdir: /home/tkloczko/rpmbuild/BUILD/esptool-4.1
collected 184 items / 4 errors

================================================================================== ERRORS ==================================================================================
______________________________________________________ ERROR collecting test/efuse_scripts/esp32/test_efuse_script.py ______________________________________________________
/home/tkloczko/rpmbuild/BUILD/esptool-4.1/test/efuse_scripts/esp32/test_efuse_script.py:3: in <module>
    espefuse(esp, efuses, args, "burn_efuse JTAG_DISABLE 1 DISABLE_SDIO_HOST 1 CONSOLE_DEBUG_DISABLE 1")
E   NameError: name 'espefuse' is not defined
_____________________________________________________ ERROR collecting test/efuse_scripts/esp32/test_efuse_script2.py ______________________________________________________
/home/tkloczko/rpmbuild/BUILD/esptool-4.1/test/efuse_scripts/esp32/test_efuse_script2.py:3: in <module>
    espefuse(esp, efuses, args, "burn_efuse JTAG_DISABLE 1 DISABLE_SDIO_HOST 1 CONSOLE_DEBUG_DISABLE 1")
E   NameError: name 'espefuse' is not defined
_____________________________________________________ ERROR collecting test/efuse_scripts/esp32xx/test_efuse_script.py _____________________________________________________
/home/tkloczko/rpmbuild/BUILD/esptool-4.1/test/efuse_scripts/esp32xx/test_efuse_script.py:3: in <module>
    espefuse(esp, efuses, args, 'burn_efuse DIS_FORCE_DOWNLOAD 1 DIS_CAN 1 DIS_DOWNLOAD_MODE 1')
E   NameError: name 'espefuse' is not defined
____________________________________________________ ERROR collecting test/efuse_scripts/esp32xx/test_efuse_script2.py _____________________________________________________
/home/tkloczko/rpmbuild/BUILD/esptool-4.1/test/efuse_scripts/esp32xx/test_efuse_script2.py:3: in <module>
    espefuse(esp, efuses, args, 'burn_efuse DIS_FORCE_DOWNLOAD 1 DIS_CAN 1 DIS_DOWNLOAD_MODE 1')
E   NameError: name 'espefuse' is not defined
========================================================================= short test summary info ==========================================================================
ERROR test/efuse_scripts/esp32/test_efuse_script.py - NameError: name 'espefuse' is not defined
ERROR test/efuse_scripts/esp32/test_efuse_script2.py - NameError: name 'espefuse' is not defined
ERROR test/efuse_scripts/esp32xx/test_efuse_script.py - NameError: name 'espefuse' is not defined
ERROR test/efuse_scripts/esp32xx/test_efuse_script2.py - NameError: name 'espefuse' is not defined
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 4 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
============================================================================ 4 errors in 0.34s =============================================================================
```
</details>

kloczek, 2022-05-27:
And after add those four files to --ignore list
<details>

```console
+ PYTHONPATH=/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.1-2.fc35.x86_64/usr/lib64/python3.8/site-packages:/home/tkloczko/rpmbuild/BUILDROOT/esptool-4.1-2.fc35.x86_64/usr/lib/python3.8/site-packages
+ /usr/bin/pytest -ra --ignore test/efuse_scripts/esp32/test_efuse_script2.py --ignore test/efuse_scripts/esp32/test_efuse_script.py --ignore test/efuse_scripts/esp32xx/test_efuse_script2.py --ignore test/efuse_scripts/esp32xx/test_efuse_script.py
=========================================================================== test session starts ============================================================================
platform linux -- Python 3.8.13, pytest-7.1.2, pluggy-1.0.0
rootdir: /home/tkloczko/rpmbuild/BUILD/esptool-4.1
collected 184 items

test/test_espefuse.py FFFFFF                                                                                                                                         [  3%]
test/test_espefuse_host.py FFFFFFFFsFFsFF.FFFsFFFFFssFsssssssssFFsssssssssssssssssssssssFFFFssFFF                                                                    [ 41%]
test/test_espsecure.py FFFFE..FFFFFFFFFFFFFFF..                                                                                                                      [ 53%]
test/test_esptool.py sssFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFsFFFFFFFFFsFFFFFsFFsFFFFFFFF                                                                                  [ 87%]
test/test_imagegen.py ..............                                                                                                                                 [ 95%]
test/test_merge_bin.py ........                                                                                                                                      [ 99%]
test/test_modules.py .                                                                                                                                               [100%]

================================================================================== ERRORS ==================================================================================
____________________________________________________ ERROR at teardown of SigningTests.test_generate_and_extract_key_v2 ____________________________________________________

self = <unittest.case._Outcome object at 0x7fa240c609d0>, test_case = <test_espsecure.SigningTests testMethod=test_generate_and_extract_key_v2>, isTest = False

    @contextlib.contextmanager
    def testPartExecutor(self, test_case, isTest=False):
        old_success = self.success
        self.success = True
        try:
>           yield

/usr/lib64/python3.8/unittest/case.py:60:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
/usr/lib64/python3.8/unittest/case.py:717: in doCleanups
    self._callCleanup(function, *args, **kwargs)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <test_espsecure.SigningTests testMethod=test_generate_and_extract_key_v2>, function = <built-in function rmdir>, args = ('/tmp/tmpqb8asebj',), kwargs = {}

    def _callCleanup(self, function, /, *args, **kwargs):
>       function(*args, **kwargs)
E       OSError: [Errno 39] Directory not empty: '/tmp/tmpqb8asebj'

/usr/lib64/python3.8/unittest/case.py:639: OSError
--------------------------------------------------------------------------- Captured stdout call ---------------------------------------------------------------------------
RSA 3072 private key in PEM format written to /tmp/tmpqb8asebj/key.pem
/tmp/tmpqb8asebj/key.pem public key extracted to /tmp/tmpwv2w4604
ECDSA NIST192p private key in PEM format written to /tmp/tmpqb8asebj/key.pem
================================================================================= FAILURES =================================================================================
________________________________________________________________ TestBurnKey.test_burn_key_34_coding_scheme ________________________________________________________________

self = <test_espefuse.TestBurnKey testMethod=test_burn_key_34_coding_scheme>

    def setUp(self):
        # reset and zero efuses
>       serialport.dtr = False
E       AttributeError: 'NoneType' object has no attribute 'dtr'

/home/tkloczko/rpmbuild/BUILD/esptool-4.1/test/test_espefuse.py:46: AttributeError
________________________________________________________________ TestBurnKey.test_burn_key_no_coding_scheme ________________________________________________________________

self = <test_espefuse.TestBurnKey testMethod=test_burn_key_no_coding_scheme>

    def setUp(self):
        # reset and zero efuses
>       serialport.dtr = False
E       AttributeError: 'NoneType' object has no attribute 'dtr'

/home/tkloczko/rpmbuild/BUILD/esptool-4.1/test/test_espefuse.py:46: AttributeError
_____________________________________________________________ TestBurnBlockData.test_burn_block_data_34_coding _____________________________________________________________
[... 142 lines cut ...]
FAILED test/test_esptool.py::TestErase::test_region_erase - TypeError: sequence item 5: expected str instance, NoneType found
[... cut ...]
FAILED test/test_esptool.py::TestVirtualPort::test_highspeed_flash_virtual_port - IndexError: tuple index out of range
FAILED test/test_esptool.py::TestReadWriteMemory::test_read_write_memory_rom - KeyError: '--ignore'
FAILED test/test_esptool.py::TestReadWriteMemory::test_read_write_memory_stub - KeyError: '--ignore'
=========================================================== 110 failed, 28 passed, 46 skipped, 1 error in 9.93s ============================================================
```
<details

radimkarnis, 2022-05-27:
Hi @kloczek,

The first four scripts (the ones you added to --ignore list) are not meant to be run directly ([more info here](https://docs.espressif.com/projects/esptool/en/latest/esp32/espefuse/index.html#execute-efuse-python-script)).

Nevertheless, `esptool.py` project tests were not written with `pytest` in mind.

We are taking a lo-fi approach of running the test python scripts one by one, [see here](https://github.com/espressif/esptool/blob/master/.gitlab-ci.yml#L52). Most of the test are also meant to be run with real hardware connected. Just running `pytest` will therefore not work out of the box.

kloczek, 2022-05-27:
> Nevertheless, `esptool.py` project tests were not written with `pytest` in mind.

Nevertheless I would try gentrly encorage you to try to have look closer on pytest as it is de facto standard test suite framework fo +8/9 of all modules 😄 
I'm not hiding that as I'm working on some Linux/Solaris distro (now with +940 modules packaged) I'm trying to standardise all tests around pytest because it allow not only test generated packages but allow as well use build procedure to evaluate/audit all packaged python code.
None of other approaches to python modules testing provides such useful platform.
If for example pytest would be useable to it is easy to install for example 'pytest-black` in build env and by execute 'pytest --black` to have integrated with regular test suite `black` module tests. The same is with many, many other pytest extensions allowing keep python code better.

Nevertheless as I found some units working I've added all failing units to --deselect list to have now some baseline for regression tests between versions. In other words I have now kind of solution but it would be good to improve pytest usability 😋

kloczek, 2022-09-14:
Here is pytest output for exptool 4.3 (new units started failing  [esptool-pytest.txt](https://github.com/espressif/esptool/files/9566074/esptool-pytest.txt)

closed 2022-10-25 by espressif-bot