espressif/esptool #748: Send SYNC in nodejs but not respond

JackieLs, 2022-05-29:
### Operating System

macOS 12.4

### Version

v3.2

### Python Version

v3.9.13

### Chip Description

ESP32-D0WDQ6 (revision 1)

### How is Esptool Run

Nodejs

### More Information

```
const { SerialPort } = require('serialport')
const port = new SerialPort({
  path: '/dev/cu.usbserial-14230',
  baudRate: 115200,
  endOnClose: true,
  autoOpen: true,
})
port.on('data', (data) => {
  console.log('data', data.toString())
})

port.on('open', () => {
  _connect_attempt()
})

async function _connect_attempt(mode) {
  if (mode !== 'no_reset') {
    await setOptions({ dtr: false, rts: true })

    await sleep(0.1)
    await setOptions({ dtr: true, rts: false })

    await sleep(0.5)
    await setOptions({ dtr: false })
  }
// above work well could print
// rst: 0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
// waiting for download

  await sleep(10)
   while (true) {
    await sync()
    console.log('sync')
    let i = 0
    while (i < 700) {
      const res = port.read()
      if (res !== null) return
      i++
    }
  }
}

const syncBuff = Buffer.from([0xc0, 0x00, 0x08, 0x24, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0x07, 0x12, 0x20, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0xc0])

function sync() {
  return new Promise((resolve) => {
    port.write(syncBuff, () => resolve())
  })
}

// when I send the SYNC, I could get nothing response.
```
I have sent the same buffer to the serial port as it is in python.
but no response form the port. Hope for a help.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

JackieLs, 2022-05-29:
the esptool.py works well, means the hareware is OK. But why under nodejs it is not woked.......

JackieLs, 2022-05-29:
```
// await setOptions({
//   dtr: true,
// })
```
I found an interesting thing.... when is comment this, it will work.....
respond like below: 
<Buffer c0 01 08 04 00 55 55 55 55 00 00 00 00 c0 c0 01 08 04 00 55 55 55 55 00 00 00 00 c0 c0 01 08 04>

dobairoland, 2022-05-30:
Hi @JackieLs. I'm sorry. We esptool developers don't have the experience with nodejs to help with you. You can find in the sources how the communication is done (in Python in the host side, in C in the ESP side for the flasher stub). You can find information regarding this in the documentation as well: https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/serial-protocol.html.

I can suggest to seek help at https://esp32.com/

closed 2022-05-30 by dobairoland

JackieLs, 2022-05-30:
Well, thanks for the reply. I would make a node-esptool if I could work out it. hhhhh

dobairoland, 2022-05-30:
One more thing. Use the `--trace` argument with `esptool.py` and you can see how the communication is done. ;-)