espressif/esptool #749: Pyinstaller recursion limit exceeded with esptool v4.x

mhummels, 2022-05-31:
### Operating System

Ubuntu 20.04

### Version

other

### Python Version

Python 3.9.2

### Full Esptool Command Line that Was Run

./helloworld.py

### Esptool Output

```shell
Traceback (most recent call last):
  File "PyInstaller/loader/pyimod03_importers.py", line 298, in get_code
  File "PyInstaller/loader/pyimod02_archive.py", line 290, in extract
  File "PyInstaller/loader/pyimod02_archive.py", line 87, in __enter__
RecursionError: maximum recursion depth exceeded while calling a Python object

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "helloworld.py", line 6, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
[... 277 lines cut ...]
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "esptool.py", line 31, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 478, in exec_module
  File "PyInstaller/loader/pyimod03_importers.py", line 300, in get_code
ImportError: Loader FrozenImporter cannot handle module esptool
[76066] Failed to execute script 'helloworld' due to unhandled exception!
```

### What is the Expected Behaviour?

Pyinstaller should be able to load esptool module correctly. With esptool 3.3 it works just fine, but when I get the latest PyPi Release it does not work and ends up with this recursive error.

### More Information

I tried also on another PC running Python 3.10.4 and esptool 4.1 and it works

radimkarnis, 2022-06-01:
Hello, @mhummels,

let me understand your issue correctly:

> Full Esptool Command Line that Was Run
 ./helloworld.py

You wrote a script `helloworld.py` which imports esptool and are trying to compile that script into executable binary with PyInstaller? (and that's where the recursion happens). Can you share ` helloworld.py` contents or the minimal code needed to reproduce the issue?

> I tried also on another PC running Python 3.10.4 and esptool 4.1 and it works

So the only difference between the PCs is the Python version (`3.9.2` vs `3.10.4`)? Could you please also report installed PyInstaller versions?

Thank you.
R.

MicroTether, 2022-06-15:
I get the same issue simply trying to flash a board or any usage of the command line tool or importing the module. I have uninstalled/reinstalled the package with no change.

radimkarnis, 2022-06-16:
@MicroTether, can you please provide more info about your environment? Python version, esptool version, OS, the output of `pip freeze`.

Does the recursion error also happen if you try running with python interpreter (clone esptool repo and run `python esptool.py -h`). Thank you!

MicroTether, 2022-06-24:
Python 3.10.4, esptool==4.1, Ubuntu 22.04 LTS. It does appear it works cloning and running esptool.py.

someburner, 2023-04-20:
Updating pyinstaller seemed to fix it for me:

`pyinstaller==5.10.1`

mhummels, 2023-05-23:
Sorry for the dump question, but I have installed esptool just using pip, so in what folder can I find the data then? I found that is is on my local machine `~/.local/lib/python3.10/site-packages/esptool/targets/stub_flasher`. But this path is not fixed, so anyone else using the spec file will fail to build the binary correctly.

So is there a way to get this right?

radimkarnis, 2023-06-01:
There are several reasons for the stub JSON files - easier maintenance, development process with git, distribution to other projects (they are used in [espflash](https://github.com/esp-rs/espflash) and [esptool.js](https://github.com/espressif/esptool-js)), stub building and wrapping,.... I understand this change might pose an additional step for some more advanced users, but it makes life easier for everyone who actually develops esptool.

@mhummels we build esptool with pyinstaller on every commit, see https://github.com/espressif/esptool/actions/workflows/build_esptool.yml. You can just download the latest binary. If you still want it to build yourself, you can pull the repository and build it [like this](https://github.com/espressif/esptool/blob/master/.github/workflows/build_esptool.yml#L51).

Installing `pyinstaller>=5.10.1` is verified to fix the recursion issue. There have also been recent [changes](https://github.com/espressif/esptool/commit/a7df5d3bee2579ce400d8d8b682546f97aa6b4cf) to prevent such self-imports in the future. 

Therefore, I am closing this issue. Thank you!

closed 2023-06-01 by radimkarnis

someburner, 2023-06-01:
@radimkarnis 

I haven't checked whether pyinstaller actually packages the json files or not. But it appears the absolute import is what breaks things. PyInstaller "works" but then when you actually go to run a program that used pyinstaller, it can't find the JSON files.

If I get some time I can try to look into whether the data is actually being packaged and just can't be found or if it's something else. I was on a pretty tight deadline to get my tool out.