espressif/esptool #752: ESP32 returns nothing but garbage

Mdes0, 2022-06-14:
### Operating System

macOS 12.3.1

### Version

latest master

### Python Version

Python 3.9

### Chip Description

ESP32-D0WDQ6-V3

### Device Description

Plain Module with TTGO T-Display soldered on to it by default.

### Hardware Configuration

Just the plain board.

### How is Esptool Run

macOS Terminal and Tonne

### Full Esptool Command Line that Was Run

esptool.py --trace -p /dev/cu.wchusbserial54240460861 flash_id

### Esptool Output

```shell
Serial port /dev/cu.wchusbserial54240460861
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.103 No serial data received.
.TRACE +0.053 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 No serial data received.
.TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.104 No serial data received.
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
[... 609 lines cut ...]
TRACE +0.105 No serial data received.
.TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 No serial data received.
.

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
puesche.matti@MacBook-Air-von-Matti ~ % screen /dev/cu.wchusbserial54240460861                        

��>0��>0�00�0�0�00�0�00�0�0>�0����>��>0�00�0�0�00�0�00�0�0��0>�0�00���00000��>������0�>��0��>���0�>�>��0��0��0��0�0�0����0�0����0�0��0�0>���0�0��0��0�����0���000000>0���>��>�>��0��>00��>�>�>�>��0��>0���>0�0�00�0�0�00�0�0�0��0>0�00���800��0>����0��>����0�>�>����0��0��0�0�0���0�0����0�0�00��0�0��0��0��0���0������0���>��>�>�0��>00��>�>�>�>��0�����0���>0�00�0�0�00�0��0��>�0>0�00��000��>����0��>���0�>�>��0�0��0���0�0����0�0���0�0��0�0>��0�0��0��0��0���0���000000>0���>��>�>�0��>00��>�>�>�>��0��>0���>0�00�0�0�00�0�00�0�0��0>�0�00���000��00>����0��0����000�>�>���0��0��0�0�0����0�0����0�0��0�0>���0�0�0����0��0�����00000>0����>�0��>00�>�>�>�>��0�����0���>0�00�0�0�00�0�00�0��0>�0�00���000��0>����0��>����>0���>0�00�0�0�00�0�00�0��>�0>0�00��000�0>����0��>���>0���>0�00�0�0�00�0�0�0�0>�0>0�00����>0�>0�000�0�00�0���>�0>0�00���000�>���0��0����0�>�>��00��00��0���0�0�0����0�0����0�0��0�0>0�0��0���0�0���8���000000>���>��>>��0��>00��>�>�>�>��0��>0���>0��00�00�0�00�0���>�0>0�00���000��>����0��>���0�>�>���00��0���0�0�0����0�0���0�0��0��0>���0�0����0��0���0������0���>��>�>�0��>00�����>�>�>��0��>0���>0�00�0�0�00�0�00�0�0��0>�0�00���000��0>����0��>���0�>�>�0��0��0��0�0�0����0�0���0�0��0�0>��0�0����0��0���0�����00000>0����>�>�0��>00��>�>�>�>��0��>0��>0�00�0�0�00�000�0��0>�0�00���000��0>����0��>���0�>�>��00��0�0��0�0�0��0����0�0��0�0�0�0��0��0��0��0�����0���>��>�>�0��>00��>�>�>�>��0�����0���>0��00�0�0�00�0�0>�0�00���000��0>������>���0�>�>�0�0��>0���>0�0�00�0�0�00�0�00�0��0>�0�00��000��0>���0��>���0�>�>��0��0��0���0�0�0����0�0���0�0��0�0>���0�0��0�0��0���0���00000>0�����>�>�0��>00�>���>�>�>�0

### Other Steps to Reproduce

Not really, my other esp8266 is working fine.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2022-06-14:
Hi @Mdes0. Do you have an auto-reset circuitry or have you put the chip into bootloader mode manually? I don't think so because the chip is not sending anything.

More about this: https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/entering-bootloader.html

Mdes0, 2022-06-14:
> Hi @Mdes0. Do you have an auto-reset circuitry or have you put the chip into bootloader mode manually? I don't think so because the chip is not sending anything.
> 
> More about this: https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/entering-bootloader.html

There is a button on my board to enter download mode which I tried using, but ended up with the same result.
[image]

dobairoland, 2022-06-14:
Are you holding the the button while you are running esptool?

BTW, you can connect with a serial terminal program using the settings from https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/serial-connection.html#serial-port-settings. After that you can enter bootloader mode manually based on https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader. You should be able to capture the serial bootloader message in the terminal program.

Mdes0, 2022-06-14:
Yes, I'm holding down the button while running esptool.
**esptool.py --trace -b 115200 -p /dev/cu.wchusbserial54240460861 flash_id**
`Serial port /dev/cu.wchusbserial54240460861
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.023 Read 1 bytes: 00
TRACE +0.000 Read invalid data: 00
TRACE +0.000 Remaining data in serial buffer: 
.TRACE +0.054 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 2 bytes: 0000
TRACE +0.000 Read invalid data: 0000
TRACE +0.000 Remaining data in serial buffer: 
.TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: 00
TRACE +0.000 Read invalid data: 00
TRACE +0.000 Remaining data in serial buffer: 
.TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read invalid data: 00
TRACE +0.000 Remaining data in serial buffer: 
.TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.032 Read 1 bytes: 00
TRACE +0.000 Read invalid data: 00
TRACE +0.000 Remaining data in serial buffer: 00
.TRACE +0.665 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.060 Read 1 bytes: 00
TRACE +0.000 Read invalid data: 00
TRACE +0.000 Remaining data in serial buffer: 00
.TRACE +0.055 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
[... 261 lines cut ...]
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.063 Read 1 bytes: 00
TRACE +0.000 Read invalid data: 00
TRACE +0.000 Remaining data in serial buffer: 00
.TRACE +0.055 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.038 Read 1 bytes: 00
TRACE +0.000 Read invalid data: 00
TRACE +0.000 Remaining data in serial buffer: 0000
.

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0x00): Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
`
Seems to have read one byte 🥲

dobairoland, 2022-06-14:
Have you tried my suggestion?

This doesn't look to me an esptool tool issue but a hardware one.

radimkarnis, 2022-06-15:
@Mdes0 

can you please try opening the port with any of [serial terminal programs](https://docs.espressif.com/projects/esptool/en/latest/esp8266/troubleshooting.html#serial-terminal-programs)? This will help to verify if there are any real data coming from the device. You can use `miniterm` the following way: `python -m serial.tools.miniterm PORT 74880`. If you connect and press the reset button, a [boot log](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#boot-log) should be printed.

If you see a boot log, there is no HW issue, but you are not switching into download mode correctly. If you don't see a boot log, something is seriously wrong and you should try going through all of the [troubleshooting guide](https://docs.espressif.com/projects/esptool/en/latest/esp8266/troubleshooting.html).

radimkarnis, 2022-07-01:
Closing this issue for inactivity. Feel free to reopen if needed!

closed 2022-07-01 by radimkarnis