espressif/esptool #753: ESP32S3 - fails to run firmware after flashing

andrew-elder, 2022-07-11:
### Operating System

Windows

### Esptool Version

v4.4.1

### Python Version

Python 3.8.7 (tags/v3.8.7:6503f05, Dec 21 2020, 17:59:51) [MSC v.1928 64 bit (AMD64)] on win32

### Chip Description

ESP32S3

### Device Description

Custom hardware.
[image]

Design is closely based on ESP32S2-DevKitM-1. 

### Hardware Configuration

Custom hardware uses the following GPIOs
```
GPIO_NUM_18  /**< Active low red LED drive */
GPIO_NUM_17  /**< Active low green LED drive */
GPIO_NUM_16  /**< Active low blue LED drive */

GPIO_NUM_48  /**< I2C bus 0 SDA */
GPIO_NUM_34  /**< I2C bus 0 SCL */

GPIO_NUM_15  /**< I2C bus 1 SDA */
GPIO_NUM_14  /**< I2C bus 1 SCL */

GPIO_NUM_21  /**< Active low output to power on sensors */
GPIO_NUM_12  /**< Active high output to power ADC input*/
GPIO_NUM_36  /**< Active low output to power on probe */

GPIO_NUM_9   /**< Active high output to enable detection circuitry */
GPIO_NUM_11  /**< Active low input used to detect probe */
GPIO_NUM_10  /**< Active low input used to detect probe */

GPIO_NUM_35  /**< Active high output used to disable USB CC pin read */
```

### How is Esptool Run

esptool is run from the command line

### Full Esptool Command Line that Was Run

 :: Write the encrypted "nvs" partition echo -- Write encrypted NVS partition python.exe %IDF_PATH%\components\esptool_py\esptool\esptool.py -p %PORT% --before default_reset ^     --after no_reset write_flash %START_NVS% manufacturing/bin/%SNFILESTUB%.bin  :: Flash the code signing key echo -- Write the code signing key partition python.exe %IDF_PATH%\components\esptool_py\esptool\esptool.py -p %PORT% --before default_reset ^     --after no_reset write_flash %START_CSK_PART% ./ota_mqtt_csk.bin  :: Write everything except the firmware echo -- Write bootloader, partition table, OTA initial data and NVS partition key python.exe %IDF_PATH%\components\esptool_py\esptool\esptool.py --chip esp32s3 -p %PORT% -b 460800 --before=default_reset ^     --after=no_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB ^     0x0 build/bootloader/bootloader.bin ^     0x8000 build/partition_table/partition-table.bin ^     0x11000 build/ota_data_initial.bin ^     %START_NVS_KEYS% manufacturing/keys/keys-%SNFILESTUB%.bin  :: Write the firmware (unencrypted the first time) echo -- Write sensor_firmware python.exe %IDF_PATH%\components\esptool_py\esptool\esptool.py --chip esp32s3 -p %PORT% -b 460800 --before=default_reset ^     --after=no_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB ^     0x20000 build/sensor-firmware.bin

### Esptool Output

```shell
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.1

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.21.1.
    C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin
[... 163 lines cut ...]
I (120) boot:  7 filesys          Unknown data     01 81 00620000 00100000
I (127) boot: End of partition table
I (131) boot: No factory image, trying OTA 0
. . . 
```

Don't see any "boot" messages output by the UART on our custom hardware.

Custom hardware on top of ESP32S3-DevKitM-1.
![PXL_20220708_205456636](https://user-images.githubusercontent.com/814195/178172241-60733636-c806-490d-b476-d138801f773c.jpg)

eFuse settings
[BlankBoardEfuses.txt](https://github.com/espressif/esptool/files/9080304/BlankBoardEfuses.txt)
[DevKitEfuses.txt](https://github.com/espressif/esptool/files/9080305/DevKitEfuses.txt)
[ProtoEFuses.txt](https://github.com/espressif/esptool/files/9080307/ProtoEFuses.txt)

### Other Steps to Reproduce

Not possible to reproduce without custom hardware, although a "blank" ESP32S3 on an ESP32S3-DevKitM-1 may reproduce the issue.

Need to understand why logging isn't working before looking at the invalid header issue.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

andrew-elder, 2022-07-11:
UART log after programming reads
```
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
invalid header: 0xacd54a85
invalid header: 0xacd54a85
...
```

dobairoland, 2022-07-11:
Hi @andrew-elder. Do you have any indication that this is an esptool issue? It looks to me that esptool works just fine.

What is printed after this?
```
I (131) boot: No factory image, trying OTA 0
```

I have checked the pins and the partition offsets. They look good to me.

Jason2866, 2022-07-11:
I had a similar behaviour when Flash type does not exactly match in type/size for Flash and PSRAM. The partition table is correct (size and entrys)?

IgorO, 2022-07-11:
Some additional info on what we are seeing- we can take an S3 dev board, run it through our flashing process and get our app to boot just fine. We then take our custom board with the blank S3, connect to ESP-PROG and get "Invalid header" scrolling on UART right out of the box, even before flashing. We then run it through the same flash process as above, it goes through with no errors and we are back to Invalid header" scrolling on UART with periodic watchdog reboots.

andrew-elder, 2022-07-11:
I made a mistake earlier when I tried to post efuse files. Here they are:

ESP32S3-DevKitM-1 efuse settings after programming
[DevKitEfuses.txt](https://github.com/espressif/esptool/files/9087352/DevKitEfuses.txt)

Our custom prototype efuse settings after programming
[ProtoEFuses.txt](https://github.com/espressif/esptool/files/9087353/ProtoEFuses.txt)

Our custom prototype board before programming
[BlankBoardEfuses.txt](https://github.com/espressif/esptool/files/9087356/BlankBoardEfuses.txt)

DIS_LEGACY_SPI_BOOT (BLOCK0)                       Disables Legacy SPI boot mode                      = True R/W (0b1) (0b0)  enabled on our board vs blank and DevKit

andrew-elder, 2022-07-11:
Also dump_mem doesn't seem to work on ESP32S3 per https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/advanced-commands.html#dump-a-memory-region-to-file-dump-mem

IgorO, 2022-07-11:
**Dump_me** returns with "A fatal error occurred: Invalid head of packet (0x47): Possible serial noise or corruption." but we are not sure if the memory should be readable with encrypted but not locked chips.

**Boot process on a working dev board  shows :**
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0270,len:0x2608
load:0x403b6000,len:0xb1c
load:0x403ba000,len:0x4258
entry 0x403b622c

**while custom board just sits at**

Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
invalid header: 0xacd54a85
invalid header: 0xacd54a85

dobairoland, 2022-07-12:
> What is the mechanism by which esptool "appears" to work, but then the esp32 does not boot?

Esptool is a low level tool. You can generate a binary of random bytes and flash the chip. It will succeed because the flashing succeeds. It does not and cannot check that the flashed binary is correct, e.g. can be executed.

> esptool can report programming success but leave the flash full of 0xff?

This is not the case. You are not getting `invalid header: 0xffffffff`.

> Dump_me returns with "A fatal error occurred: Invalid head of packet (0x47): Possible serial noise or corruption." but we are not sure if the memory should be readable with encrypted but not locked chips.

It wouldn't be too much for protection if you would be able to read back.

Have you followed https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html? Incorrect setup of encryption can result in that the application cannot be executed.

I'm sorry but I don't think we (esptool developers) can help with this.

You can seek help from the community (www.esp32.com) or contact the business development team (https://www.espressif.com/en/contact-us/technical-inquiries/other-technical-issues).

closed 2022-07-12 by dobairoland