espressif/esptool #760: can't burn_efuse esp32c3

stingdau1206, 2022-07-21:
### Operating System

ubuntu 22

### Esptool Version

v4.3.3

### Python Version

3.10.4

### Chip Description

esp32c3

### How is Esptool Run

terminal

### Full Esptool Command Line that Was Run

espefuse.py burn_efuse SPI_BOOT_CRYPT_CNT

### Esptool Output

```shell
Connecting...
Detecting chip type... ESP32-C3
espefuse.py v3.3.2-dev

=== Run "burn_efuse" command ===
The efuses to burn:
  from BLOCK0
     - SPI_BOOT_CRYPT_CNT

Burning efuses:
Traceback (most recent call last):
  File "/home/khoatn/.espressif/python_env/idf4.3_py3.10_env/lib/python3.10/site-packages/bitstring.py", line 831, in __new__
    return _cache[auto]
KeyError: 'uint:3=15'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/khoatn/.espressif/python_env/idf4.3_py3.10_env/lib/python3.10/site-packages/bitstring.py", line 1296, in _init_with_token
    b = cls(**{_tokenname_to_initialiser[name]: value})
KeyError: 'uint'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/khoatn/.espressif/python_env/idf4.3_py3.10_env/lib/python3.10/site-packages/bitstring.py", line 876, in _initialise
    init_without_length_or_offset[k](self, v)
KeyError: 'uint'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/khoatn/esp/esp-idf_v4.3.3/components/esptool_py/esptool/espefuse.py", line 198, in <module>
    _main()
  File "/home/khoatn/esp/esp-idf_v4.3.3/components/esptool_py/esptool/espefuse.py", line 191, in _main
    main()
  File "/home/khoatn/esp/esp-idf_v4.3.3/components/esptool_py/esptool/espefuse.py", line 178, in main
    operation_func(esp, efuses, args)
  File "/home/khoatn/esp/esp-idf_v4.3.3/components/esptool_py/esptool/espressif/efuse/base_operations.py", line 228, in burn_efuse
    print("\n    - '{}' ({}) {} -> {}".format(efuse.name, efuse.description, efuse.get_bitstring(), efuse.convert_to_bitstring(new_value)))
  File "/home/khoatn/esp/esp-idf_v4.3.3/components/esptool_py/esptool/espressif/efuse/base_fields.py", line 491, in convert_to_bitstring
    return BitArray(self.efuse_type + "={}".format(new_value))
  File "/home/khoatn/.espressif/python_env/idf4.3_py3.10_env/lib/python3.10/site-packages/bitstring.py", line 3117, in __new__
    y = Bits.__new__(BitArray, auto, length, offset, **kwargs)
  File "/home/khoatn/.espressif/python_env/idf4.3_py3.10_env/lib/python3.10/site-packages/bitstring.py", line 844, in __new__
    x._datastore._appendstore(Bits._init_with_token(*token)._datastore)
  File "/home/khoatn/.espressif/python_env/idf4.3_py3.10_env/lib/python3.10/site-packages/bitstring.py", line 1301, in _init_with_token
    b = cls(**{name: int(value), 'length': token_length})
  File "/home/khoatn/.espressif/python_env/idf4.3_py3.10_env/lib/python3.10/site-packages/bitstring.py", line 855, in __new__
    x._initialise(auto, length, offset, **kwargs)
  File "/home/khoatn/.espressif/python_env/idf4.3_py3.10_env/lib/python3.10/site-packages/bitstring.py", line 881, in _initialise
    init_with_length_only[k](self, v, length)
  File "/home/khoatn/.espressif/python_env/idf4.3_py3.10_env/lib/python3.10/site-packages/bitstring.py", line 1461, in _setuint
    raise CreationError(msg, uint, length, (1 << length) - 1)
bitstring.CreationError: 15 is too large an unsigned integer for a bitstring of length 3. The allowed range is [0, 7].
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

KonstantinKondrashov, 2022-07-21:
Hi @stingdau1206!
I see that this error occurs when SPI_BOOT_CRYPT_CNT is maximum = 0x7 (all 3 bits are set). In this case, this efuse can no longer be burned. Is that your case?
Could you take a look `espefuse.py summary`? you will be able to see the log like:

```
SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables encryption and decryption, when an SPI boo = Enabled R/W (0b111)
                                                   t mode is set. Enabled when 1 or 3 bits are set,di
```

I will make this error more user friendly.

stingdau1206, 2022-07-21:
hi @KonstantinKondrashov 
this is summary output cmd:
Detecting chip type... ESP32-C3
espefuse.py v3.3.2-dev

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
VDD_SPI_AS_GPIO (BLOCK0)                           Set this bit to vdd spi pin function as gpio       = False R/W (0b0)
BTLC_GPIO_ENABLE (BLOCK0)                          Enable btlc gpio                                   = 0 R/W (0b00)
POWERGLITCH_EN (BLOCK0)                            Set this bit to enable power glitch function       = False R/W (0b0)
POWER_GLITCH_DSENSE (BLOCK0)                       Sample delay configuration of power glitch         = 0 R/W (0b00)
DIS_DIRECT_BOOT (BLOCK0)                           Disables direct boot mode                          = False R/W (0b0)
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             Disables USB-Serial-JTAG ROM printing              = False R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FORCE_SEND_RESUME (BLOCK0)                         Force ROM code to send a resume command during SPI = False R/W (0b0)
                                                    bootduring SPI boot                              
ERR_RST_ENABLE (BLOCK0)                            Use BLOCK0 to check error record registers         = without check R/W (0b0)
BLOCK_USR_DATA (BLOCK3)                            User data                                         
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Efuse fuses:
WR_DIS (BLOCK0)                                    Disables programming of individual eFuses          = 8388864 R/W (0x00800100)
RD_DIS (BLOCK0)                                    Disables software reading from BLOCK4-10           = 1 R/W (0b0000001)

Flash Config fuses:
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms                                              

Identity fuses:
SECURE_VERSION (BLOCK0)                            Secure version (used by ESP-IDF anti-rollback feat = 0 R/W (0x0000)
                                                   ure)                                              
MAC (BLOCK1)                                       Factory MAC Address                               
   = 10:91:a8:34:c7:d8 (OK) R/W 
WAFER_VERSION (BLOCK1)                             WAFER version                                      = 3 R/W (0b011)
PKG_VERSION (BLOCK1)                               Package version                                    = ESP32-C3 R/W (0b000)
BLOCK1_VERSION (BLOCK1)                            BLOCK1 efuse version                               = 4 R/W (0b100)
OPTIONAL_UNIQUE_ID (BLOCK2)                        Optional unique 128-bit ID                        
   = 86 6c 1d 3f a4 e1 8a a5 ee 85 9a be 74 3e 79 18 R/W 
BLOCK2_VERSION (BLOCK2)                            Version of BLOCK2                                  = 5 R/W (0b101)
CUSTOM_MAC (BLOCK3)                                Custom MAC Address                                
   = 00:00:00:00:00:00 (OK) R/W 

Jtag Config fuses:
SOFT_DIS_JTAG (BLOCK0)                             Software disables JTAG. When software disabled, JT = 0 R/W (0b000)
                                                   AG can be activated temporarily by HMAC peripheral
DIS_PAD_JTAG (BLOCK0)                              Permanently disable JTAG access via pads. USB JTAG = False R/W (0b0)
                                                    is controlled separately.                        

Security fuses:
DIS_DOWNLOAD_MANUAL_ENCRYPT (BLOCK0)               Disables flash encryption when in download boot mo = False R/W (0b0)
                                                   des                                               
SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables encryption and decryption, when an SPI boo = Enable R/W (0b111)
                                                   t mode is set. Enabled when 1 or 3 bits are set,di
                                                   sabled otherwise                                  
SECURE_BOOT_KEY_REVOKE0 (BLOCK0)                   If set, revokes use of secure boot key digest 0    = False R/W (0b0)
SECURE_BOOT_KEY_REVOKE1 (BLOCK0)                   If set, revokes use of secure boot key digest 1    = False R/W (0b0)
SECURE_BOOT_KEY_REVOKE2 (BLOCK0)                   If set, revokes use of secure boot key digest 2    = False R/W (0b0)
KEY_PURPOSE_0 (BLOCK0)                             KEY0 purpose                                       = XTS_AES_128_KEY R/- (0x4)
[... cut ...]
DIS_USB_DEVICE (BLOCK0)                            Disables USB DEVICE                                = False R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                            Exchanges USB D+ and D- pins                       = False R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disables USB-Serial-JTAG download feature in UART  = False R/W (0b0)
                                                   download boot mode                                

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = False R/W (0b0)

stingdau1206, 2022-07-21:
can't I reset to burn again :(

KonstantinKondrashov, 2022-07-21:
FYI, https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/security/flash-encryption.html#relevant-efuses
SPI_BOOT_CRYPT_CNT is max, so you can not turn off flash encryption back.

closed 2022-07-26 by espressif-bot