espressif/esptool #761: Docs: -e / —erase-all option does not exist

andypiper, 2022-07-22:
### Operating System

All

### Esptool Version

4.x

### Python Version

3.10

### More Information

https://docs.espressif.com/projects/esptool/en/latest/esp32s2/esptool/basic-commands.html

This page refers to a -e option to erase the chip before flashing - no such option exists.

radimkarnis, 2022-07-25:
Hello @andypiper,

to perform a full flash chip erase, you can use the `erase_flash` command. 

When writing to flash with the `write_flash` command, you can append the `-e/--erase-all` option to erase the flash chip beforehand (without having to run a separate `erase_flash` command). This option indeed exists:

```
$ esptool.py write_flash --erase-all 0 in.bin

Connecting....
Detecting chip type... ESP32-C2
Chip is ESP32-C2 (revision 1)
Features: Wi-Fi
Crystal is 26MHz
MAC: 10:97:bd:f1:db:a8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Erasing flash (this may take a while)...         # You can see the erase happening here ...
Chip erase completed successfully in 0.8s.       # ... and finishing here
Compressed 8 bytes to 16...
Wrote 8 bytes (16 compressed) at 0x00000000 in 0.0 seconds (effective 1.4 kbit/s)...
Hash of data verified.
```

If you run esptool.py help, you can see the option being present:

```
$ esptool.py write_flash -h
usage: esptool write_flash [-h] [--erase-all] ... 
```

andypiper, 2022-07-25:
Thanks for the update, I'll recheck that locally!

radimkarnis, 2022-07-28:
@andypiper do you want to discuss more, or can I close this issue? Thanks!

andypiper, 2022-07-28:
Thanks for asking! Go ahead and close; you've answered!

closed 2022-07-28 by radimkarnis