espressif/esptool #763: stub_flasher_32c3.json not found

reppad, 2022-08-04:
### Operating System

Windows 10

### Esptool Version

v4.2

### Python Version

no local python (prebuilt win64 esptool.exe)

### Chip Description

ESP32-C3 native USB

### Device Description

LuatOS CORE-ESP32 ("new version": the one using ESP internal USB instead of CH343)
[image]

### Hardware Configuration

nothing connected

### How is Esptool Run

No IDE, direct from PowerShell

### Full Esptool Command Line that Was Run

.\esptool.exe -p COM5 -b 460800 flash_id

### Esptool Output

```shell
PS C:\softs\esptool-v4.2-win64> .\esptool.exe -p COM5 -b 460800 flash_id
esptool.py v4.2
Serial port COM5
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:79:a4:5c
Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
  File "esptool\__init__.py", line 1004, in _main
  File "esptool\__init__.py", line 684, in main
  File "esptool\loader.py", line 891, in run_stub
  File "esptool\loader.py", line 136, in __init__
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\Reppad\\AppData\\Local\\Temp\\_MEI39042\\esptool\\./targets/stub_flasher/stub_flasher_32c3.json'
[1860] Failed to execute script 'esptool' due to unhandled exception!
```

### More Information

The error appears with any command I tried (flash_id, chip_id, read_flash, write_flash...).
No tested with other ESP variant nor throught an usb to UART adapter.

All works fine with previous esptool versions.
I use v4.1 for now :)

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-08-04:
Hi @reppad,

thanks for the report! Seems like pyInstaller is not bundling the stub JSON files. Will look into this!

radimkarnis, 2022-08-04:
@reppad, I tried drafting a solution in a fork repo. If you download the windows artifact from [this run](https://github.com/radimkarnis/esptool/actions/runs/2798213999), can you please verify the issue is solved? Thanks!

reppad, 2022-08-04:
@radimkarnis Seems to work well with the artifact from your run.
```
PS C:\Users\Reppad\Downloads\esptool-win64> .\esptool.exe -p COM5 -b 460800 flash_id
esptool.py v4.3-dev
Serial port COM5
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:79:a4:5c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Manufacturer: cd
Device: 6016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

radimkarnis, 2022-08-04:
@reppad great, thanks a lot for confirmation. I will keep this issue open until we merge the fix.

closed 2022-08-08 by espressif-bot