espressif/esptool #765: Cannot flash ESP32C3 chip (Wrong boot mode detected (0x1)!)

marco711, 2022-08-08:
### Operating System

Debian 5.10.120-1

### Esptool Version

4.2

### Python Version

3.9.2

### Chip Description

ESP32C3

### Device Description

ESP32C3, customized board connected to two sensors communicating in I2C.

### Hardware Configuration

 Wrong boot mode detected (0x1)! The chip needs to be in download mode.
GPIO0 --> DAC
GPIO1 --> SCL
GPIO2 --> SDA
GPIO3 --> SCL
GPIO19 --> SDA
GPIO18 --> GPIO OUT
GPIO21 --> TX PROGRAMMING
GPIO20 --> RX PROGRAMMING
GPIO10 --> GPIO OUT
GPIO4 --> TMS
GPIO5 --> TDI
GPIO6 --> TCK
GPIO7 --> TDO
GPIO8 --> GPIO IN
GPIO9 --> GPIO IN

### How is Esptool Run

Linux terminal

### Full Esptool Command Line that Was Run

idf.py -p /dev/ttyUSB1 flash

### Esptool Output

```shell
Executing action: flash
Running ninja in directory /home/msurca/esp/IoTlift/mqtt-tcp/build
Executing "ninja flash"...
[1/5] cd /home/msurca/esp/IoTlift/mqtt...sp/IoTlift/mqtt-tcp/build/mqtt_tcp.bin
mqtt_tcp.bin binary size 0xcf610 bytes. Smallest app partition is 0x100000 bytes. 0x309f0 bytes (19%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/msurca/esp/IoTlift/mqtt-tcp/build/bootloader/esp-idf/esptool_py && /home/msurca/.espressif/python_env/idf5.0_py3.9_env/bin/python /home/msurca/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/msurca/esp/IoTlift/mqtt-tcp/build/bootloader/bootloader.bin
Bootloader binary size 0x5180 bytes. 0x2e80 bytes (36%) free.
[2/3] cd /home/msurca/esp/esp-idf/comp...nents/esptool_py/run_serial_tool.cmake
esptool esp32c3 -p /dev/ttyUSB1 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 mqtt_tcp.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.2
Serial port /dev/ttyUSB1
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-C3: Wrong boot mode detected (0x1)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:55 (message):
  
  /home/msurca/.espressif/python_env/idf5.0_py3.9_env/bin/python;;/home/msurca/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32c3
  failed

FAILED: CMakeFiles/flash 
cd /home/msurca/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/msurca/esp/esp-idf -D "SERIAL_TOOL=/home/msurca/.espressif/python_env/idf5.0_py3.9_env/bin/python;;/home/msurca/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32c3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/msurca/esp/IoTlift/mqtt-tcp/build -P /home/msurca/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### More Information

Before I was able to flash the ESP32. The problem started after I flashed a code calling twice the 'esp_event_loop_create_default()' function.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2022-08-08:
Hi @marco711. Do you have a reset circuitry? (I don't think so based on your PIN description). Esptool has some requirements to be able to put the chip into download mode automatically: https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html#automatic-bootloader

Please also take a look at Chapter 7 of ESP32-C3 TRM. "GPIO2" used for SDA is also a strapping pin. It needs to be high during boot.

> Before I was able to flash the ESP32. The problem started after 

I think the device was put manually into bootloader mode. Later it was restarted and flashing doesn't work anymore. If you don't have a reset circuitry then every time (before flashing) you will have to put the chip into bootloader mode manually.

marco711, 2022-08-09:
Hello @dobairoland. No, there is not reset circuitry. The EN pin is connected to VCC through a 22 K resistor, to GND through a 100 nF capacitor and to a Reset button connected to GND.
I have tried by pulling GPIO9 down and pulling up GPIO2 and GPIO8 (Download Boot mode) and this way I am able to program the ESP32. Nevertheless, I would like to know if it is possible to restore the chip to its initial state in order to avoid putting it manually into Download Boot mode every time I need to flash it.

dobairoland, 2022-08-09:
> Nevertheless, I would like to know if it is possible to restore the chip to its initial state in order to avoid putting it manually into Download Boot mode every time I need to flash it.

I'm afraid there is no initial state of the chip and you need to use manual reset into bootloader mode if you don't have a reset circuitry connected to the serial interface.

closed 2022-08-09 by dobairoland