espressif/esptool #771: Couldn't open port, the port doesn't exist. Random ESP32 deaths and resurrections.

czupirrek, 2022-08-25:
### Operating System

Windows 10

### Esptool Version

esptool.py v4.3-dev

### Python Version

Python 3.10.6

### Chip Description

ESP-Wroom-32

### Device Description

just the esp32 module connected via micro usb cable (of confirmed quality) to a pc

### Hardware Configuration

there is nothing else connected to the board

### How is Esptool Run

windows command line/cmd.exe

### Full Esptool Command Line that Was Run

esptool.py --trace -p 10 erase_flash

### Esptool Output

```shell
esptool.py v4.3-dev
Serial port 10

A fatal error occurred: Could not open 10, the port doesn't exist
```

### More Information

After being plugged in for extended periods of time (sometimes ~30min, ~1hr) the board would randomly decide to start working again. The device is currently flashed with Tasmota software, which I've been tinkering with today. At the moment of "death", the board was not being used actively, it was only lazily publishing mqtt data from a sensor. At this moment I cannot access the ESP's log through USB (I've tried Arduino IDE serial monitor, which printed out nothing, only a blank page) or flash new firmware. I see the device pop up in Windows Device Manager and I hear the sound of it being plugged in. I'm sure of the micro-usb cable's quality as it had worked fine before. Holding the EN/BOOT button doesn't seem to work.

### Other Steps to Reproduce

This issue seems to have risen after I tried Tasmota through their webpage. The installation froze midway and I pulled the plug on ESP. Later I flashed it again, this time with NodeMCU-PyFlasher, and it had been working fine for a few days with one or two random "seizures" in which the board was totally unresponsive for a random period of time.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2022-08-26:
Hello @czupirrek,

it doesn't matter what code your ESP is currently running. If there is a working serial connection and the board is switched to download mode, esptool.py should work.

Based on the fact you can see a new device connected and hear a sound, the USB-to-UART bridge on your devkit is working. I don't understand you selecting the port as `-p 10`, though. Esptool tells you that this "port" doesn't exist. On windows, serial ports are usually named `COMX`, in your case `COM10`, I suppose.

1) If you run esptool with ``-p COM10` (or however else the serial port is named right now), does it work?
2) If you connect to the chip with a serial terminal (Arduino, or `python -m serial.tools.miniterm COM10 115200`) and start pressing the `RESET` button, do you see any output?

czupirrek, 2022-08-26:
Hi @radimkarnis,
thanks for the reply.

1. 
> If you run esptool with -p COM10 (or however else the serial port is named right now), does it work?
No, it still doesn't work. Here's a snippet of the output:

`    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.117 No serial data received.
.TRACE +0.117 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.114 No serial data received.
.

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html`

I ran `-p 10` instead of `-p COM10` since I automatically assumed it would just mean COM. My bad.

2.
>If you connect to the chip with a serial terminal (Arduino, or python -m serial.tools.miniterm COM10 115200) and start pressing the RESET button, do you see any output?
No. The output is blank still. Same with the BOOT button (which used to put the device to download mode before).

I should also mention now that the board is ambient termperature (cold) and that it used to heat up to ~50 deg. Celsius before, which worries me.

radimkarnis, 2022-08-26:
Sorry to tell you, but this seems like a serious hardware issue.

The USB-to-UART bridge works (that's why you see a serial port and can connect to it), but the ESP chip is probably dead (otherwise you would see at least some data).

>  it used to heat up to ~50 deg

This is also alarming, the ESP shouldn't heat up that much when running Tasmota.

If this was a cheaper low-quality devkit, I would just advise you to replace it.

radimkarnis, 2022-09-01:
Will close this issue. Feel free to reopen if you feel we should discuss this more!

closed 2022-09-01 by radimkarnis